google/aiyprojects-raspbian #257: step 13 page 45 book start voice unit

Lepracon68, 2018-01-17:
In the book "Magpi essentials" AIY projects , page 45 , step 13 it says:
start the voice unit
Double click the "start dev terminal" icon and enter:
src/assistant_library_demo.py

That gives back the error message:
No such file or directory

when i search i found them at:
src/examples/voice

When i enter there (in src/examples/voice):
assistant_library_demo.py

it returns:
bash: assistant_library_demo.py: command not found

But when i do "ls" i see it's there and green.

I am new to this RaspberryPI thing and Voice is my first thing i do. Can anyone tell me how to proceed ?

Lepracon68, 2018-01-17:
i found on the internet https://aiyprojects.withgoogle.com/voice#users-guide-3-using-your-device

3.1. START THE ASSISTANT LIBRARY DEMO APP
For the device to begin acting as your Google Assistant much like Google Home, start the assistant library demo app by double-clicking "Start dev terminal" on the Desktop and entering:

src/examples/voice/assistant_library_demo.py

The assistant library app has hotword detection built-in. To start a conversation with the Google Assistant, say "Okay, Google" or "Hey Google". When you are done, press Ctrl-C to end the application.

This returns :
You need client secrets to use the assistant API

So , also not correct, but with a solution beneath it (follow these instructions), so i am moving on and destroying the book which came with the package

Lepracon68, 2018-01-17:
my mistake, 
assitent.json  instead of assistant.json, changed that, and will try on

drigz, 2018-01-17:
The book is still pretty :)

But yes, unfortunately it is not up-to-date for the latest image - the examples moved when the Vision Kit was published to avoid confusing Vision Kit users. I'll look into adding a warning to help users like yourself.

Lepracon68, 2018-01-17:
Thanks . i am a bit of a noob here, but loving it so far !

Lepracon68, 2018-01-17:
It works with the assitant_library_demo
Thank you.

drigz, 2018-01-26:
Glad to hear it's working :) We've added a warning which will be in the next image.

closed 2018-01-26 by drigz