google/aiyprojects-raspbian #261: VoiceHat not working

markjhughes, 2018-01-23:
Hey ya'.  I have been tasked with attaching a voicehat to a raspberrypi3, and unfortunately, things are not working.

First attempt was to use the "hack" instructions to add it to a preexisting Raspbian installation.  I noticed there are several discrepancies between the project name in the instructions and the one referenced on github. 

Second attempt was to write the aiyprojects-2018-01-03.img.xy to a fresh microsd card.  No luck.  The first error messages appear in dmesg:
[    3.405055] i2c i2c-1: Failed to register i2c client aiy-io-i2c at 0x51 (-16)
[    3.405072] i2c i2c-1: of_i2c: Failure registering /soc/i2c@7e804000/aiy-gpio-i2c@51
[    3.405082] i2c i2c-1: Failed to create I2C device for /soc/i2c@7e804000/aiy-gpio-i2c@51
[    3.416030] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    3.490891] voicehat-codec voicehat-codec: property 'voicehat_sdmode_delay' not found default 5 mS
[    3.495824] snd-googlevoicehat-soundcard soc:sound: voicehat-hifi <-> 3f203000.i2s mapping ok
[    3.515543] Error: Driver 'voicehat-codec' is already registered, aborting...
[    3.692805] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    3.700512] usbcore: registered new interface driver brcmfmac
[    3.908679] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    3.909558] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.41 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-08-07 00:37:47 
[    3.993699] aiy-io-i2c 1-0051: AIYIO Status:
[    3.993714] aiy-io-i2c 1-0051: Failed to initialize status message
[    3.994448] ktd202x 1-0030: ktd202x driver loaded for a ktd2026.
[    3.996116] ktd202x 1-0030: could not set power management for the chip (-121)!

The Check cloud and check WiFi scripts run fine, but CheckAudio returns:
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 158, in <module>
    main()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 153, in main
    do_checks()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 111, in do_checks
    if not check_voicehat_present():
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 70, in check_voicehat_present
    card_id = CARDS_ID[get_aiy_device_name()]
KeyError: None
Press Enter to close...

aplay -l provides the following information
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Any suggestions would be appreciated.  Please let me know if you require additional information.
Thanks,
Mark

markjhughes, 2018-01-23:
Okay -- so I went back into AIY-projects-python and issued "git checkout voicekit" and now it is working.  Have you folks changed the git repository structure recently?

drigz, 2018-01-23:
Thanks for the report, Mark. It sounds like a duplicate of https://github.com/google/aiyprojects-raspbian/issues/255. Do the demo scripts work even though Check Audio fails?

> I noticed there are several discrepancies between the project name in the instructions and the one referenced on github.

Could you point me to the discrepancies you noticed? I can only see references to `AIY-projects-python` in [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md), which is the path we want to use from now on.

markjhughes, 2018-01-23:
Hi drigz,
     It was worse than that -- the errors occurred and there was no audio until I checked out the voice brance again.  The projects with varying names include the user guide at 
https://aiyprojects.withgoogle.com/voice#makers-guide-1-2--python-api-reference which points users towards:  https://github.com/google/aiyprojects-raspbian/tree/voicekit/src/aiy.

I'm not worried about the hacking part not working on my Raspbian installation -- I had tinkered with that for a long while and changed many things.  I'm more concerned about it not working out of the box on the SD card distribution.  

Perhaps the issue is that I worked from sets of competing instructions?  The MagPI and the AIY Projects guide?

I'm going to reinstall on a fresh SD card to see what happens.  Go ahead and close for now and if I can recreate the error, I'll open a new, more detailed case.

Thanks,
Mark

drigz, 2018-01-24:
>  I'm more concerned about it not working out of the box on the SD card distribution.

Call me an idealist, but I'd like to fix both :) Thanks for pointing out the errors on the website, I've got a fix in process.

The Check Audio issue (#255) is linked to the hardware. Voice Kits from the first production run didn't get their EEPROM correctly programmed. This wasn't an issue in the first AIY images as they were only designed to work with the Voice HAT, but when Vision Kit support was added, the Check Audio script stopped working with these Voice HATs.

> It was worse than that -- the errors occurred and there was no audio until I checked out the voice brance again.

Do you mean that even with the aiyprojects-2018-01-03 image, the demo scripts didn't work until you switched to the `voicekit` branch? If so, that certainly is worse that I was expecting, as the demo scripts work for me even on my older hardware.

Could you post the error you get with the demo scripts on the aiyprojects-2018-01-03 image?

Thanks for bearing with me!

markjhughes, 2018-01-24:
Hi Rodrigo -- I'm happy to play along.  To answer your question "Do you
mean that even with the aiyprojects-2018-01-03 image, the demo scripts
didn't work until you switched to the voicekit branch", yes, that's what I
meant -- and that's with the DIY image.  So, after I switched to the
voicekit branch, the errors disappeared and the kit began working.

So here's my plan today (while I'm doing other things of course).
Attempt the Hacking.md instructions from a fresh install of Raspbian
Stretch on a Raspberry Pi 3 B+ on a 16 GB SD card.  I ran apt-get update &&
apt-get upgrade last night, so it should be ready to go today.  I'll keep
track of history and any errors that I might encounter (if I encounter them
again).

Then I'll try it off of the aiyprojects-2018-01-03 image on an different SD
card and keep track of the same.

drigz, 2018-01-24:
OK, good luck and let me know how it goes! A heads up: the path to the example script in HACKING.md is wrong (cf #264)

markjhughes, 2018-01-24:
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

Follow Procedure Produced Errors:
Download https://www.raspberrypi.org/downloads/raspbian/ unzip, and burn to SD card with https://etcher.io/

From GUI Enable SSH and VNC
Connect to Wireless Network  (rest of commands run remotely via ssh)
    1  sudo apt-get update && sudo apt-get upgrade -y
    2  sudo apt-get install git
    3  git clone https://github.com/google/aiyprojects-raspbian.git AIY-projects-python
    4  cd ~/AIY-projects-python/
    5  scripts/install-deps.sh 
    6  sudo scripts/install-services.sh
    7  sudo scripts/configure-driver.sh
    8  sudo reboot
    9  cd ~/AIY-projects-python/
   10  sudo scripts/install-alsa-config.sh
   11  python3 checkpoints/check_audio.py <-- Errors here
   12  uname -a
   13  history >> ~/Desktop/history.txt

Results of history line 11:
_pi@raspberrypi:~/AIY-projects-python $ python3 checkpoints/check_audio.py 
Traceback (most recent call last):
  File "checkpoints/check_audio.py", line 27, in <module>
    import aiy.audio  # noqa
ImportError: No module named 'aiy'

markjhughes, 2018-01-24:
Okay -- so the quick fix is to again run 'git checkout voicekit'

I ran grep -r 'aiy.audio' * in the project directory before and after I ran the git command.
**Before:**
checkpoints/check_audio.py:import aiy.audio  # noqa
checkpoints/check_audio.py:    aiy.audio.play_wave(TEST_SOUND_PATH)
checkpoints/check_audio.py:        aiy.audio.record_to_wave(temp_path, RECORD_DURATION_SECONDS)
checkpoints/check_audio.py:        aiy.audio.play_wave(temp_path)
src/examples/voice/cloudspeech_demo.py:import aiy.audio
src/examples/voice/cloudspeech_demo.py:    aiy.audio.get_recorder().start()
src/examples/voice/assistant_grpc_demo.py:import aiy.audio
src/examples/voice/assistant_grpc_demo.py:    with aiy.audio.get_recorder():
src/examples/voice/assistant_grpc_demo.py:                aiy.audio.play_audio(audio)
src/examples/voice/assistant_library_with_local_commands_demo.py:import aiy.audio
src/examples/voice/assistant_library_with_local_commands_demo.py:    aiy.audio.say('Good bye!')
src/examples/voice/assistant_library_with_local_commands_demo.py:    aiy.audio.say('See you in a bit!')
src/examples/voice/assistant_library_with_local_commands_demo.py:    aiy.audio.say('My IP address is %s' % ip_address.decode('utf-8'))
src/aiy/cloudspeech.py:import aiy.audio
src/aiy/cloudspeech.py:        self._recorder = aiy.audio.get_recorder()
src/aiy/cloudspeech.py:                    aiy.audio.say('OK, which one?')
src/aiy/assistant/grpc.py:import aiy.audio
src/aiy/assistant/grpc.py:        self._recorder = aiy.audio.get_recorder()
src/aiy/assistant/grpc.py:                aiy.audio.play_audio(audio)
src/aiy/assistant/grpc.py:                aiy.audio.play_audio(audio)
src/aiy/audio.py:    use this. Instead, use 'aiy.audio.play_wave' if you would like to play some
src/aiy/audio.py:    Example: aiy.audio.say('This is an example', lang="en-US", volume=75, pitch=135)
src/aiy/audio.py:        volume = aiy.audio.get_tts_volume()
src/aiy/audio.py:        pitch = aiy.audio.get_tts_pitch()
src/aiy/audio.py:    aiy._drivers._tts.say(aiy.audio.get_player(), words, lang=lang, volume=volume, pitch=pitch)
src/aiy/_drivers/_status_ui.py:import aiy.audio
src/aiy/_drivers/_status_ui.py:            aiy.audio.play_wave(self._trigger_sound_wave)

**After:**
checkpoints/check_audio.py:import aiy.audio  # noqa
checkpoints/check_audio.py:    aiy.audio.play_wave(TEST_SOUND_PATH)
checkpoints/check_audio.py:        aiy.audio.record_to_wave(temp_path, RECORD_DURATION_SECONDS)
checkpoints/check_audio.py:        aiy.audio.play_wave(temp_path)
src/cloudspeech_demo.py:import aiy.audio
src/cloudspeech_demo.py:    aiy.audio.get_recorder().start()
src/assistant_grpc_demo.py:import aiy.audio
src/assistant_grpc_demo.py:    with aiy.audio.get_recorder():
src/assistant_grpc_demo.py:                aiy.audio.play_audio(audio)
Binary file src/aiy/__pycache__/audio.cpython-35.pyc matches
src/aiy/cloudspeech.py:import aiy.audio
src/aiy/cloudspeech.py:        self._recorder = aiy.audio.get_recorder()
src/aiy/assistant/grpc.py:import aiy.audio
src/aiy/assistant/grpc.py:        self._recorder = aiy.audio.get_recorder()
src/aiy/assistant/grpc.py:                aiy.audio.play_audio(audio)
src/aiy/assistant/grpc.py:                aiy.audio.play_audio(audio)
src/aiy/audio.py:    use this. Instead, use 'aiy.audio.play_wave' if you would like to play some
src/aiy/audio.py:    Example: aiy.audio.say('This is an example', lang="en-US", volume=75, pitch=135)
src/aiy/audio.py:        volume = aiy.audio.get_tts_volume()
src/aiy/audio.py:        pitch = aiy.audio.get_tts_pitch()
src/aiy/audio.py:    aiy._drivers._tts.say(aiy.audio.get_player(), words, lang=lang, volume=volume, pitch=pitch)
src/aiy/_drivers/_status_ui.py:import aiy.audio
src/aiy/_drivers/_status_ui.py:            aiy.audio.play_wave(self._trigger_sound_wave)
src/assistant_library_with_local_commands_demo.py:import aiy.audio
src/assistant_library_with_local_commands_demo.py:    aiy.audio.say('Good bye!')
src/assistant_library_with_local_commands_demo.py:    aiy.audio.say('See you in a bit!')
src/assistant_library_with_local_commands_demo.py:    aiy.audio.say('My IP address is %s' % ip_address.decode('utf-8'))

Now that's a whole lot to take in in a text browser, so I ran diff to look for differences in where aiy.audio appears.  Granted this is a limited sample focused on the one keyword I chose to lookfor -- but it appears that the file structure is moved around after import -- this shows that files originally in src/examples/voice/ are moved up two directories.

What would you like me to do next?

markjhughes, 2018-01-24:
As I continue down this merry road of issues, a new one appears.

pi@raspberrypi:~/AIY-projects-python/checkpoints $ python3 check_cloud.py
Traceback (most recent call last):
  File "check_cloud.py", line 26, in <module>
    import aiy._apis._speech  # noqa
  File "/home/pi/AIY-projects-python/src/aiy/_apis/_speech.py", line 25, in <module>
    import google.auth
ImportError: No module named 'google'

markjhughes, 2018-01-24:
Alright -- so, we need to include the following in the instructions:
instead of saving client secrets to assistant.json, save them to cloud_speech.json
Then run

source env/bin/activate
python3 checkpoints/check_cloud.py

markjhughes, 2018-01-24:
Alright -- that's it for the fresh Raspbian installation.  I'm going to continue with my work for the day before trying the next one.  I'm already a couple hours behind playing with this.  But to summarize:

After fresh Raspbian Installation:
sudo apt-get update && sudo apt-get upgrade -y
sudo apt-get install git
git clone https://github.com/google/aiyprojects-raspbian.git AIY-projects-python
cd ~/AIY-projects-python/
scripts/install-deps.sh
sudo scripts/install-services.sh
sudo scripts/configure-driver.sh
sudo reboot
cd ~/AIY-projects-python/
sudo scripts/install-alsa-config.sh
python3 checkpoints/check_audio.py

Do the google authentication stuff -- save to ~/cloud_speech.json
If everything works, great.  If not run 'git checkout voicekit'
'source env/bin/activate'
'python3 ~/AIY-projects-python/checkpoints/check_cloud.py'

Here is the complete history (including things that didn't work)
    1  sudo apt-get update && sudo apt-get upgrade -y
    2  sudo apt-get install git
    3  git clone https://github.com/google/aiyprojects-raspbian.git AIY-projects-python
    4  cd ~/AIY-projects-python/
    5  scripts/install-deps.sh 
    6  sudo scripts/install-services.sh
    7  sudo scripts/configure-driver.sh
    8  sudo reboot
    9  cd ~/AIY-projects-python/
   10  sudo scripts/install-alsa-config.sh
   11  python3 checkpoints/check_audio.py 
   12  uname -a
   13  history >> ~/Desktop/history.txt 
   14  diff -y grepaiyaudio.txt grepaiyaudioafter.txt aiyaudiodiff.txt
   15  diff -y grepaiyaudio.txt grepaiyaudioafter.txt >> aiyaudiodiff.txt
   16  mv "Desktop/client_secret_832804580228-llt01rq14lj67eru99k81gqh8qu7p165.apps.googleusercontent.com.json" Desktop/assistant.json
   17  head client_secret_832804580228-llt01rq14lj67eru99k81gqh8qu7p165.apps.googleusercontent.com.json 
   18  cd ../AIY-projects-python/
   19  cd checkpoints/
   20  python3 check_cloud.py
   21  python3 -c 'import google; print(google.__file__)'
   22  pip install google-auth
   23  python3 check_cloud.py
   24  pip install --upgrade google-api-python-client
   25  python3 check_cloud.py
   26  easy_install --upgrade google-api-python-client
   27  sudo easy_install --upgrade google-api-python-client
   28  cd AIY-projects-python/
   29  source env/bin/activate
   30  cp ../assistant.json ../cloud_speech.json
   31  python3 checkpoints/check_cloud.py 
   32  pushd ../Desktop/
   33  cp VoiceHatAIY-19023530c142.json ../cloud_speech.json 
   34  popd
   35  python3 checkpoints/check_cloud.py 
   36  history >> Desktop/historyfinal.txt

drigz, 2018-01-25:
Excellent feedback, thanks Mark!

> 11 python3 checkpoints/check_audio.py <-- Errors here

This is a known issue and a fix is on the way. It should read `env/bin/python3 checkpoints/check_audio.py`.

> it appears that the file structure is moved around after import -- this shows that files originally in src/examples/voice/ are moved up two directories.

That's correct, but it's not the only change. The biggest difference is that in the `aiyprojects` branch, `check_audio.py` needs to be run inside the virtualenv, which is the cause of the `ImportError`s.

> pi@raspberrypi:~/AIY-projects-python/checkpoints $ python3 check_cloud.py
> Traceback (most recent call last):
> File "check_cloud.py", line 26, in 
> import aiy._apis._speech # noqa
> File "/home/pi/AIY-projects-python/src/aiy/_apis/_speech.py", line 25, in 
> import google.auth
> ImportError: No module named 'google'

The correct way to run `check_cloud.py` is to follow the [Change to the Cloud Speech API instructions](https://aiyprojects.withgoogle.com/voice/#makers-guide-3-1--change-to-the-cloud-speech-api) and then either use `env/bin/python3 checkpoints/check_cloud.py` or run the script after `source env/bin/activate`, as you discovered. Hopefully this will be clearer once HACKING.md is updated.

> instead of saving client secrets to assistant.json, save them to cloud_speech.json

I guess you followed the separate process (downloading service account credentials) here. ~/assistant.json (client secrets) and ~/cloud_speech.json (service account credentials) need to have different contents.

Let me know how your next attempt goes!

Brad94, 2018-01-27:
Hi,

My issue is exactly the same as this so didn't want to create a new one. Anything I can do to assist let me know. 

Currently I am running a RaspberryPi Zero Wireless with Google AIY hardware from May 2017 (Initial MayPi release). 

Previously was running an old build of unknown date which worked fine.

Today, after burning the aiyprojects-2018-01-03 which was successful.

Both assistant.json and cloud_speech.json have been copied to the system.

Check audio failed with the same error as @markjhughes ending in:

`card_id = CARDS_ID[get_aiy_device_name()]
KeyError: None`

None of the examples work either with assistant examples opening the browser to sign in to Google account but then after entering email address navigate back to Most Visited sites (not sure what it's meant to do after this). This results in needing to kill the terminal which brings the following error:

`Traceback (most recent call last):
  File "./assistant_grpc_demo.py", line 54, in <module>
    main()
  File "./assistant_grpc_demo.py", line 33, in main
    assistant = aiy.assistant.grpc.get_assistant()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 75, in get_assistant
    credentials = aiy.assistant.auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 127, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 118, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 71, in _credentials_flow_interactive
    credentials = flow.run_local_server()
  File "/home/pi/AIY-projects-python/env/lib/python3.5/site-packages/google_auth_oauthlib/flow.py", line 414, in run_local_server
    local_server.handle_request()
  File "/usr/lib/python3.5/socketserver.py", line 291, in handle_request
    ready = selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 376, in select
    fd_event_list = self._poll.poll(timeout)`

The cloud speech demo appears to execute, but after pressing the arcade button nothing happens.

After doing the 'git checkout voicekit' the check audio started working, but all examples continue to fail.

markjhughes, 2018-01-28:
Did you by any chance skip the 'source env/bin/activate' step?  That got me
a couple of times.

sheridat, 2018-01-28:
@Brad94  The _None of the examples work either with assistant examples opening the browser to sign in to Google account but then after entering email address navigate back to Most Visited sites (not sure what it's meant to do after this)_ resonates with me. 

I've burnt the image a few times now and each time I've had the experience you describe where you enter your email address and then instead of getting the password page - you get the most visited sites.

I just kept on trying and after a few goes the password page displayed - put in the password and then you get a page about access to your account on you have to click accept.  

So have a few more tries

drigz, 2018-01-29:
@Brad94 Sounds like two separate issues there. The first (Check Audio failing) is #255 - could you try the following workaround?

> Run the following commands to enable the audio driver. Check Audio will still fail, but the demos should work normally:

```
cd ~/AIY-projects-python
sudo scripts/configure-driver.sh
sudo dtoverlay googlevoicehat-soundcard
```

For the web browser issue, I've filed #269 - please comment there if you have any more info.

Regarding the messy error - with Python it's pretty much expected that if you have to kill the script while it's doing something, you'll get an error that looks like that.

Brad94, 2018-01-30:
@markjhughes _Did you by any chance skip the 'source env/bin/activate' step?  That got me
a couple of times.

If you use the dev terminal you shouldn't have to run that command which is what I've been doing.

@drigz That workaround was successful Check Audio worked straight away!

Glad to hear that has been raised. 

Thanks.

closed 2019-11-14 by dmitriykovalev