google/aiyprojects-raspbian #262: GPIO warning

mj585, 2018-01-23:
I updated my google AIY voice kit by getting voice kit SD Image from https://aiyprojects.withgoogle.com/voice/#assembly-guide-1-get-the-voice-kit-sd-image.

I works well before updated but it doesn't work now. Belows are  Error messages a

==========================================================
(env) pi@raspberrypi:~/AIY-projects-python/src $ ./assistant_grpc_demo.py
/home/pi/AIY-projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
Press the button and speak
[2018-01-21 23:10:32,211] INFO:recorder:started recording
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM dsnoop
arecord: main:788: audio open error: No such file or directory
[2018-01-21 23:10:32,225] ERROR:recorder:Microphone recorder died unexpectedly, aborting..

drigz, 2018-01-24:
Do you have a file at `~/.asoundrc`? Sometimes the Raspberry Pi volume control creates it and it interferes with the AIY Voice Kit. If you have it, could you post its contents here? Then try deleting it and seeing if the demos work afterwards.

mj585, 2018-01-24:
I deleted it and problem solved. Thank you.

drigz, 2018-01-24:
Thanks for reporting back! I'm going to try to prevent this (with the approach from #172, if you're curious) so hopefully we can avoid it with the next AIY image.

closed 2018-01-24 by drigz

andrewtvuong, 2018-02-20:
I am experience the same problem as OP but do not have a ~/.asoundrc file

drigz, 2018-02-20:
@andrewtvuong Are you using the AIY image, and if so which version?

You can check use `diff /etc/asound.conf ~/AIY-projects-python/scripts/` to see if the config was correctly installed, and if so try using `strace -eopen aplay -q -d 1 -c 1 -t raw -f S32_LE /dev/zero` to see if it's loading a config file from somewhere else.

andrewtvuong, 2018-02-20:
Hi @drigz thanks for your help; the voice image was Jan 3rd, 2018. I managed to resolve the issue by running the driver scripts: `configure-driver.sh` and `install-alsa-config.sh` from `/home/pi/AIY-projects-python/scripts`. Not sure why I ran into this problem the second time after reflashing and following the same setup procedure.

drigz, 2018-02-20:
@andrewtvuong Thanks for following up! Maybe an apt update or something else overwrite the ALSA config?

If you or anyone else see this again, please file a new issue with the output of the commands in my [previous comment](https://github.com/google/aiyprojects-raspbian/issues/262#issuecomment-366884316).

trickiedick, 2018-02-21:
It's a me too on this - I burned a new sd from the Jan 3rd 2018 image.  Same error reports - and the two driver scripts worked fine for me.