google/aiyprojects-raspbian #267: aiyprojects fails to install on C-STEM raspbian

jjfretired, 2018-01-27:
Download and install C-STEM and clone aiyprojects from github.  Installed failed on scripts/install-deps.sh.  See attached for failure.
[stemfail.txt](https://github.com/google/aiyprojects-raspbian/files/1670633/stemfail.txt)

drigz, 2018-01-29:
The error is:

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python-pip-whl : Breaks: virtualenv (< 14.0.5+ds-5) but 1.11.6+ds-1 is to be installed
 python3-all-dev : Depends: python3 (= 3.4.2-2) but 3.5.3-1 is to be installed
                   Depends: libpython3-all-dev (= 3.4.2-2) but it is not going to be installed
                   Depends: python3-all (= 3.4.2-2) but it is not going to be installed
                   Depends: python3-dev (= 3.4.2-2) but 3.5.3-1 is to be installed
                   Depends: python3.4-dev but it is not going to be installed
 virtualenv : Depends: python3-virtualenv but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

@jjfretired, we don't support other distros than Raspbian - it's hard since package names/versions vary - but others have been able to get it to work (eg on Diet-Pi) by looking inside install-deps.sh and adjusting the commands.

If you figure out what commands you need to run, post them here so others can copy your method.

jjfretired, 2018-01-31:
I finally got it to work.
Changed 
< sudo apt-get -y install alsa-utils python3-all-dev python3-pip python3-numpy \
<   python3-rpi.gpio python3-pysocks virtualenv rsync libttspico-utils ntpdate
< sudo pip3 install --upgrade pip virtualenv
to
> sudo apt-get -y install alsa-utils python3-dev python3-pip python3-numpy \
>   python3-rpi.gpio python3-pysocks rsync libttspico-utils ntpdate
> sudo pip3 install virtualenv

Commented out the following to get check_audio.py  to execute properly.
if aiy_device_name() == 'Voice Hat":
   enable_audio_driver()
in check_audio.py

Execute some files in example successfully.
copied AIY-projects-shell from a raspbian load, rename to voice.sh and chmod 755 voice.sh.
Added bin under home/pi and and file to PATH - PATH=/home/pi/bin:$PATH
voice.sh rename to voice.txt to load.

[voice.txt](https://github.com/google/aiyprojects-raspbian/files/1683325/voice.txt.

closed 2018-01-31 by jjfretired