google/aiyprojects-raspbian #268: Issue with configure-driver.sh

jjfretired, 2018-01-28:
In scripts/configure-driver.sh there are three greps

grep -q "dtoverlay=i2s-mmap" /boot/config.txt || \
  echo "dtoverlay=i2s-mmap" >> /boot/config.txt
grep -q "dtoverlay=googlevoicehat-soundcard" /boot/config.txt || \
  echo "dtoverlay=googlevoicehat-soundcard" >> /boot/config.txt
grep -q "dtparam=i2s=on" /boot/config.txt || \
  echo "dtparam=i2s=on" >> /boot/config.txt
I may be wrong but the 3 greps returne true as the config.txt file has the search items commented out.
If they test returns true the || (or) statement is never executed and the lines are never uncommented.

sheridat, 2018-01-28:
Relates to the aiyprojects branch image. To get going I had to edit /boot/config.txt and uncomment 
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard

drigz, 2018-01-29:
Great catch - thank you! We've updated the workaround posted on #255 and we'll look into fixing configure_driver.sh.

closed 2018-02-27 by mbrooksx