google/aiyprojects-raspbian #269: Can't log in - browser redirects to Most Visited sites

drigz, 2018-01-29:
From https://github.com/google/aiyprojects-raspbian/issues/261#issuecomment-361024395:

> None of the examples work either with assistant examples opening the browser to sign in to Google account but then after entering email address navigate back to Most Visited sites (not sure what it's meant to do after this).

And from https://github.com/google/aiyprojects-raspbian/issues/261#issuecomment-361049641:

> I've burnt the image a few times now and each time I've had the experience you describe where you enter your email address and then instead of getting the password page - you get the most visited sites.
> 
> I just kept on trying and after a few goes the password page displayed - put in the password and then you get a page about access to your account on you have to click accept.
> 
> So have a few more tries

drigz, 2018-01-29:
We've noticed that on the 2018-01-03 image, a different browser (not Chrome) is being opened - maybe that's causing the login problem?

If anyone runs in to this, please try logging in with Chrome (you can open it from the taskbar/"start menu"). You can copy the link from the terminal or from the URL bar of the browser that is automatically opened, paste it into the Chrome URL bar, and then try to log in there. Let us know if that works better.

LondonSi72, 2018-01-29:
The epiphany browser opens by default on that image. It caused my install to hang and corrupted the voice_recognizer cache file in /home/pi whilst I was doing the authorisation step. I had to delete the cache file and manually enter the URL into Chrome...

drigz, 2018-01-29:
Thanks for the info, @LondonSi72! We'll make the AIY scripts open chromium by default. Users can change the default browser then if they want to.

The change has been made internally, and it should be released in a week or so.

cougar-b, 2018-02-26:
@drigz 
We had the same problem at a hands-on to make a Voice Kit.
At the time around 5 of 16 people had the problem.
(I don't know why only some had got it with the same image and the same process)
Anyway I confirmed that the way changing the default browser was working well for all. :+1: 

Thank you for the info.

closed 2018-02-27 by mbrooksx