google/aiyprojects-raspbian #270: Traceback (most recent call last):

seanandre, 2018-01-30:
I'm getting a weird message when I run the Check Audio on my desktop. The console shows this

Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 158, in <module>
    main()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 153, in main
    do_checks()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 111, in do_checks
    if not check_voicehat_present():
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 70, in check_voicehat_present
    card_id = CARDS_ID[get_aiy_device_name()]
KeyError: None
Press Enter to close...

Possible bad Voice Hat? I'm running a new Pi 3 B with a fresh image flash of 2018-01-03. I even tried cloning a copy of the repo and still get the same message. Everything is also wired perfectly. Check Cloud and Check Wifi work, just not Check audio. I also get some kinda system kernel modules not loaded at startup. Not sure if this is related, but I did do sudo apt-get update, dpkg --configure -a, and dist-upgrade and still doesn't seem to fix it. Any ideas?

seanandre, 2018-01-30:
I did a "git checkout voicekit" and now I get a different message. 

Failed to find the voiceHAT soundcard. Refer to HACKING.md for how to
setup the voiceHAT driver: https://git.io/v99yK
Press Enter to close...

drigz, 2018-01-30:
This looks like a duplicate of #255. Could you try this workaround and let us know if it works for you?

Run the following commands to enable the audio driver. Check Audio will still fail, but the demos should work normally:

```
git checkout aiyprojects
sudo dtoverlay googlevoicehat-soundcard
sudo leafpad /boot/config.txt
```

In the text editor, find the lines:

```
# dtoverlay=i2s-mmap
# dtoverlay=googlevoicehat-soundcard
```

and uncomment them (remove the `#`s) to give:

```
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard
```

seanandre, 2018-01-30:
Okay great, that seemed to do the trick. The "Check Audio" file on desktop doesn't work but when you run a demo script it still runs. To new users who bought the kit is using the MagPi instructions, ignore the part in Chapter 6 where it says to run "src/cloudspeech_demo.py" since it no longer exist in that folder. Go to src/examples/voice and then run python cloudspeech_demo.py and have at it. Thanks for your help drigz!

seanandre, 2018-01-30:
I'll keep this open if anyone has any ideas what the fail to load system kernels is at boot. I also have some bad lag, but I'll give the benefit of the doubt it's my wifi location in this house lol.

drigz, 2018-01-30:
"fail to load system kernels" is a known issue (just filed #272) - you can safely ignore it.

Thanks for the feedback about the workaround! Enjoy the kit :)

closed 2018-01-30 by drigz