google/aiyprojects-raspbian #271: AIY Vision Project - AssertionError

tranqkhue, 2018-01-30:
I've used the Vision kit with the built-in Joy detection example. Now I'm using the object_detection.py example and I get this error
`(env) pi@raspberrypi:~ $ python3 ~/AIY-projects-python/src/examples/vision/object_detection.py --input=Desktop/test.jpg --output=Desktop/output.jpg
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/object_detection.py", line 56, in <module>
    main()
  File "/home/pi/AIY-projects-python/src/examples/vision/object_detection.py", line 47, in main
    result = inference.run(image_center)
  File "/home/pi/AIY-projects-python/src/aiy/vision/inference.py", line 70, in run
    return self._engine.image_inference(self._key, image, params)
  File "/home/pi/AIY-projects-python/src/aiy/vision/inference.py", line 279, in image_inference
    assert False, 'Only RGB and L modes are supported.'
AssertionError: Only RGB and L modes are supported.
`

drigz, 2018-01-30:
I guess the image is in not in a supported format. Can you post the output of `identify -verbose Desktop/test.jpg`?

You could try `convert Desktop/test.jpg -colorspace RGB Desktop/test_rgb.jpg` but conversion doesn't seem to be easy in general ([source](http://www.imagemagick.org/discourse-server/viewtopic.php?t=16464)).

If you don't have these commands, try `sudo apt-get install imagemagick`.

tranqkhue, 2018-01-30:
I tried converting it with `convert Desktop/test.jpg -colorspace RGB Desktop/test_rgb.jpg`. And It works :D . By the way, I think it should be marked as a bug because I just use an ordinary picture from my Ubuntu laptop. 
For more details, this is the identification of my original test.jpg image
```
identify -verbose Desktop/test.jpg
Image: Desktop/test.jpg
Format: PNG (Portable Network Graphics)
Mime type: image/png
Class: DirectClass
Geometry: 1920x1080+0+0
Units: Undefined
Type: TrueColorAlpha
Endianess: Undefined
Colorspace: sRGB
Depth: 8-bit
Channel depth:
red: 8-bit
green: 8-bit
blue: 8-bit
alpha: 1-bit

```

And test_rgb.jpg `convert Desktop/test.jpg -colorspace RGB Desktop/test_rgb.jpg`. Seems like the alpha channel was removed:
```
 identify -verbose Desktop/test_rgb.jpgImage: Desktop/test_rgb.jpg
  Format: JPEG (Joint Photographic Experts Group JFIF format)
  Mime type: image/jpeg
  Class: DirectClass
  Geometry: 1920x1080+0+0
  Units: Undefined
  Type: TrueColor
  Endianess: Undefined
  Colorspace: sRGB
  Depth: 8-bit
  Channel depth:
    red: 8-bit
    green: 8-bit
    blue: 8-bit

```

drigz, 2018-01-30:
@tranqkhue Thanks for the update!

@dmitriykovalev Would it be easy to ignore an alpha channel if present?

tranqkhue, 2018-01-30:
I've just asked my friend about alpha channel. It's used in PNG format for transparency. Normally, in graphics applications, they ignore it (I can open it in my image viewer). So I think it should be okay to throw it away :D

dmitriykovalev, 2018-01-31:
@tranqkhue,

Regarding 'Only RGB and L modes are supported', I'll try to add support for images with alpha channel, thanks for reporting this!

Regarding 'generator' object has no attribute 'tensors', your code is slightly incorrect: inference.run() returns a generator object, so you basically need to iterate over it in order to get inference results for camera frames:

```
with CameraInference(image_classification.model()) as inference:
  for result in inference.run():
    classes = image_classification.get_classes(result)
    for i, (label, score) in enumerate(classes):
      print('Result %d: %s (prob=%f)' % (i, label, score))
```

tranqkhue, 2018-01-31:
Another issue:
I'm using the arcade button. When I use it to blink the light OR to press the button, it works fine. But when I use those functions at the same time, the arcade button doesn't light, but the button pressing is still okay
```
from aiy._drivers._rgbled import PrivacyLED
from aiy._drivers._rgbled import RGBLED
from aiy.vision.pins import BUTTON_GPIO_PIN
from gpiozero import Button

from picamera import PiCamera
import time

def main():
    rgbled = RGBLED()
    rgbled.SetAnimation(rgbled.BLUE, rgbled.BREATHE, 0.4)
    time.sleep(5)
    button = Button(BUTTON_GPIO_PIN)
```

dmitriykovalev, 2018-04-26:
We have new API to control RGB LED:
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/leds.py

and corresponding examples: 
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/leds_example.py

closed 2018-04-26 by dmitriykovalev