google/aiyprojects-raspbian #273: Continuous Listening Available?

idealist33, 2018-01-30:
From the python example, a person should trigger AIY kit by saying 'Okay google' or press a button whenever saying commands.
I want to let AIY keep listening. First a person say 'change' and AIY executes a command. After that without triggering AIY again, I want to let AIY waits for person's speech.

seanandre, 2018-01-30:
I believe you're talking about the assistant_library_demo.py
You can say "Okay Google" for every question just as a normal Google Home without pressing the button. The problem I'm having now is my speaker, it seems to be crackling when she speaks :(

divx118, 2018-01-30:
@idealist33 You can restart the listening after it has finished with `self._assistant.start_conversation()`

seanandre, 2018-01-31:
I seemed to fix the crackling in the speaker by overclocking the RPI in boot.

idealist33, 2018-02-01:
Thanks a lot I used assistant.start_conversation().

closed 2018-02-01 by idealist33