google/aiyprojects-raspbian #275: is it possible to use google assistant without internet?

idealist33, 2018-01-31:
I need to use speech recognition to communicate to server by tcp.
But the network is internal(not connected to internet) and blocked to many services(google, gmail, facebook...) because of security.
Can I use google assistant without connecting to google assistant service?

divx118, 2018-01-31:
@idealist33 
The actual speech recognition is done on google servers. So you will need internet. There are however other rpi projects with speech recognition that can work local. Which seems to be your only option. https://www.raspberrypi.org/forums/ should be a better place to ask these kind of questions.

idealist33, 2018-02-01:
Thanks a lot!

drigz, 2018-02-01:
@divx118 Good answer, thanks!

closed 2018-02-01 by drigz