google/aiyprojects-raspbian #276: Alarms and timers stopped working?

LondonSi72, 2018-01-31:
Has something happened at the Google servers to stop alarms and timers working?  I have two voicekits - one running the May 2017 image and one running the Jan 2018 image and they have both stopped processing alarm and timer requests. Google responds saying alarm is set then it immediately says 'there was a glitch - please try again in a few moments'. The log on the Jan 2018 image shows 
ERROR:assistant_output_handler.cc(543) Ignoring kUnimplemented action error!
ERROR:assistant_output_handler.cc(618) Failed with a gerneral speech recognition error

I'm running the Assistant_library_with_local_commands_demo.py sample

I thought I'd seen someone else with this error quite a while back but can't find it in the issues in this repo, or a general Internet search.

LondonSi72, 2018-01-31:
OK - seems it's actually just the alarm that is no longer working - timers still work.  Is this a known problem?  Alarms still seem to work on the Google Mini I have.

drigz, 2018-02-01:
I can't reproduce this. What was the exact query you used?

sheridat, 2018-02-01:
@drigz - while you wait for the OP to reply - I am running the Jan 2018 aiyprojects image. 
I tried setting an alarm a couple of dozen times - in each instance I asked the assistant to "set an alarm for hh mm" The assistant replied with something along the lines of "your alarm is set for hh mm" and then said "hmm something went wrong - try again in a few seconds" interestingly sometimes she says there has been a glitch rather than saying Hmmm

The trackeback below shows the error for 3 attempts to set the allarm - all failed.
  GPIO.setup(channel, GPIO.OUT)
Say "OK, Google" then speak, or press Ctrl+C to quit....
You said: set an alarm for 13 23
[1289:1301:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
[1289:1301:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error
You said: set an alarm for 13 23
[1289:1301:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
[1289:1301:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error
You said: set an alarm for 13 24
[1289:1301:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
[1289:1301:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error

LondonSi72, 2018-02-01:
Thanks @sheridat.  That's pretty much what I'm seeing too.

@drigz Here's the output when I ask to set an alarm in 5 minutes.  Google replies with the time that would equate to 5 minutes later, and says it's setting an alarm, then immediately either says there's been a glitch or says something went wrong, try again in a few seconds.  Similar thing happens if I ask it to set an alarm for quarter past 6:

Build info: Thu Jan  4 03:38:36 UTC 2018 @ bec0157
Dev terminal is ready! See the demos in 'src/examples' to get started.
(env) pi@BlueGooglePi:~/AIY-projects-python $ src/examples/voice/assistant_library_with_local_commands_custom.py
Say "OK, Google" then speak, or press Ctrl+C to quit...
You said: set an alarm for 5 minutes
[1116:1129:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
[1116:1129:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error
You said: set an alarm for quarter past 6
[1116:1129:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
[1116:1129:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error

sheridat, 2018-02-01:
Hi @LondonSi72 
I tried mine doing it your way of asking to set an alarm for 5 minutes - i love the way it comes back and tells you the time it will be set for. It then failed in exactly the same manner as yours

LondonSi72, 2018-02-01:
And just for completeness, here is output from journalctl on my older GoogleAIY with the image from May last year when I ask for the same alarms (it mis-hears my request for quarter past 6):

Feb 01 16:24:03 GooglePi python3[20138]: [2018-02-01 16:24:03,583] INFO:root:ON_CONVERSATION_TURN_STARTED
Feb 01 16:24:03 GooglePi python3[20138]: [2018-02-01 16:24:03,587] INFO:main:listening...
Feb 01 16:24:08 GooglePi python3[20138]: [2018-02-01 16:24:08,372] INFO:root:ON_END_OF_UTTERANCE
Feb 01 16:24:08 GooglePi python3[20138]: [2018-02-01 16:24:08,375] INFO:main:thinking...
Feb 01 16:24:08 GooglePi python3[20138]: [2018-02-01 16:24:08,454] INFO:root:ON_RECOGNIZING_SPEECH_FINISHED:
Feb 01 16:24:08 GooglePi python3[20138]: {'text': 'set an alarm for 5 minutes'}
Feb 01 16:24:09 GooglePi python3[20138]: [2018-02-01 16:24:09,151] INFO:root:ON_RESPONDING_STARTED:
Feb 01 16:24:09 GooglePi python3[20138]: {'is_error_response': False}
Feb 01 16:24:09 GooglePi kernel: voicehat-codec voicehat-codec: Enabling audio amp...
Feb 01 16:24:12 GooglePi python3[20138]: [20138:22533:ERROR:assistant_output_handler.cc(517)] Ignoring kUnimplemented action error!
Feb 01 16:24:12 GooglePi python3[20138]: [20138:22533:ERROR:assistant_output_handler.cc(589)] Failed with a general speech recognition error
Feb 01 16:24:12 GooglePi python3[20138]: [2018-02-01 16:24:12,254] INFO:root:ON_RESPONDING_FINISHED
Feb 01 16:24:12 GooglePi kernel: voicehat-codec voicehat-codec: Disabling audio amp...
Feb 01 16:24:12 GooglePi python3[20138]: [2018-02-01 16:24:12,255] INFO:root:ON_RESPONDING_STARTED:
Feb 01 16:24:12 GooglePi python3[20138]: {'is_error_response': True}
Feb 01 16:24:12 GooglePi kernel: voicehat-codec voicehat-codec: Enabling audio amp...
Feb 01 16:24:16 GooglePi python3[20138]: [2018-02-01 16:24:16,022] INFO:root:ON_RESPONDING_FINISHED
Feb 01 16:24:16 GooglePi python3[20138]: [2018-02-01 16:24:16,024] INFO:root:ON_CONVERSATION_TURN_FINISHED:
Feb 01 16:24:16 GooglePi python3[20138]: {'with_follow_on_turn': False}
Feb 01 16:24:16 GooglePi kernel: voicehat-codec voicehat-codec: Disabling audio amp...
Feb 01 16:24:16 GooglePi python3[20138]: [2018-02-01 16:24:16,026] INFO:main:ready...
Feb 01 16:24:18 GooglePi python3[20138]: [2018-02-01 16:24:18,885] INFO:root:ON_CONVERSATION_TURN_STARTED
Feb 01 16:24:18 GooglePi python3[20138]: [2018-02-01 16:24:18,887] INFO:main:listening...
Feb 01 16:24:21 GooglePi python3[20138]: [2018-02-01 16:24:21,014] INFO:root:ON_END_OF_UTTERANCE
Feb 01 16:24:21 GooglePi python3[20138]: [2018-02-01 16:24:21,016] INFO:main:thinking...
Feb 01 16:24:21 GooglePi python3[20138]: [2018-02-01 16:24:21,186] INFO:root:ON_RECOGNIZING_SPEECH_FINISHED:
Feb 01 16:24:21 GooglePi python3[20138]: {'text': 'set an alarm for 4 to 6'}
Feb 01 16:24:21 GooglePi python3[20138]: [2018-02-01 16:24:21,754] INFO:root:ON_RESPONDING_STARTED:
Feb 01 16:24:21 GooglePi python3[20138]: {'is_error_response': False}
Feb 01 16:24:21 GooglePi kernel: voicehat-codec voicehat-codec: Enabling audio amp...
Feb 01 16:24:24 GooglePi python3[20138]: [20138:22533:ERROR:assistant_output_handler.cc(517)] Ignoring kUnimplemented action error!
Feb 01 16:24:24 GooglePi python3[20138]: [20138:22533:ERROR:assistant_output_handler.cc(589)] Failed with a general speech recognition error
Feb 01 16:24:24 GooglePi python3[20138]: [2018-02-01 16:24:24,942] INFO:root:ON_RESPONDING_FINISHED
Feb 01 16:24:24 GooglePi python3[20138]: [2018-02-01 16:24:24,945] INFO:root:ON_RESPONDING_STARTED:
Feb 01 16:24:24 GooglePi python3[20138]: {'is_error_response': True}
Feb 01 16:24:24 GooglePi kernel: voicehat-codec voicehat-codec: Disabling audio amp...
Feb 01 16:24:24 GooglePi kernel: voicehat-codec voicehat-codec: Enabling audio amp...
Feb 01 16:24:27 GooglePi python3[20138]: [2018-02-01 16:24:27,875] INFO:root:ON_RESPONDING_FINISHED
Feb 01 16:24:27 GooglePi python3[20138]: [2018-02-01 16:24:27,877] INFO:root:ON_CONVERSATION_TURN_FINISHED:
Feb 01 16:24:27 GooglePi python3[20138]: {'with_follow_on_turn': False}
Feb 01 16:24:27 GooglePi kernel: voicehat-codec voicehat-codec: Disabling audio amp...
Feb 01 16:24:27 GooglePi python3[20138]: [2018-02-01 16:24:27,878] INFO:main:ready...

drigz, 2018-02-01:
@proppy, have you seen this? I just tried "set an alarm for 5 minutes" and it worked fine (although the alert was very quiet...)

LondonSi72, 2018-02-01:
@drigz - I don't know if it makes a difference to the back-end servers that are used, but I'm in the UK.

sheridat, 2018-02-01:
@drigz - I'm in the UK as well

proppy, 2018-02-02:
I haven't seen this, maybe we could track this on https://github.com/googlesamples/assistant-sdk-python as it's unlikely to be a AIY specific issue.

LondonSi72, 2018-02-02:
@proppy I'm not sure what you mean by reproducing with google-assistant-demo. Do you mean one of the other samples in the /src/examples/voice, or do you mean the separate google-assistant SDK?  If it's the latter, Inwoukdnt know where to start...

sheridat, 2018-02-02:
@proppy I wouldn't know where to begin either.
I did try the AIY's GRPC demo. The behaviour is a little bit different - if you say set an alarm for 5 minutes it says "done and calculates the date and time that will be and tells you" and then says " i dont undertand" or  "I can't help with that yet"

sheridat, 2018-02-02:
@drigz - should @LondonSi72 raise this as an issue on https://github.com/googlesamples/assistant-sdk-python ?

LondonSi72, 2018-02-02:
I will raise it in the other forum when I get back home later today. I did suspect it wasn't an AIY specific problem, but it's interesting it doesn't seem to affect everyone. 
I have one of the voicekits in the bedroom controlling 433MHz lights etc and my wife liked the fact she could also set an alarm with it without having to play around with the alarm clock. 
I have to keep the better half happy :)

drigz, 2018-02-02:
@proppy means following these instructions: https://pypi.python.org/pypi/google-assistant-library

That's a demo of the Assistant Library that doesn't use the AIY code.

LondonSi72, 2018-02-19:
It seems the problem occurs with google-assistant-demo too, as another poster on that forum has confirmed they see the same issue.  It’s not universal, though, so makes me think it is a back-end server problem, and possible geographic.

closed 2018-06-25 by dmitriykovalev