google/aiyprojects-raspbian #278: ImportError: No Module named 'aiy'

soulproprietor, 2018-02-01:
I have read the other closed issue that states this happens if you don't follow the instructions in the HACKING file inside the "start dev terminal".  however, I am having this issue even in the 'start dev' environment. 

I've spent so many hours on this and I would love some help!!

1) I tried with the latest image (1-03-18) and I couldnt get the google assistant to work, I don't remmber what the error was, but I eventually read that downgrading to the previous image would work.  I did that and then the google samples worked flawlesslsy, so now I am using the older image.

2) I get the Import error for modlue 'aiy' after follwing the instructions many times.  I have tried it in the virtual environemt, and without the environment.  same error.  

Can someone please provide some insight?

soulproprietor, 2018-02-02:
I downloaded the voicekit branch and now it seems to work great :)

closed 2018-02-02 by soulproprietor

wilksy, 2018-06-28:
this worked for me.. from above.

If trying to install on an existing pi image or a current Voice image and using HACKING instructions...

just do this...

be in AIY_PROJECTs dir already

cp -R src/aiy checkpoints/aiy
python3 checkpoints/check_audio.py