google/aiyprojects-raspbian #280: hacking guide: Error trying to run /boot/check_audio.py

akramh, 2018-02-04:
I am getting PermissionError: [Errno 13] Permission denied: '/boot/config.txt' -> '/boot/config.txt.bak' 

I already have virtualenv enabled this file seems to be inaccessible. Trying to comment that code generates even another error any one know what is going on why it is not working

PaulFinch, 2018-02-04:
I had to use sudo with that script.

akramh, 2018-02-05:
u did and it didn't work. It worked when i use python from virtualenv like this - sudo env/bin/python

drigz, 2018-02-05:
Thanks for the report, @akramh! This is a known issue and a fix is on the way. In the short term, the workaround is to use:

`sudo env/bin/python checkpoints/check_audio.py`

as you discovered.

hack-r, 2018-04-29:
Per the instructions you would've had the (env) virtual env loaded via the settings of the "dev terminal" configured in the AIY image, right? So that part is implicit I guess, that if you're using another terminal you need to load the env youself. So that being said its just a matter of sudo as @PaulFinch commented

dmitriykovalev, 2018-08-16:
We are not using `virtualenv` anymore, everything should be working just by running `python3` directly.

closed 2018-08-16 by dmitriykovalev