google/aiyprojects-raspbian #281: Voice HAT Drive/Servo Pin Outs

jjfretired, 2018-02-05:
I been trying for a number of days to get a led to blink from the VoiceHAT pins.  I found a pin out at
https://pinout.xyz/pinout/voice.hat that show the pin outs and corresponding driver/servo pins.  

The picture in the manual shows them described as GPIO ports but  when I did a spread sheet file
to the BCM pins, WiringPi pins and board pin the GPIO pin should be list as BCM pin numbering.

As soon as I used the correct GPIO numbers the led began flashing.

[BCM-GPIO-pins.pdf](https://github.com/google/aiyprojects-raspbian/files/1697028/BCM-GPIO-pins.pdf)

drigz, 2018-02-06:
Thanks for the feedback. If I understand you right, the numbers are correct, but it's not clear that the manual uses BCM numbering. @jjfretired, how can we improve this?

We can't change the printed manuals, but we can update the website.

jjfretired, 2018-02-07:
After more googling and search, the manual is correct for BCM gpio pin addressing.  WiringPi has a different pin addressing.

closed 2018-02-07 by jjfretired

LondonSi72, 2018-02-07:
@jjfretired - yes - pin numbering is far from straightforward on the Pis. It took me a while to figure out the wiring Pi system - it's historic (from the original 20pin Pi) and the developer has a page on his site explaining why he hasn't  changed it.

It would be too easy if it were straightforward :)