google/aiyprojects-raspbian #282: Troubles with cloudspeech onAIYproject

eavitan, 2018-02-07:
hello, I'm having a problem activating cloudspeech_demo.py on the aiy-project kit.

`^(env) pi@raspberrypi:~/AIY-projects-python/src/ea $ python cloudspeech_demo.py`

`Traceback (most recent call last):
  File "cloudspeech_demo.py", line 19, in <module>
    import aiy.cloudspeech
  File "/home/pi/AIY-projects-python/src/aiy/cloudspeech.py", line 19, in <module>
    import aiy._apis._speech
  File "/home/pi/AIY-projects-python/src/aiy/_apis/_speech.py", line 27, in <module>
    import google.auth.transport.grpc
  File "/home/pi/AIY-projects-python/env/lib/python3.5/site-packages/google/auth/transport/grpc.py", line 28, in <module>
    class AuthMetadataPlugin(grpc.AuthMetadataPlugin):
AttributeError: module 'grpc' has no attribute 'AuthMetadataPlugin'`

What am I doing wrong?

when running the "Check Cloud" file from desktop it says that everything is working fine, i configured it propely.

drigz, 2018-02-08:
Could you post the output of `env/bin/pip freeze | grep  grpc`?

eavitan, 2018-02-08:
returns
google-assistant-grpc==0.0.2
grpcio==1.6.6

drigz, 2018-02-08:
Thanks. I've not seen the problem before... Seems similar to https://github.com/googlesamples/assistant-sdk-python/issues/83, but that's on a Mac.

eavitan, 2018-02-08:
just to make sure, for some reason somtimes it works, i have no idea why

drigz, 2018-02-08:
Curiouser and curiouser!

eavitan, 2018-02-09:
ill just post the all respond in case someone will know

(env) pi@raspberrypi:~/AIY-projects-python $ python src/cloudspeech.py
Traceback (most recent call last):
  File "src/cloudspeech.py", line 19, in <module>
    import aiy.cloudspeech
  File "/home/pi/AIY-projects-python/src/aiy/cloudspeech.py", line 19, in <module>
    import aiy._apis._speech
  File "/home/pi/AIY-projects-python/src/aiy/_apis/_speech.py", line 27, in <module>
    import google.auth.transport.grpc
  File "/home/pi/AIY-projects-python/env/lib/python3.5/site-packages/google/auth/transport/grpc.py", line 28, in <module>
    class AuthMetadataPlugin(grpc.AuthMetadataPlugin):
AttributeError: module 'grpc' has no attribute 'AuthMetadataPlugin'

HELPPPPPP

drigz, 2018-02-09:
You could try editing editing `/home/pi/AIY-projects-python/env/lib/python3.5/site-packages/google/auth/transport/grpc.py` and adding this line before line 28:

`print('grpc comes from', grpc.__file)`

Then try starting it a few times in different ways, and seeing if the output is different when it works and work it doesn't. That might give us an indication of what's causing the problem.

eavitan, 2018-02-18:
@drigz 
 File "/home/pi/AIY-projects-python/env/lib/python3.5/site-packages/google/auth/transport/grpc.py", line 27, in <module>
    print('grpc comes from', grpc.__file)
AttributeError: module 'grpc' has no attribute '__file'

eavitan, 2018-02-18:
@drigz i think i came up with the beginning of the solution, tell me if it helps.

sometimes, when i want to run a one of the scripts, i dont need to write python before, and sometime i do need.

means sometime i just write 
(env) pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ cloudspeech.py
-- and it works.

but sometime it answer: -bash: cloudspeech.py: command not found.

when i dont use python before the cloudspeech work just fine. im not really good at python in rpi but this is something i noticed, hope it helps. :)

thanks for your help

eavitan, 2018-02-18:
new highlights!!!
we gonna do it!!!!!

this works: 
(env) pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ python cloudspeech.py

this doest work:
(env) pi@raspberrypi:~/AIY-projects-python/src $ python cloudspeech.py

** same file!!! **

drigz, 2018-02-19:
Great investigation, @eavitan!

Regarding the debugging command, it should have been `__file__` instead of `__file` - sorry.

Are you still getting the `AttributeError: module 'grpc' has no attribute 'AuthMetadataPlugin'` error, or is the `cloudspeech.py: command not found.` error an unrelated issue?

You should be able to avoid `command not found` by always using the full path to the script, eg:

```
(env) pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ python cloudspeech_demo.py
(env) pi@raspberrypi:~/AIY-projects-python/src $ python examples/voice/cloudspeech_demo.py
(env) pi@raspberrypi:~/AIY-projects-python $ python src/examples/voice/cloudspeech_demo.py
```

Perhaps you already know this, but if you press <Tab> when typing a path, it will autocomplete it based on the current directory, which can help you check you're typing the right path. For example:

```
cd ~/A<Tab>
python sr<Tab>e<Tab>vo<Tab>c<Tab>
```

becomes

```
cd ~/AIY-projects-python/
python src/examples/voice/cloudspeech_demo.py
```

hack-r, 2018-04-30:
I have this problem as well, but I don't want to run it from the other location (there's a reason, which is that I'm doing a larger project in my current directory). Any idea about the underlying problem and solution, so that I can fix it within my folder?

I'm guessing there's something in the folder you used that I need. I'll be looking.

closed 2018-08-16 by dmitriykovalev