google/aiyprojects-raspbian #283: Voice kit 'night mode' for LED

mpember, 2018-02-10:
Is there any way of setting a lower LED peak brightness during 'night' hours?

Before I hack away at **src/aiy/_drivers/_led.py**,  I was wondering if there was an existing option.  If not, is **src/aiy/_drivers/_led.py** the best place to implement such a feature?

I was thinking that it should be possible to replace all hard-coded peak brightnesses of 100 to a variable which can be changed to reflect the 'mode'.  Am I wrong?

drigz, 2018-02-12:
It might be easier to just change the line:

`self.pwm.ChangeDutyCycle(next(self.iterator))`

to multiply by the max brightness / 100. Then the main script can call set_max_brightness(50) at night.

mpember, 2018-02-19:
Does such a function already exist?  If not, is there a preferred placement for such a function?  Would the voicehat object be the best place to attach these variables / functions?

mpember, 2018-02-19:
My custom service script already supports the use of a settings file to store various details (podcast URLs, RF codes for remote devices, credentials for other web services).  I can easily store the settings there, but I would require a way of passing these variables to the underlying objects in a way that would not break any coding bible that this project adheres to.

drigz, 2018-02-19:
I would add a new `set_max_brightness()` method to the `LED` class, which can optionally be used to adapt the brightness. The user's `main.py` can then set the max brightness however it wants.

mpember, 2018-03-02:
After taking the advice from @drigz, I tried the following:

I added this to the __init__ function of the led driver:
```
self.brightness = 1
```

I then added the following function:
```
    def set_brightness(self, brightness):
        try:
            self.brightness = int(brightness) / 100
        except ValueError:
            raise ValueError('unsupported brightness: %s' % brightness)
```

The last change was to modify every call to **self.pwm.ChangeDutyCycle** as follows:
```
self.pwm.ChangeDutyCycle(0 *** self.brightness**)
```
or
```
self.pwm.ChangeDutyCycle(100 *** self.brightness**)
```
or
```
self.pwm.ChangeDutyCycle(next(self.iterator) * self.brightness)
```

Now, I can set the brightness by simply calling
```
aiy.voicehat.get_led().set_brightness(10)
```

Is there anything I have overlooked?

Since I'm not very familiar with git, I don't know how to generate a pull request.

t1m0thyj, 2018-10-17:
I just submitted a PR to add this functionality. Also wanted to make a note that although 10% seems like a low value to use, it actually corresponds to around 38% of the LED's full brightness according to the equation for linear LED PWM found [here](https://jared.geek.nz/2013/feb/linear-led-pwm).

mpember, 2018-10-17:
Since originally submitting this code, I have amended it.  The latest version can be found here:
https://github.com/mpember/aiyprojects-raspbian/blob/aiyprojects/src/aiy/_drivers/_led.py

I have two AIY kits.  Both came with the green button.  Since I wanted to use a different colour on one of the kits, I replaced the button an arcade button I bought in a kit of 10.  The kit that contained the original button suffered from problem of spiking to full brightness at random points in the cycle.  I did not have this issue with the kit that included a replacement button.

Since switching to the new method of adjusting the brightness, the problem appears to have gone away.

I'm not sure if this new approach makes it possible to better correlate the brightness percentage with visual brightness.

t1m0thyj, 2018-10-18:
I recently bought an AIY kit and am using it with the original green button. The light is much brighter than the tiny LED I was using before which is why I wanted night mode.

I've been using the code in my PR for the past few days and haven't noticed any weird behavior with the light. In theory the two approaches should behave exactly the same, they're just multiplying by `self.brightness` at different places in the code.

I know that in practice things don't always work as theoretically expected, so there could be a problem with the current approach that I'm not aware of. However the correlation of the brightness percentage with visual brightness is definitely the same between the two approaches.

mpember, 2018-10-18:
I guess my second approach was a bit of overkill.  If others are not having the same problem that I experienced, that's great.  I've got a few more of those replacement buttons.  I should probably try another LED/button and see if it just happens to be a dodgy bit of hardware in that specific kit.  I guess the only real improvement is that my second approach kept the maximum brightness for the **ON** status.

While I have modified the value of **BEACON_DARK** to be **0** as a way of stopping the idle flashing, I wonder if a 'night mode' would also involve stopping the AIY kit from acting as an accidental night light.