google/aiyprojects-raspbian #288: How to re-authenticate Google Assistant API with a new client secret json from another cloud account?

andrewtvuong, 2018-02-20:
Thanks!

drigz, 2018-02-20:
If you delete the cached credentials: `rm ~/.cache/voice-recognizer/assistant_credentials.json`

Then you can overwrite `~/assistant.json` with the new client secrets and rerun your script to trigger the auth process again.

Let us know how it goes!

andrewtvuong, 2018-02-20:
@drigz sorry I had reflashed already. I believed I could have just deleted the old `~/assistant.json` and replacing it with a new credential file, but then receive "('deleted_client: The OAuth client was deleted.', '{\n "error" : "deleted_client",\n "error_description" : "The OAuth client was deleted."\n}')"

drigz, 2018-02-20:
Maybe the code should handle that error by saying to the user: "If you intentionally changed your client secrets, please delete ~/.cache/voice-recognizer/assistant_credentials.json" and/or provide a command to reauth with new client secrets.

closed 2018-06-25 by dmitriykovalev

Joel4763, 2020-04-29:
This worked for me perfectly! I also feel that running any of the code with the framework should alert back to remove the cached file, as I was stuck for a while till I found this GitHub issue!