google/aiyprojects-raspbian #289: bonnet_model_compiler.par Error

jeonghopark, 2018-02-20:
Hi,

I have a error of bonnet_model_compiler.par from
[https://aiyprojects.withgoogle.com/vision/#makers-guide-4-1-tensorflow-model-compiler](https://aiyprojects.withgoogle.com/vision/#makers-guide-4-1-tensorflow-model-compiler)

My vision kit and zero work fine. I want to build make a custom model.

`(env) pi@raspberrypiZero:~/Desktop/model $ ./bonnet_model_compiler.par --frozen_graph_path=./mobilenet_v1_160res_0.5_imagenet.pb --output_graph_path=./mobilenet_v1_160res_0.5_imagenet.binaryproto --input_tensor_name="input" --output_tensor_names="MobilenetV1/Predictions/Softmax" --input_tensor_size=160`

and then the error message. ( just a help option is working.)

> Traceback (most recent call last): File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
>     "__main__", mod_spec)
>   File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
>     exec(code, run_globals)
>   File "./bonnet_model_compiler.par/__main__.py", line 129, in <module>
>   File "./bonnet_model_compiler.par/__main__.py", line 71, in main
>   File "./bonnet_model_compiler.par/compiler/bonnet_model_compiler_utils.py", line 11, in run_cmd
>   File "/usr/lib/python3.5/subprocess.py", line 676, in __init__
>     restore_signals, start_new_session)
>   File "/usr/lib/python3.5/subprocess.py", line 1282, in _execute_child
>     raise child_exception_type(errno_num, err_msg)
> OSError: [Errno 8] Exec format error

thanks

PeterMalkin, 2018-02-20:
Jeon,

The compiler is intended to be run on your host linux desktop machine. not on raspberry pi zero. Please compile your model on your desktop, then transfer over to the raspberry pi to run.

jeonghopark, 2018-02-20:
@PeterMalkin thanks :-) it's working.

closed 2018-02-20 by jeonghopark