google/aiyprojects-raspbian #290: Raspberry pi crashes when playing youtube video

Timerush, 2018-02-25:
I was running the December version of the AIY kit, and could watch youtube videos when I plugged in a keyboard and mouse. I saw that this new update would allow me to add custom commands into the pi, so I got the new iso and set up my raspberry pi. now anytime I try to load a youtube video the entire pi freezes and I cannot do anything but unplug it to get it to stop.

drigz, 2018-02-28:
I haven't tried this yet, but I just noticed an entry in the [ChangeLog for Raspbian Stretch](http://downloads.raspberrypi.org/raspbian/release_notes.txt) 2017-09-07 (which is what the AIY images are currently based on):

`* Bug fix for crash on some videos and animations in Chromium`

So maybe this was fixed upstream but somehow undone when creating the AIY image?

@Timerush, do these instructions help? https://github.com/guysoft/FullPageOS/issues/86#issuecomment-312369762

schmide, 2018-03-10:
I just setup a pi for the first time and got this crash over and over.

did a "sudo apt-get update" and it went away.

The current Raspbian at https://www.raspberrypi.org/downloads/ seems to have this problem.

Please rebuild package with update

gyukiss55, 2021-08-24:
Disable Chrome extension, it help in my RPI4.
[image]