google/aiyprojects-raspbian #292: Raspberry Pi Crashes On Startup

jdquaff, 2018-03-03:
So, I'm brand new to the Pi scene and this is my first project, the Voice Kit. I formatted my SD Card, burned the image file onto it, but when I insert it and boot the Pi it runs the start up and the first kernel fails. The whole system starts to loop, jumping from my monitor receiving no signal, to an attempted boot (no code crawl, though) back to no signal.

Any ideas? 

I've reburned the image file multiple times, as well as downloaded a new file from the AIY website.

Thanks for any help!

jdquaff, 2018-03-03:
The image file is 2/21/2018, because that's important as well!

omalleyck20, 2018-03-03:
Yes! I've been having the same issue!

Other info that applies to my circumstance for the sake of troubleshooting (but may likely apply to @jdquaff ) are some of the following:
-Image Formatted on Windows 10 Default Format, then SD Association format program (issue persisted)
-Upon Pi boot, text on a blue screen says "Resized RootKit. Restarting." (and is not seen again during the looping boots after)

[This is a picture I snapped of the failed kernel text.](https://drive.google.com/file/d/1Foyxx8NSQEnJRymPLRFgIkRitkYcqjlO/view?usp=sharing) It only appears once, as @jdquaff said, before only the Raspian logos appear in the following boots.

Thank you anyone that responds!

jdquaff, 2018-03-03:
To take this a step further, I hav

jdquaff, 2018-03-03:
Sorry, fat fingers closed the issue. I tried the same things as @omalleyck20, and they are describing the same issue.

I formatted on Windows Default first as well, but also tried it on a brand new SD and got the same result - so it doesn't seem to be that.

dmitriykovalev, 2018-03-03:
@jdquaff,  @omalleyck20, can you please accurately describe your hardware setup? Are you using Pi 3 board with the attached VoiceHat?

jdquaff, 2018-03-03:
@dmitrykovalev

The Pi I am using is the Pi 3 Model B. And yes, with the attached VoiceHat. I am using a 16gb Sandisk microSD card. The microphone, speaker, and button are all wired securely in the proper locations.

LondonSi72, 2018-03-03:
What power supply are you using? If it doesn't have enough power the Pi3 can end up rebooting as peripherals are powered up / used. Also watch the wiring of the lamp/button. If you get this wrong it can cause the Pi to reboot too.

jdquaff, 2018-03-03:
I’ve triple checked the lamp wiring, so that isn’t the issue. For power, I’m using a Samsung power brick - which is making me think that’s it. (After putting it together last night I scrambled to find just about anything to plug in to).

How much power does the Pi3 require?

LondonSi72, 2018-03-03:
It varies depending on what it's doing and the peripherals used, but officially it needs 2.5A. I've found power supplies vary widely in how much they actually provide, despite what they claim. If you see a lightning bolt symbol appear in the top right hand side of the screen, that's a warning that the supply can't provide enough current. It might flicker on and off depending on what the Pi is doing. I've had a few supplies that claim they provide 2.5A but still cause the lightning bolt. Best to get at least one official Pi supply as they definitely work.

omalleyck20, 2018-03-03:
@dmitriykovalev 
That is correct. Specifically Model B.

@LondonSi72 
I have noticed the lightning bolt, and that's good to know about the power supply. That being said, I've successfully booted the default NOOBS and ran Raspian from the main site without a problem.

jdquaff, 2018-03-03:
Then i definitely am not getting enough power. Looking at the brick, it’s saying it outputs 5V/1A. Looks like I’ll be making a trip to the store, and I’ll report back when I try w/ more power!

jdquaff, 2018-03-03:
I've gotten it booted. Seems power was the issue. Now I'm facing other problems, but those are probably for another board unless anyone has experience with this kit. My demos aren't working

closed 2018-03-03 by PeterMalkin