google/aiyprojects-raspbian #294: cloudspeech_demo.py - Syntax error, line 381

AKLitman, 2018-03-04:
After successfully running the assistant_library_demo.py, setting up my service api, successfully running the Check Cloud script, and establishing a billing method, I tried running the cloudspeech_demo.py script and obtained the following error:

Traceback (most recent call last):
File "/usr/local/bin/cloudspeech_demo.py", line 6, in <module>
exec(compile(open(__file__).read(), __file__, 'exec'))
File "/opt/aiy/projects-python/src/examples/voice/cloudspeech_demo.py", line 20, in <module>
import aiy.cloudspeech
File "/opt/aiy/projects-python/src/aiy/cloudspeech.py", line 19, in <module>
import aiy._apis._speech
File "/opt/aiy/projects-python/src/aiy/_apis/_speech.py", line 30, in <module>
from google.cloud import speech
File "/usr/local/lib/python3.5/dist-packages/google/cloud/speech/__init__.py", line 22, in <module>
from google.cloud.speech.client import Client
File "/usr/local/lib/python3.5/dist-packages/google/cloud/speech/client.py", line 25, in <module>
from google.cloud.speech._gax import GAPICSpeechAPI
File "/usr/local/lib/python3.5/dist-packages/google/cloud/speech/_gax.py", line 17, in <module>
from google.cloud.gapic.speech.v1.speech_client import SpeechClient
File "/usr/local/lib/python3.5/dist-packages/google/cloud/gapic/speech/v1/speech_client.py", line 31, in <module>
from google.gapic.longrunning import operations_client
File "/usr/local/lib/python3.5/dist-packages/google/gapic/longrunning/operations_client.py", line 50, in <module>
from google.longrunning import operations_pb2
File "/usr/local/lib/python3.5/dist-packages/google/longrunning/operations_pb2.py", line 23, in <module>
from google.longrunning.operations_grpc_pb2 import *
File "/usr/local/lib/python3.5/dist-packages/google/longrunning/operations_grpc_pb2.py", line 341, in <module>
from grpc.beta import implementations as beta_implementations
File "/usr/local/lib/python3.5/dist-packages/grpc/beta/implementations.py", line 22, in <module>
from grpc import _auth
File "/usr/local/lib/python3.5/dist-packages/grpc/_auth.py", line 17, in <module>
from concurrent import futures
File "/usr/local/lib/python3.5/dist-packages/concurrent/futures/__init__.py", line 8, in <module>
from concurrent.futures._base import (FIRST_COMPLETED,
File "/usr/local/lib/python3.5/dist-packages/concurrent/futures/_base.py", line 381
raise exception_type, self._exception, self._traceback
^
SyntaxError: invalid syntax

I saw another user had posted an identical issue on the AIY Projects forum a few days ago but had but received any nibbles.  I am stuck right now.  The assistant_library_demo.py works over & over but this one (cloudspeech_demo.py) under a service account fails with the above error.  I searched for the line 381 in the py script mentioned and it is part of the distribution with my kit.  I ran Update & Upgrade which did have some new files in the kit but the error persists.  Any thoughts on what I may try next?  Austin

AKLitman, 2018-03-04:
I will move to close this topic.  The other user who had the same error as I had - wrote to tell me he saw a related error in other forums about the "futures" library existing for Python 2 compatibility.  He offered that the conversation concluded it was not needed in Python 3.5.  I renamed the entire futures library on my Pi and tried the same cloudspeech_demo.py and it worked without error.  Austin

closed 2018-03-04 by AKLitman

SUDHARMP, 2018-03-29:
Hi Austin , 
I'm facing the same exact issue, can you give tell me how you solved it in detail?

drigz, 2018-03-29:
I would guess they did something like:

`sudo mv /usr/local/lib/python3.5/dist-packages/concurrent/futures /usr/local/lib/python3.5/dist-packages/concurrent/futures.hidden`

Although it is not a very elegant solution... a better option would be to find out what has installed futures, since it should not be installed for Python 3.

AKLitman, 2018-03-30:
Dear Rodrigo - Yes, I used a different suffix for the name but yours is
very elegant and should work.  BTW, I recently updated/upgraded my OS and
found it overwrote all my AIY directories with new ones.  I should have
made copies of my new python code & scripts in a private directory but
recreating the changes should not be too hard.  I will know better next
time.

Thanks for the reply.  What you suggest is terrific.  Thanks - Austin