google/aiyprojects-raspbian #295: Can't install as a service on official image

bfenty, 2018-03-05:
I've got the voice assistant running when I run the demo py script, but I can't seem to get it installed as a service via systemd. I've tried running the installer script, following the instructions from hacking.md, and the service fails every time I try to start it.

Here's what the console says:

 sudo systemctl status voice-recognizer.service
● voice-recognizer.service - voice recognizer
   Loaded: loaded (/lib/systemd/system/voice-recognizer.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-03-04 20:44:01 CST; 4s ago
  Process: 2134 ExecStart=/opt/aiy/projects-python/env/bin/python3 -u src/main.py (code=exited, status=203/EXEC)
 Main PID: 2134 (code=exited, status=203/EXEC)

Mar 04 20:44:01 gamepi2 systemd[1]: voice-recognizer.service: Main process exited, code=exited, status=203/EXEC
Mar 04 20:44:01 gamepi2 systemd[1]: voice-recognizer.service: Unit entered failed state.
Mar 04 20:44:01 gamepi2 systemd[1]: voice-recognizer.service: Failed with result 'exit-code'.
Mar 04 20:44:01 gamepi2 systemd[1]: voice-recognizer.service: Service hold-off time over, scheduling restart.
Mar 04 20:44:01 gamepi2 systemd[1]: Stopped voice recognizer.
Mar 04 20:44:01 gamepi2 systemd[1]: voice-recognizer.service: Start request repeated too quickly.
Mar 04 20:44:01 gamepi2 systemd[1]: Failed to start voice recognizer.
Mar 04 20:44:01 gamepi2 systemd[1]: voice-recognizer.service: Unit entered failed state.
Mar 04 20:44:01 gamepi2 systemd[1]: voice-recognizer.service: Failed with result 'exit-code'.

divx118, 2018-03-10:
I noticed it too. It seems `VIRTUAL_ENV` is not used anymore so the [.service file](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/systemd/voice-recognizer.service) should be adjusted.
Something like below should do it.

```
# This service can be used to run your code automatically on startup. Look in
# HACKING.md for instructions on creating main.py and enabling it.

[Unit]
Description=voice recognizer
After=network.target ntpdate.service

[Service]
Environment=PATH=/home/pi/AIY-projects-python/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/usr/bin/python3 -u src/main.py
WorkingDirectory=/home/pi/AIY-projects-python
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```
If I am correct then [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) should also be updated by removing the `source env/bin/activate`

mcduarte2000, 2018-04-22:
Just tried the solution above, without success (as any others I found through the web)... Very frustrating. The script as many people say, ran well by itself. The copy main.py, actually, not (should we make it executable?)...

My service file right now, based on the comment above is like the one below. Can I help anyone test a different solution?

# This service can be used to run your code automatically on startup. Look in
# HACKING.md for instructions on creating main.py and enabling it.

[Unit]
Description=voice recognizer
After=network.target ntpdate.service

[Service]
Environment=VIRTUAL_ENV=/home/pi/AIY-projects-python/env
Environment=PATH=/home/pi/AIY-projects-python/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr$
ExecStart=/usr/bin/python3 -u src/main.py
WorkingDirectory=/home/pi/AIY-projects-python
StandardOutput=inherit
StandardError=inherit
Restart=always
User=root
RestartSec=50

[Install]
WantedBy=multi-user.target

divx118, 2018-04-23:
@mcduarte2000 It should be without the VIRTUAL_ENV so like below
```
# This service can be used to run your code automatically on startup. Look in
# HACKING.md for instructions on creating main.py and enabling it.

[Unit]
Description=voice recognizer
After=network.target ntpdate.service

[Service]
Environment=PATH=/home/pi/AIY-projects-python/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/usr/bin/python3 -u src/main.py
WorkingDirectory=/home/pi/AIY-projects-python
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

main.py doesn't need to be executable, because you start it with `/usr/bin/python3  -u src/main.py`
Is main.py in `/home/pi/AIY-projects-python/src`?
`/home/pi/AIY-projects-python/bin` isn't there anymore. so it could be removed from `Environment=PATH=` however it doesn't break anything if `/home/pi/AIY-projects-python/bin` is in the service file.
Can you manually start main.py with `/usr/bin/python3  src/main.py`?

mcduarte2000, 2018-04-24:
Many thanks. Copied/pasted your code, unfortunately still not working.

 Yes, /usr/bin/python3 /home/pi/AIY-projects-python/src/main.py works perfectly.

floresboy, 2018-04-25:
when i redirect output (/usr/bin/python3 /home/pi/AIY-voice-kit-python/src/examples/voice/main.py >> /tmp/out.txt) it doen't work either. When i leave the redirection out everything is fine.

Beats me but i don't get it.

divx118, 2018-04-25:
@mcduarte2000 Just to be sure, you did enable the service with `sudo systemctl enable voice-recognizer.service`
Can you give the output of `sudo systemctl --all | grep voice-recognizer.service`
and the output of `sudo journalctl -u voice-recognizer -n 14 --no-hostname`

divx118, 2018-04-26:
@mcduarte2000 there seems to be a problem with your main.py . Could you post the code?
Or you could add yourself some more debugging info to see where it fails.

mcduarte2000, 2018-05-01:
@divx118 

If you need more logs, please tell me what command I should run. I know how to do basic stuff, but not a linux pro. :) I am using the latest version of the script examples, which were updates 2 months ago. Just as a precaution I copied again the script to main.py and tried again to activate the service. The logs results changed (but still no success):

`May 01 10:58:40 systemd[1]: voice-recognizer.service: Failed with result 'exit-code'.
May 01 10:58:40 systemd[1]: voice-recognizer.service: Service hold-off time over, scheduling restart.
May 01 10:58:40 systemd[1]: Stopped voice recognizer.
May 01 10:58:40 systemd[1]: Started voice recognizer.
May 01 10:58:40 systemd[440]: voice-recognizer.service: Failed at step EXEC spawning /home/pi/AIY-projects-python/env/bin/python3: No s
May 01 10:58:40 systemd[1]: voice-recognizer.service: Main process exited, code=exited, status=203/EXEC
May 01 10:58:40 systemd[1]: voice-recognizer.service: Unit entered failed state.
May 01 10:58:40 systemd[1]: voice-recognizer.service: Failed with result 'exit-code'.
May 01 10:58:41 systemd[1]: voice-recognizer.service: Service hold-off time over, scheduling restart.
May 01 10:58:41 systemd[1]: Stopped voice recognizer.
May 01 10:58:41 systemd[1]: voice-recognizer.service: Start request repeated too quickly.
May 01 10:58:41 systemd[1]: Failed to start voice recognizer.
May 01 10:58:41 systemd[1]: voice-recognizer.service: Unit entered failed state.
May 01 10:58:41 systemd[1]: voice-recognizer.service: Failed with result 'exit-code'.`

divx118, 2018-05-03:
@mcduarte2000 Can you post the output of `cat /etc/systemd/system/multi-user.target.wants/voice-recognizer.service`
I believe you still are trying to run python in the environment because of 
```
voice-recognizer.service: Failed at step EXEC spawning /home/pi/AIY-projects-python/env/bin/python3: No such file or directory
-- Subject: Process /home/pi/AIY-projects-python/env/bin/python3 could not be executed
```

nasseifp, 2018-05-04:
@divx118

Thanks for posting here! I've followed what you said and was able to get my service running when checking the status of the service but now the voice kit won't respond. If I start the main.py manually it works as it should. 

I ended up checking sudo journal like you mentioned and found that it says I need client secrets for API to work. Add a assistant.json to root. I've already done this, API works manually why would it not be able to other wise? Any suggestions? Thank you again.

RowanZee, 2018-05-27:
@divx118 I am also experiencing the same issue. My service will not start due to looking the assistant.json file in the /root/ directory which obviously doesny exist. Should be looking in /home/pi/...

Have you solved this? Any other solutions?

divx118, 2018-05-27:
@RowanZee Can you post the output of `cat /etc/systemd/system/multi-user.target.wants/voice-recognizer.service`

RowanZee, 2018-05-28:
@divx118 
[Unit]
Description=voice recognizer
After=network.target ntpdate.service

[Service]
Environment=VIRTUAL_ENV=/home/pi/voice-recognizer-raspi/env
Environment=PATH=/home/pi/voice-recognizer-raspi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/voice-recognizer-raspi/env/bin/python3 -u src/main.py
WorkingDirectory=/home/pi/voice-recognizer-raspi
StandardOutput=inherit
StandardError=inherit
Restart=always
User=root

[Install]
WantedBy=multi-user.target

I've moved my package to an older one which still used the virtual env. I couldnt get the new one to work. Works perfectly when run manually inside the virtual environment

divx118, 2018-05-28:
@RowanZee It should be `User=pi` and not `User=root` in the service file.

RowanZee, 2018-05-28:
@divx118 I have tried that, makes no difference. Service also seems to start but when I run sudo journalctl -u voice-recognizer -n 14 --no-hostname shows the following

May 28 18:41:53 python3[14317]: You need client secrets to use the Assistant API.
May 28 18:41:53 python3[14317]: Follow these instructions:
May 28 18:41:53 python3[14317]:     https://developers.google.com/api-client-library/python/auth/installed-ap
May 28 18:41:53 python3[14317]: and put the file at /root/assistant.json
May 28 18:41:53 systemd[1]: voice-recognizer.service: Main process exited, code=exited, status=1/FAILURE
May 28 18:41:53 systemd[1]: voice-recognizer.service: Unit entered failed state.
May 28 18:41:53 systemd[1]: voice-recognizer.service: Failed with result 'exit-code'.

divx118, 2018-05-28:
@RowanZee Just to be absolutely sure, you have in `/etc/systemd/system/multi-user.target.wants/voice-recognizer.service` the below code and you get the above error you posted when you restart the servive or after a reboot?

```
# This service can be used to run your code automatically on startup. Look in
# HACKING.md for instructions on creating main.py and enabling it.

[Unit]
Description=voice recognizer
After=network.target ntpdate.service

[Service]
Environment=PATH=/home/pi/AIY-projects-python/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/usr/bin/python3 -u src/main.py
WorkingDirectory=/home/pi/AIY-projects-python
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

RowanZee, 2018-05-31:
@divx118 That is correct. But I have managed to solve it. I changed the user to pi and reran the authentication process. This seems to have solved the problem as the service now runs. Thanks for  all the help :)

KevinZhou92, 2018-06-11:
@divx118 Holy shit! You are a fking genius!!!. I played around with the autostart stuff all day but just can't get it work.
Can you explain how this magic environment setting works?
`Environment=DISPLAY=:0`

divx118, 2018-06-11:
@KevinZhou92

It just defines `$DISPLAY` in the environment you are launching the service, because there was the below error in the log. I didn't have this problem, but am still on an old version without pulseaudio.

```
Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
```

Btw no genius, just looking at the error messages. :)

ShapeShifter499, 2018-06-23:
Changing my unit file to the following works for me.

```

# This service can be used to run your code automatically on startup. Look in
# HACKING.md for instructions on creating main.py and enabling it.

[Unit]
Description=voice recognizer
After=network.target ntpdate.service

[Service]
Environment=DISPLAY=:0
Environment=PATH=/home/pi/AIY-projects-python/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/usr/bin/python3 -u src/main.py
WorkingDirectory=/home/pi/AIY-projects-python
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

commit 1828cb7b (2018-08-07): Fix the service

These wasn't updated when the virtualenv was removed.

This also replaces the dependency on ntpdate (untested) and defines
DISPLAY (which is required for PulseAudio).

Fixes #295.

Change-Id: Ide4ba19b5588533991af998734a95bbc45ba752a