google/aiyprojects-raspbian #296: Hotword 'Ok google" is not working.

joserochabh, 2018-03-05:
Hi All,

With the new version of AIYProject(aiyprojects-2018-01-03.img.xz) the assistant_grpc_demo.py work fine in portuguese:
===============
pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ ./assistant_grpc_demo.py
/opt/aiy/projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
[2018-03-05 09:05:57,068] INFO:recorder:started recording
Press the button and speak
Listening...
[2018-03-05 09:06:03,825] INFO:speech:transcript: Olá
[2018-03-05 09:06:03,901] INFO:speech:transcript: olá  como
[2018-03-05 09:06:03,954] INFO:speech:transcript: Olá  como vai
[2018-03-05 09:06:03,995] INFO:speech:transcript: Olá como  vai você
[2018-03-05 09:06:04,059] INFO:speech:event_type: 1
[2018-03-05 09:06:04,065] INFO:speech:transcript: Olá como vai  você
[2018-03-05 09:06:04,107] INFO:speech:transcript: Olá como vai você
[2018-03-05 09:06:04,296] INFO:speech:transcript: Olá como vai você
You said " Olá como vai você "
Press the button and speak
=============

The hotword "Ok google"is not working im assistant_library_with_local_commands_demo.py and assistant_library_demo.py 

Thanks,

Jose Rocha

drigz, 2018-03-05:
Please explain what exactly is not working and how it is failing. eg can you copy-paste the error message

joserochabh, 2018-03-05:
Hi, Rodrigo
When I run the assistant_library_with_local_commands_demo.py  it show message like that:

pi@raspberrypi:~/AIY-projects-python/jrproject $ ./assistant_library_with_local_commands_demo.py
/opt/aiy/projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
Say "OK, Google" then speak, or press Ctrl+C to quit...
[3646:3663:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
[3646:3663:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
[3646:3663:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates

I say Ok google but nothing happend, and the Led  turns on and off slowly.

Thanks.

PeterMalkin, 2018-03-05:
Just to make sure, you are running on raspberry pi 3?

PyT, 2018-03-11:
I have exactly the same issue on Rpi3, any idea ?

drigz, 2018-03-12:
The certificate error can be ignored: https://github.com/googlesamples/assistant-sdk-python/issues/192

@joserochabh Has "OK Google" worked for you before in a previous version? It's possible that it's expecting an en-US accent even though the request language is pt-BR.

@PyT, does "OK Google" not work for you either? Which image version are you using?

joserochabh, 2018-03-12:
> The certificate error can be ignored: googlesamples/assistant-sdk-

It's possible. I had replace the language to pt-BR.

Thanks.

Jose Rocha

closed 2018-06-21 by dmitriykovalev

rafaelassis16, 2018-11-01:
@joserochabh Você conseguiu colocar em português ? Estou com os mesmos problemas, já mudei o i18n e até em outros arquivos que estavam en-US.