google/aiyprojects-raspbian #297: Stops talking after about 3 seconds

bacsom, 2018-03-05:
I played with my AIY kit about a month ago and it worked just fine on a Raspberry Pi Zero (not W). Now I've installed it on a Raspberry Pi 1B+ and it seems to be working, but it wouldn't finish its last sentences. For example if I ask about the weather she starts talking, but stops before she could tell me the temperature. "Today In Budapest it'll be mostly sunny with the forecast hig..." The program itself doesn't freeze. If I push the button I can ask another question no problem, but of course she won't finish her answer. It seems that she has no problem with answers less then 3 seconds short. Like "Who is the president of the USA?". But If I ask who Donald Trump is, she stops at 3 seconds. Also, if I ask her to sing, she stops at 3... I've made some reinstalls, I've even replaced my old Rpi1 with an Rpi Zero W and it still stops talking after about 3 seconds... Any help would be greatly appreciated!!!!

bacsom, 2018-03-05:
OK, so it seems that the "stops" happen at different intervals. For example, if I ask her to count to 10, she stops at 7, but if I ask her to count to 50 she stops at 2. Also it seems that she is still speaking in the background because the button stays glowing and "Press the button and speak" line doesn't appear in the SSH window until she supposed to stop answering.

BayAreaCoders, 2018-03-09:
Hi, we are having our the same issue. Not sure what to do.

bacsom, 2018-03-09:
Unfortunately I haven't found any solution. BUT luckily I could find the exact name of an earlier build on one of the forums and then with that information I could find the link for the file on Google's server.

This one do work:
https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2018-01-03.img.xz

What I also noticed that while this version reports no sound card in raspbian (GUI, red x next to the speaker icon), the new non working ISO says there is one.

dmitriykovalev, 2018-03-10:
@bascom, by any chance, have you tried to use Pi 3 (instead of Pi 1B+) with the latest SD card image?

dmitriykovalev, 2018-03-10:
@BayAreaCoders, are you also using Pi 1B+ board or something else?

bacsom, 2018-03-10:
I have a Pi3 board as well but I haven't tried the new iso on that one. I have tried to run the latest image on a Pi1B+ and a PiZeroW... same result. Now it works wonderfully on a PiZeroW with an earlier image.

Tetsuo30, 2018-03-13:
Same issue for me ... RPI3 with last image ... assistant_grpc_demo.py (french i18n config) ... stop the response after 3 or 4 seconds ...

omaewamoushindeiru, 2018-03-14:
I have the same issue. I'm running a Pi 3 with the latest image (aiyprojects-2018-02-21.img.xz), using assistant_grpc_demo.py.  Is the recommended fix to downgrade to aiyprojects-2018-01-03.img.xz?

lordrubenbp, 2018-03-14:
I have the same issue, i try to delete .pyc files and dont work

protovist, 2018-03-14:
This seems to be caused by the PulseAudio suspend-on-idle module.  Please try disabling it and restarting PulseAudio:

```
sudo sed -i -e "s/^load-module module-suspend-on-idle/#load-module module-suspend-on-idle/" /etc/pulse/default.pa
pulseaudio --kill
pulseaudio --start
```

omaewamoushindeiru, 2018-03-19:
@protovist Thank you, your suggested solution worked for me!

srcshelton, 2020-03-17:
This issue doesn't seem to be documented for users installing following instructions in `HACKING.md` on top of an existing Raspbian installation.

(Although adjusting the sample-rate is covered)