google/aiyprojects-raspbian #300: Response by Voice

anh1e, 2018-03-07:
Currently the voice kit will light up when I say a wake word “Hey, Google”

Can I make a change so that so that when I say a wake word “Hey, Google” the system will response by voice such as “Yes” or ‘OK’ as well as light up.

Like Amazon Alexa, The system will response with voice "Yes" when I say a wake word "Alexa"

Thanks,

drigz, 2018-03-08:
This isn't part of the Assistant SDK as far as I'm aware. However, if you don't mind getting your hands dirty you could probably implement it: if you're using the Assistant Library, the `ON_CONVERSATION_TURN_STARTED` event will occur when the Assistant starts listening. You could detect this event and play a pre-recorded WAV of someone saying "OK". You could even use the Assistant Service to get a WAV of the Assistant saying "OK".

af987, 2018-03-20:
A slight addition to the above comment.  If you're using the SDK then you can use:

`aiy.voicehat.get_status_ui().set_trigger_sound_wave('./resources/ding.wav')`

to set an overall sound that is played when the trigger word is heard (obviously changing the path to the wav file as appropriate).  So for the assistant_library_with_local_commands_demo you could change the main procedure to:

```
def main():
    if platform.machine() == 'armv6l':
        print('Cannot run hotword demo on Pi Zero!')
        exit(-1)

    credentials = aiy.assistant.auth_helpers.get_assistant_credentials()
    aiy.voicehat.get_status_ui().set_trigger_sound_wave('./resources/ding.wav')
    with Assistant(credentials) as assistant:
        for event in assistant.start():
            process_event(assistant, event)
```
and the wav file specified would be played each time the trigger word is heard.  Hope that helps.

drigz, 2018-03-26:
Great point, @af987. @anh1e, does the trigger sound work for your use case?

dmitriykovalev, 2018-05-10:
Please reopen if you are still having this issue.

closed 2018-05-10 by dmitriykovalev