google/aiyprojects-raspbian #301: Certificate errors

sheridat, 2018-03-10:
Hi, 
I've just burnt the aiyprojects-2018-02-21.img.xz and installed.

I've tried the assistant_library_demo.py and the local commands version. Both work.

Although both demos work some errors are being thrown out regarding authentication - I've just copied an example below. 

Build info: Thu Feb 22 00:43:02 UTC 2018 @ 3368a38
Dev terminal is ready! See the demos in 'src/examples' to get started.
Note: the MagPi Essentials are not up-to-date. See https://goo.gl/8hEcfo
pi@raspberrypi:~/AIY-projects-python $ src/examples/voice/assistant_library_with_local_commands_demo.py
[1791:1808:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
[1791:1808:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
/opt/aiy/projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
Say "OK, Google" then speak, or press Ctrl+C to quit...
[1791:1808:ERROR:assistant_nupnp_client.cc(103)] Cloud discovery nupnp request failed: [60] Peer certificate cannot be authenticated with given CA certificates
You said: what time is it

divx118, 2018-03-10:
Yes, noticed them too, but it seems they can be ignored see https://github.com/googlesamples/assistant-sdk-python/issues/192

sheridat, 2018-03-11:
@divx118 Cheers - will close

closed 2018-03-11 by sheridat