google/aiyprojects-raspbian #302: Assistant Voice volume level is no longer adjustable via alsamixer/the gui

sheridat, 2018-03-10:
I've just installed the latest image aiyprojects-2018-02-21.img.xz on my Voicekit.

In the past I have adjusted the assistant's voice level by using either the alsamixer or the little loudspeaker GUI control on the desktop. That no longer works - instead I have adjusted the volume level by asking the assistant to increase/decrease it by x percent. Is thus deliberate.

sangj, 2018-03-11:
My Voice kit (aiyprojects-2018-02-21.img.xz) has same problem.

Mixer volume is too loud to listen to music (sound noisy and distorted) 
When I changed volume at amixer or at GUI, actual speaker volume does not change at all. Even I muted the sound at GUI, I hear still loud sound.

Tetsuo30, 2018-03-13:
Same for me, can't adjust the volume ... the guy or amixer doesn't affect the volume, and in assistant_grpc_demo.py, every command i try doesn't affect it ... when asking what's the volume, the answer is always 50% ...

drigz, 2018-03-26:
Thanks for the report! The switch to PulseAudio is intentional, but disabling the volume control was not intentional. We're working on a new image that will fix this.

bfsmith, 2018-04-19:
How do I update to a working volume control if I'm on the 2018-02-21 img?

highsociety2, 2018-05-17:
Experiencing similar issues, and this is a school project for me... Has this been fixed yet? Is there an alternative way to control audio that could bypass this?

babyrobotx, 2018-08-01:
Really appreciate it, drigz.
Would you guys mind releasing a fix rather than a new "image"? or an alternative to choose "not an image"?
I have done many times re-image... and everytime I have to sacrifice losing the work on top of the existing image.... : (

dmitriykovalev, 2018-08-16:
We now install https://freedesktop.org/software/pulseaudio/pavucontrol/ by default.

closed 2018-08-16 by dmitriykovalev

ThePrankMonkey, 2020-10-20:
Changes to Pulse Audio in gui have no impact on the speaker. It remains painfully loud. In the interim, I'm just going to toss a potentiometer inline with the speaker and manually drop the volume that way.