google/aiyprojects-raspbian #303: TTS voice default volume level incorrect?

sheridat, 2018-03-10:
I have just burnt the aiyprojects-2018-02-21.img.xz and installed it on my voicekit.

I ran the assistant_library_with_local_commands_demo.py. 

Requested "ip address" - the output is very loud and distorted. I've had to reduce the setting (in src/aiy/audio.py) of _tts_volume = 60 to _tts_volume = 20 to get a reasonable volume level.

LondonSi72, 2018-03-10:
Yes I noticed that too. I re-applied the "fix" that was suggested for one of the earlier images to improve the sound of the TTS voice. I'll look it up and post here later when I'm back at my terminal.

divx118, 2018-04-01:
@sheridat FYI you don't need to change the source, you just can use the following in your script:
```
aiy.audio.set_tts_volume(20)
aiy.audio.set_tts_pitch(100)
```
See https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/audio.py#L145

LondonSi72, 2018-05-24:
In the latest image I can't find anyway of changing the tts volume or pitch.  Either changing it in the audio.py, or directly within the script, as @divx118 suggests above.  It doesn't seem to make a difference where I change the value, the volume is unchanged.  And it's loud!

kochinc, 2018-05-31:
aiy.audio.say() uses a local TTS engine on your Raspberry Pi. That makes the responses inconsistent to normal Assistant interactions. I found that very annoying. 

To make them consistent in voice and volume, if you don't strictly require them responded locally, you can let Assistant to respond using an assistant.send_text_query('Repeat after me: %s' % your_text) call and handle EventType.ON_RESPONDING_FINISHED to do any actions required.

LondonSi72, 2018-06-19:
@kochinc I'm using the voice kit v1 to control a variety of local devices that aren't natively supported by Google Assistant (like the Edimax smartplug - I'm using a 'hack' to control it via it's API.)  I also have custom responses to the commands I send it, which is kind of fun.  When they aren't shouted back at you...

Do you have an example of the code you suggested to get the Assistant to respond instead?  I'd like to give that a go - though I guess I'd lose the custom response, at least I'd get the local action.

kochinc, 2018-06-23:
@LondonSi72 
Only the newer version of google.assistant.library implements assistant.send_text_query(). The version comes with AIY Voice Kit v1 is old and doesn't have it. So you have to update your google.assistant.library first.
```
$ sudo pip install --upgrade google.assistant-library
```
Once that's done, you can try this modified src/examples/voice/assistant_library_with_local_commands_demo.py. It's a ugly hack, so treat it as a demonstration only.
```
#!/usr/bin/env python3
# Copyright 2017 Google Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Run a recognizer using the Google Assistant Library.

The Google Assistant Library has direct access to the audio API, so this Python
code doesn't need to record audio. Hot word detection "OK, Google" is supported.

It is available for Raspberry Pi 2/3 only; Pi Zero is not supported.
"""

import logging
import platform
import subprocess
import sys

import aiy.assistant.auth_helpers
from aiy.assistant.library import Assistant
import aiy.audio
import aiy.voicehat
from google.assistant.library.event import EventType

logging.basicConfig(
    level=logging.INFO,
    format="[%(asctime)s] %(levelname)s:%(name)s:%(message)s"
)

def power_off_pi():
    subprocess.call('sudo shutdown now', shell=True)

def reboot_pi():
    subprocess.call('sudo reboot', shell=True)

def get_ip():
    ip_address = subprocess.check_output("hostname -I | cut -d' ' -f1", shell=True)
    return ip_address.decode('utf-8')

def process_event(assistant, event, command):

    def say(text):
        assistant.send_text_query('Repeat after me: %s' % text)
        
    status_ui = aiy.voicehat.get_status_ui()
    if event.type == EventType.ON_START_FINISHED:
        status_ui.status('ready')
        if sys.stdout.isatty():
            print('Say "OK, Google" then speak, or press Ctrl+C to quit...')

    elif event.type == EventType.ON_CONVERSATION_TURN_STARTED:
        status_ui.status('listening')

    elif event.type == EventType.ON_RECOGNIZING_SPEECH_FINISHED and event.args:
        print('You said:', event.args['text'])
        text = event.args['text'].lower()
        if text == 'power off':
            command = text
            assistant.stop_conversation()
            say('good bye!')
        elif text == 'reboot':
            command = text
            assistant.stop_conversation()
            say('see you in a bit!')
        elif text == 'ip address':
            command = text
            assistant.stop_conversation()
            say('My IP address is %s.' % get_ip())

    elif event.type == EventType.ON_END_OF_UTTERANCE:
        status_ui.status('thinking')

    elif event.type == EventType.ON_RESPONDING_FINISHED:
        if command == 'power off':
            power_off_pi()
        elif command == 'reboot':
            reboot_pi()
        elif command == 'ip address':
            pass
        command = None

    elif (event.type == EventType.ON_CONVERSATION_TURN_FINISHED
          or event.type == EventType.ON_CONVERSATION_TURN_TIMEOUT
          or event.type == EventType.ON_NO_RESPONSE):
        status_ui.status('ready')

    elif event.type == EventType.ON_ASSISTANT_ERROR and event.args and event.args['is_fatal']:
        sys.exit(1)

    # Local command utterred.
    return command

def main():
    if platform.machine() == 'armv6l':
        print('Cannot run hotword demo on Pi Zero!')
        exit(-1)

    command = None
    credentials = aiy.assistant.auth_helpers.get_assistant_credentials()
    with Assistant(credentials) as assistant:
        for event in assistant.start():
            command = process_event(assistant, event, command)

if __name__ == '__main__':
    main()
```

LondonSi72, 2018-06-26:
Thanks @kochinc I’ll give that a try. Although I’m using the voicekit v1, I’ve got the latest image, so may have the right assistant library anyway, but I’ll check.

mikeyroth, 2018-09-28:
I get
    assistant.send_text_query('Repeat after me: %s' % text)
AttributeError: 'Assistant' object has no attribute 'send_text_query'

mikeyroth, 2018-09-30:
I thought I had.  I am a newbie, but this time I used pip3 instead, and now I got it working   Thanks!  This was very helpful

csmcvea, 2018-11-21:
I am having the same and following your help.
Thank you.