google/aiyprojects-raspbian #304: Support of Media Responses

hyzhak, 2018-03-11:
I'm working right now on actions which use Media Responses but they don't work on AIY Voice. Is there any plan to support it on AIY Voice?
https://developers.google.com/actions/assistant/responses#media_responses

drigz, 2018-03-12:
@hyzhak The AIY library just provides a wrapper around the Assistant SDK, so we're limited to what that supports. Does https://github.com/googlesamples/assistant-sdk-python/issues/132 describe your issue accurately? If not, you could open an upstream issue.

hyzhak, 2018-03-12:
@drigz thanks! I've created dedicated issue https://github.com/googlesamples/assistant-sdk-python/issues/203 because Media Response is a very fresh thing, so that issue could be obsolete.

hyzhak, 2018-10-16:
@drigz @dmitriykovalev could you get new Assistant SDK, so we be able to try Media Response here?
This issue might be solved by new update https://github.com/googlesamples/assistant-sdk-python/issues/203

hyzhak, 2018-10-16:
As I see you still use old `0.1.0` version here https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/setup.py#L17 but we already could use modern one `1.0.0` https://github.com/googlesamples/assistant-sdk-python/blob/master/google-assistant-library/CHANGELOG.rst

whoo, 2019-04-07:
Hi Guys,

Have you got information regarding that?
- It works with x86 version of SDK
- it does not on RaspberryPi3 /B+

Tks