google/aiyprojects-raspbian #305: Hotword detection trigger instead of button trigger in cloudspeech_demo.py

varad1311, 2018-03-13:
Hello! In the cloudspeech_demo.py button trigger is used. How to use hotword  detection,that is, ok google or hey google instead of button trigger? I'm trying to control GPIO pins of raspberry pi 3B in cloudspeech_demo.py. Please help me with this.

varad1311, 2018-03-13:
> Hi @varad1311 <https://github.com/varad1311>! I'm afraid hotwording is only
> supported with the Google Assistant Library.
>
> As for controlling GPIO pins, you can have a look at the AIY Essentials
> <https://www.raspberrypi.org/magpi/issues/essentials-aiy-v1/>, although
> watch out for the changes in the AIY functions that have happened since AIY
> Essentials was written.

Is there any way to control the GPIO pins using voice only? There should be
no need to press the button and then the voice command.

drigz, 2018-03-13:
@varad1311 If you want to use "OK Google" hotwording, you can use assistant_library_demo.py. For Cloud Speech, aallan's Magic Mirror post describes how to use a custom hotword, although there are caveats described in [the post](https://medium.com/@aallan/a-magic-mirror-powered-by-aiy-projects-and-the-raspberry-pi-e6a0fea3b4d6) (Ctrl+F "hotword"). Alternatively, you could look into using other hotwording engines such as Snowboy.