google/aiyprojects-raspbian #307: Unable to Record via Voicehat Mic Board

Flowdeeps, 2018-03-15:
I'm using the latest image, updated and upgraded via apt but the kit is the first revision.

The button light pulses when using the demo scripts, assistant_library_demo.py for example. The speakers work when testing the audio but no mic recording. When I use audacity I can see and record the mic board and get play back.

```
$aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

/boot/config.txt has the dtoverlay lines enabled

```
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard
```

If anyone can help with that I'd be really grateful.

drigz, 2018-03-15:
Can you clarify what you mean by "no mic recording"? For example, what happens when you run the Check Audio script - do you get an error message or just silence?

Flowdeeps, 2018-03-15:
It's just silence. As it's a brand new installation and I can get sound out of it with no issues in Audacity it makes me think it's a setup or software issue.

KarmaMonk, 2018-03-16:
Same Problem here. The microphone stopped working after I upgraded (sudo apt-get upgrade) and restarted the raspi.

Flowdeeps, 2018-03-16:
I had a spare SD card so I thought I'd try reverting to the older image. Worked straight away.

wesmininger, 2018-03-19:
Ive had the same issue with the Feb 21 2018 image but only after doing update/upgrade/reboot.

drigz, 2018-03-26:
Thanks for the feedback, everyone! We're working on a new release that fixes this issue. Until then, you'll need to use the 2018-01-03 image or reflash 2018-02-21 and avoid running `apt-get upgrade`.

af987, 2018-04-17:
I see that a new image has been released (dated 2018-04-13).  Is this issue fixed in this new image?  I might be being a bit thick, but is there a release note for the image indicating which issues it fixes?  If there is, where can it be found as I can't seem to find it.

LondonSi72, 2018-04-17:
Documentation on updates seems to be a bit ad-hoc at best. I tend to wait for others to try and find the bits that have been added/broken and comment here before I spend ages trying to figure out why something has changed.

bwims, 2018-04-27:
The 2018-04-13 version works - BUT.

I have the original installation (2017-05-03), upgraded according the instructions here: http://eduncan911.com/stem/embedded/aiy-google-assistant-upgrades-may-2017.html

And the playback volume of a voice recorded wav file is about twice as loud as one you get from the newest version (2018-04-13)

I want to run the latest version as the old one is almost entirely deprecated, but this recording volume thing is a killer!

Here are the test wav files (Testing, 1 2 3 ) produced by both versions, if you are interested.

https://www.dropbox.com/s/ogi1rhmd5u6vz44/OriginalAIY.wav?dl=0
https://www.dropbox.com/s/9aryvvy1i9j29kf/NewAIY.wav?dl=0

drigz, 2018-05-02:
@bwims Just to confirm, is your issue that the volume is too low with 2018-04-13, or that it is too high when you do the upgrade yourself?

If the latter, I'm afraid we don't support that process, although maybe the workaround to #47 would help you.

bwims, 2018-05-02:
Hi.  

The answer to your question is: "that the volume is too low with 2018-04-13".  

I have just noticed that there is a way to attach files, for which I previously gave dropbox links
I dragged a zip file here called TestWavFiles.zip

[TestWavFiles.zip](https://github.com/google/aiyprojects-raspbian/files/1967213/TestWavFiles.zip)

You should notice that the 2017-05-03 version (OriginalAIY )
produces a much louder, more audible recording than the 2018-04-13 version (NewAIY).

Something about the changes in the new version produced to incorporate the V2 boards has reduced the volume of the recording to unacceptably quiet levels, and I can find no way to affect them.

What I would hope that you mightprovide is a means of adjusting the recording level as per alsamixer, or at least a .ini file with the default setting in an editable form.

Thanks,
Brian

closed 2018-08-16 by dmitriykovalev