google/aiyprojects-raspbian #308: Cannot convert object detection graph from archive from site

ostfor, 2018-03-15:
Hello, I have downloaded an [archive](https://dl.google.com/dl/aiyprojects/vision/tf_models_2017_12_13.tgz) (from https://aiyprojects.withgoogle.com/vision) with `mobilenet_v1_160res_0.5_imagenet.pb` and `mobilenet_ssd_256res_0.125_person_cat_dog.pb` and tried to convert it  with a `bonnet_model_compiler.par`. A `mobilenet_v1_160res_0.5_imagenet.pb` case works well, but `mobilenet_ssd_256res_0.125_person_cat_dog.pb` has not been converted. 
I have used a following command:
```bash
./bonnet_model_compiler.par 
--frozen_graph_path="./mobilenet_ssd_256res_0.125_person_cat_dog.pb" 
--input_tensor_name="image_tensor" 
--output_tensor_names="concat,concat_1" 
--input_tensor_size=256 --output_graph_path="./converted_graph.binaryproto"
```
It seems that graph in archive just hasn't been prepared for conversion. For example it has some preprocessing operations (like `tf.Cast` and normalization ops). Could you help me to find a ready-to-convert  `./mobilenet_ssd_256res_0.125_person_cat_dog.pb` graph?

weiranzhao, 2018-03-15:
@ostfor Thanks for bring this to our attention.
Could you try the compiler tool with a different input_tensor_name, 'Preprocessor/sub'? This works on my end. We'll update the website to include this information as well.

ostfor, 2018-03-16:
Yes, it works, thank you! 

And one more question. Can I train model using https://github.com/tensorflow/models/tree/master/research/object_detection tool and just use a graph obtained by export_inference_graph script or it need to be modified? 
The matter is that i successfully convert such graph. But when i tried to test it on device it raise a error:
```bash
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 128, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, self._tbuf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/object_detection.py", line 50, in main
    result = inference.run(image_center)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 98, in run
    return self._engine.image_inference(self._key, image, params)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 319, in image_inference
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 143, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/object_detection.py", line 60, in <module>
    main()
  File "/home/pi/AIY-projects-python/src/examples/vision/object_detection.py", line 56, in main
    image.save(args.output)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 108, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 101, in close
    self._engine.unload_model(self._key)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 241, in unload_model
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Model 'object_detection' is not loaded.
```

P.S. Default model (`mobilenet_ssd_256res_0.125_person_cat_dog`) works well

sainttelant, 2018-03-19:
@ostfor 
Hi guy,
         i 've tired to compile my own frozen model by using compiler tool, the compiler is installed and tested successfully in my ubuntu system, however, when i converted my Pb file trained by object_detection api of tensorflow to binaryproto file, it popped the errors like the following:

~/anaconda3/envs/tensorflow/models/object_detection/vehiclemodel$ ./bonnet_model_compiler.par \
> --frozen_graph_path=./vehicledemo_graph.pb \
> --output_graph_path=./vehicledemo_graph.binaryproto \
> --input_tensor_name="input" \
> --output_tensor_names="MobilenetV1/Predictions/Softmax" \
> --input_tensor_size=160
Traceback (most recent call last):
  File "/home/wx/anaconda3/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/wx/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "./bonnet_model_compiler.par/__main__.py", line 129, in <module>
  File "./bonnet_model_compiler.par/__main__.py", line 75, in main
  File "./bonnet_model_compiler.par/compiler/bonnet_model_compiler_utils.py", line 189, in user_friendly_error_msg
TypeError: a bytes-like object is required, not 'str'

what is wrong? anyone knows?

sainttelant, 2018-03-19:
@weiranzhao 
what does the input_tensor_name mean? is there any format we should follow when tried to compile pb files? or just give it a random name?

weiranzhao, 2018-03-19:
@ostfor Could you provide some more details on how you run your own model? 

Generally, you will need to write your own code to load your model to the device and interpret the result returned from device. A good starting point is to look at 
src/aiy/vision/models/object_detection.py
src/examples/vision/object_detection.py

weiranzhao, 2018-03-19:
@sainttelant For example, you have a trained TF graph, and you want to convert it to on-device graph. input_tensor_name points to the node you want to use as input for the on-device graph. It could be a different node than the input node of your TF graph. Does this make sense?

sainttelant, 2018-03-20:
@weiranzhao
thanks for your help, get it, i printed all nodes'name in jupyter, and i also found there are 'Preprocessor/sub' in my model, what i need to do is to specify or pick up random nodes'name from them, and give them in compiler tool, however, i am still confused the input_tensor_size, how can i ensure the accurate value of it?  i am a new one in TF,  plz help me, thanks for your help

weiranzhao, 2018-03-20:
@sainttelant You should pick up a node from the graph. Usually you do NOT want to pick a random one because the node you pick as input will be used to take input image when inference. The node you pick should be the starting node for inference.

Input tensor size is the image size you use to train your model, as you mentioned in issues314, you were using 160x160. So you should pass 'input_tensor_size' 160 here.

sainttelant, 2018-03-22:
@weiranzhao perfect explanation, i checked my graph and the graph of example(160_mobilenetV1),by comparison with  them displayed on tesorboard, i found their structure are almost same,no difference,  i think the start node name both should  be "input",  the output_tensor_name of mine, i took the node of MobilenetV1/Predictions/Softmax as well, put it into the compiler, it still had error, and then, i tried to use the last node of graph('num_detections'), it didn't work, the error still popped up

File "./bonnet_model_compiler.par/compiler/bonnet_model_compiler_utils.py", line 189, in user_friendly_error_msg
TypeError: a bytes-like object is required, not 'str'

minakhan01, 2018-04-22:
@weiranzhao @ostfor can you please take a look at #335? It'd be super helpful. I think your comments are relevant but I am not sure how to apply them to my case...

minakhan01, 2018-04-22:
@weiranzhao @ostfor I also tried the following, but neither of them work:

./bonnet_model_compiler.par \
    --frozen_graph_path=./mobilenet_ssd_256res_0.125_person_cat_dog.pb \
    --output_graph_path=./converted_graph.binaryproto \
    --input_tensor_name="Preprocessor/sub" \
    --output_tensor_names=“concat,concat_1” \
    --input_tensor_size=256

./bonnet_model_compiler.par \
    --frozen_graph_path=./mobilenet_ssd_256res_0.125_person_cat_dog.pb \
    --output_graph_path=./converted_graph.binaryproto \
    --input_tensor_name="image_tensor" \
    --output_tensor_names=“concat,concat_1” \
    --input_tensor_size=256

abhinavrawat27, 2019-02-19:
> mobilenet_ssd_256res_0.125_person_cat_dog

@ostfor What is the exact model name for `mobilenet_ssd_256res_0.125_person_cat_dog`