google/aiyprojects-raspbian #309: ImportError: No module named 'aiy'

melzoghbi, 2018-03-16:
I am trying to setup aiy projects on RPI3, as i follow the inrecutions on the [hacking.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) file, i am stuck at this step:

env/bin/python checkpoints/check_audio.py

First, i do not have env folder, so i adjusted the command to be:

[image]

How can i fix this issue?

I searched other old threads, but none of them helped me to resolve this issue. Please note that i use Raspbian Stretch on RPI3.

Thanks.

dmitriykovalev, 2018-03-16:
Please use `python3` instead of `python`, or run the script directly: `./check_audio.py`

melzoghbi, 2018-03-16:
Same thing:

[image]

When i debugged my file, i found this is the import that causes the exception:

import aiy._drivers._alsa

MithicalM, 2018-03-19:
https://github.com/google/aiyprojects-raspbian/blob/2a78558cda2ac390638d2e1f062b77d491c13482/checkpoints/check_audio.py#L28
This is where I am getting stuck on.

Jack2905, 2018-03-19:
Hi,
I managed to get this working by copying the aiy folder found in src to the checkpoint directory. I used the following commands from the root of the project:

`cp -R src/aiy checkpoints/aiy`
`./checkpoints/check_audio.py`

melzoghbi, 2018-03-20:
I am having the same issue on two boards. The mic does not play a recording but the speaker is working. I tried the check audio bash file but it did not work on both boards!

MithicalM, 2018-03-20:
Yeah I have the same issue as you do
https://github.com/google/aiyprojects-raspbian/issues/309#comment-374707981

pyorgue, 2018-03-24:
Same for me. I am following the HACKING,md guideline and have the same error.
I don't have the env folder as well.

pyorgue, 2018-03-24:
Ok here is the solution:

- follow the instructions in https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md
- just after cloning the git repository, rollback to a previous version by using >git checkout c5745ec
- continue the instruction

The bug is due to a change in the install-deps.sh that has remove the creation of the /env folder.

NicoLivesey, 2018-04-16:
Hi,

I'm raising again this issue since I can't find any good solution and even though the one provided by @Jack2905 helped for the installation, when you want to run `assistant_library_demo.py` you still need the venv.

Thank you for your help !

str8edgedave, 2018-04-17:
I noticed there are new kits and new image file available at https://dl.google.com/dl/aiyprojects/aiyprojects-latest.img.xz. The actual file downloaded is aiyprojects-2018-04-13.xz. Has anyone tried the new image out to see if it resolves this problem? I'm at work right now.

weboide, 2018-04-21:
You can actually run it by modifying PYTHONPATH env variable (same for running the examples). Like so:
`PYTHONPATH="src/" checkpoints/check_audio.py`

But I'm stuck on installing the drivers for the voicehat. I'm getting `Failed to find the voiceHAT soundcard` even after installing the alsa configs.

theochva, 2018-04-22:
I had the same problem with the missing aiy module, but all i had to do was "activate" the python environment "env" using `source ./env/bin/activate` :
```
pi@pi3plus:~/AIY-projects-python $ src/examples/voice/assistant_grpc_demo.py 
Traceback (most recent call last):
  File "src/examples/voice/assistant_grpc_demo.py", line 20, in <module>
    import aiy.assistant.grpc
ImportError: No module named 'aiy'
pi@pi3plus:~/AIY-projects-python $ source ./env/bin/activate
(env) pi@pi3plus:~/AIY-projects-python $ env/bin/python checkpoints/check_audio.py^C
(env) pi@pi3plus:~/AIY-projects-python $ src/examples/voice/assistant_grpc_demo.py
Please visit this URL to authorize this application: https://accounts.google....
```

dmitriykovalev, 2018-05-10:
We got rid of virtualenv, please use the latest SD card image. Everything should just work:

```
pi@raspberrypi:~ $ python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import aiy
>>> 
```

closed 2018-05-10 by dmitriykovalev

tobyweston, 2018-06-05:
@dmitriykovalev it doesn't! 😉 Well, at least building from source doesn't...

Building from source and following `HACKING.md` doesn't work. There's a bunch of related issues including errors like the following which I receive on a fresh Stretch install.

```
$ checkpoints/check_audio.py 
Traceback (most recent call last):
  File "checkpoints/check_audio.py", line 28, in <module>
    import aiy.audio  # noqa
ImportError: No module named 'aiy'
```

I've forked and am testing by reverting the removal of `install-deps.sh` and potentially reintroducing virtualenv, but I'm not a python guy and aren't sure about the impact. I'm happy to hack away and submit a PR for feedback but I think this warrants some proper attention and testing.

If you create a fresh Stretch Lite image and follow the instructions from `HACKING.md`, you'll quickly see the issues. 

Is there competition between the image and building from source? Personally, I'd expect to be able to manually install things so would favour support of both.

damasiomg, 2018-06-13:
Hello.
Has anyone been able to install by HACKING.md?
Any update forecast?
Thanks!

tobyweston, 2018-06-13:
This is just plain broken and there seems to be no interest in even commenting here. I’ve started a fork but it’s not 100% working with a fresh raspbian :sad:

damasiomg, 2018-06-13:
Wow. This is so frustrating!

wilksy, 2018-06-16:
this worked for me.. from above.

be in AIY_PROJECTs dir already

cp -R src/aiy checkpoints/aiy
python3 checkpoints/check_audio.py

patoberli, 2018-07-29:
Sadly having the same problem once I try to run assistant_grpc_demo.py.

ndmax, 2018-07-30:
This is indeed very broken and very frustrating.  The kit website (https://aiyprojects.withgoogle.com) needs an update or it will continue to waste everyone's time.  Is there any place to find instructions on the current image linked above?

recroviral, 2018-12-11:
I also get the same problem but I slove it using
**import aiy.voice.audio**
instead of **import aiy.audio**

zzulanas, 2018-12-16:
I've been having this issue too but with 
`Traceback (most recent call last):
  File "./led_demo.py", line 7, in <module>
    import aiy.voicehat
ImportError: No module named 'aiy.voicehat'`

I can't find the voicehat module anywhere when searching through the /aiy/ directory

note: I'm also using the older version of the AIY MagPi guide, so when I was trying to do the led_demo.py I had this issue

feh123, 2019-01-26:
I am trying to use a similar idea. I have a working Voice Kit and I have seen Github threads that install vlc, add commands to get internet radio stations and hence play internet radio. I can get this to work with vlc on a raspberry pi only kit using google assistant library where I have a usb mic and external speaker. I use the assistant_library_with_local_commands_demo.py to which I add the extra command lines. However a parallel setup on the Voice kit only gives the error "I cannot play that yet". The setup will answer the usual questions. 
I guess there is no solution to this problem either?

aseelothman, 2019-03-24:
hellow there, im having the same problem; (No module named aiy.audio)
when using the following code,  help me please im stuck
`
#!/usr/bin/env python3
# Copyright 2017 Google Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Check that the voiceHAT audio input and output are both working."""

import os
import sys
import tempfile
import textwrap
import traceback

sys.path.append(os.path.realpath(os.path.join(__file__, '..', '..')) + '/src/')

import aiy.audio  

CARDS_PATH = '/proc/asound/cards'
VOICEHAT_ID = 'googlevoicehat'

STOP_DELAY = 1.0

TEST_SOUND_PATH = '/usr/share/sounds/alsa/Front_Center.wav'

RECORD_DURATION_SECONDS = 3

def get_sound_cards():
    """Read a dictionary of ALSA cards from /proc, indexed by number."""
    cards = {}

    with open(CARDS_PATH) as f:  # pylint: disable=invalid-name
        for line in f.read().splitlines():
            try:
                index = int(line.strip().split()[0])
            except (IndexError, ValueError):
                continue

            cards[index] = line

    return cards

def ask(prompt):
    """Get a yes or no answer from the user."""
    ans = input(prompt + ' (y/n) ')

    while not ans or ans[0].lower() not in 'yn':
        ans = input('Please enter y or n: ')

    return ans[0].lower() == 'y'

def check_voicehat_present():
    """Check that the voiceHAT is present."""
    return any(VOICEHAT_ID in card for card in get_sound_cards().values())

def check_voicehat_is_first_card():
    """Check that the voiceHAT is the first card on the system."""
    cards = get_sound_cards()
    return 0 in cards and VOICEHAT_ID in cards[0]

def check_speaker_works():
    """Check the speaker makes a sound."""
    print('Playing a test sound...')
    aiy.audio.play_wave(TEST_SOUND_PATH)

    return ask('Did you hear the test sound?')

def check_mic_works():
    """Check the microphone records correctly."""
    temp_file, temp_path = tempfile.mkstemp(suffix='.wav')
    os.close(temp_file)

    try:
        input("When you're ready, press enter and say 'Testing, 1 2 3'...")
        print('Recording...')
        aiy.audio.record_to_wave(temp_path, RECORD_DURATION_SECONDS)
        print('Playing back recorded audio...')
        aiy.audio.play_wave(temp_path)
    finally:
        try:
            os.unlink(temp_path)
        except FileNotFoundError:
            pass

    return ask('Did you hear your own voice?')

def do_checks():
    """Run all audio checks and print status."""
    if not check_voicehat_present():
        print(textwrap.fill(
            """Failed to find the voiceHAT soundcard. Refer to HACKING.md for
how to setup the voiceHAT driver: https://git.io/v99yK"""))
        return

    if not check_voicehat_is_first_card():
        print(textwrap.fill(
            """The voiceHAT not the first sound device, so the voice recognizer
may be unable to find it. Please try removing other sound drivers."""))
        return

    if not check_speaker_works():
        print(textwrap.fill(
            """There may be a problem with your speaker. Check that it's
connected properly."""))
        return

    if not check_mic_works():
        print(textwrap.fill(
            """There may be a problem with your microphone. Check that it's
connected properly."""))
        return

    print('The audio seems to be working.')

def main():
    do_checks()

if __name__ == '__main__':
    try:
        main()
        input('Press Enter to close...')
    except Exception:  # pylint: disable=W0703
        traceback.print_exc()
        input('Press Enter to close...')
`

thechrisdavis, 2019-06-12:
Hey peeps, 

I purchased my AIY voice kit on clearance from Target and had the same trouble.  I was able to get the demos to run via the following command.  

cd ~/AIY-projects-python/src/examples

voice/assistant_grpc_demo.py

Hope that helps someone else out there that was having trouble.  Good luck!