google/aiyprojects-raspbian #310: Current image (2018-02-31) does not work with Raspberry Pi 3 B+

Green7, 2018-03-17:
Current image can't boot on RPi3B+  (rainbow screen, power led off, act led blinking in patern).
I think that kernel is too old ...

MithicalM, 2018-03-19:
I have the same issue with my one.
This explains the reason why. 
https://www.raspberrypi.org/forums/viewtopic.php?t=58151
I hope this gets updated/patched soon

jverelst, 2018-03-19:
Same issue here, hope it gets updated soon. I just assembled my AIY with a brand new 3B+ board and it won't boot up...

MithicalM, 2018-03-19:
I just noticed that this was an option. I am going to give it a shot and see how it goes.
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

drigz, 2018-03-19:
@MithicalM As a heads up, some users have reported problems with HACKING.md and the latest code, eg https://github.com/google/aiyprojects-raspbian/issues/309. We'd be grateful if you could report any issues or workarounds you find!

Green7, 2018-03-19:
I tried https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md
But without success. I stuck on #309

jverelst, 2018-03-20:
I worked around the issue by putting the SD card into an older raspberry pi and using 'apt-get update' and 'apt-get upgrade' to update rapbian to the latest version. 
The AIY kit itself now boots up correctly on the 3B+, only the microphone doesn't work. I can't verify if that's a hardware issue or software issue yet.

jverelst, 2018-03-20:
Retried with a different (older) Raspberry Pi board: microphone does work. 

Conclusions:
- Updating the AIY distro from raspbian 2.1 to 2.3 might allow it to boot up on a 3B+
- The 3B+ with the updated distro doesn't support the microphone yet.

melzoghbi, 2018-03-20:
@jverelst it is a software issue. i had the same issue on two boards!

AdrienLacroix123, 2018-03-30:
Hello,
I have the same issue with a Raspberry Pi 3B+, I can't boot and since it's my only board I don't have any solution.
Do you know how often the code is updated ?
Thank you.

MithicalM, 2018-03-30:
I managed to get it working using this guys tutorial. https://github.com/shivasiddharth/GassistPi

kengberg, 2018-04-04:
Noooo... Bought the kit and a 3 b+
- and it's not working!

Stuck on the rainbow screen.. thought it was backwards compatible... :-/

drigz, 2018-04-17:
@str8edgedave It ought to, although I haven't tried it myself. Good luck!

weboide, 2018-04-21:
Using the 2.0 voice bonnet, I've tried the 2018-04-13 image and it boots fine on a rpi3b+:

- Audio and microphone work fine.
- assistant_grpc_demo.py works.
- The assistant library ones just seem to hang and do nothing.
- alsamixer works to change volume but it's a quite low, even at 100%.

theochva, 2018-04-22:
I tried using the latest raspbian image as of today (2018-04-18) on rpi3b+ and it worked.  I used the latest version of the master branch of this repo but included the "scripts/install-deps.sh" script from the commit "c5745ec"  (https://github.com/google/aiyprojects-raspbian/blob/c5745ec6cbde0a2269ca2c6e2ba02f0538bbc926/scripts/install-deps.sh). The script is needed to create the python environment "env".

mattncsu, 2018-04-22:
I had no luck with an AIY 1.0 kit on a 3B+ using aiyprojects-2018-04-13.xz -- Got past the rainbow screen to 4 raspberries but just a cursor at the top.  Also struck out following hacking.md on a fresh install of raspbian.  Will have to try the install-deps.sh script and see how that works.

windigoHub, 2018-04-22:
I have AIY 1.0 and raspi 3B+ using aiy-projects-281804.xz and it's not booting either. I am getting to the 4 raspberries and one _ ticking just below, but it stays like that indefinitely.

windigoHub, 2018-04-22:
@theochva I tried this, but now I have the same problem as referenced in issue #309 .
When I tried to check_audio.py the error message "ImportError: No module named 'aiy'". Then I used "source env/bin/activate" and when I try check_audio.py now I get the message "Failed to find the voiceHAT soundcard"

mattncsu, 2018-04-23:
Started over and followed the instructions above (including pulling the old version of install-deps.sh but I get this error:
`The voiceHAT not the first sound device, so the voice recognizer may
be unable to find it. Please try removing other sound drivers.
Press Enter to close...`
Was the onboard sound driver supposed to be disabled by one of the included scripts?

```
pi@raspberrypi-AIY:~/AIY-projects-python $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
I followed [some steps](http://www.instructables.com/id/Disable-the-Built-in-Sound-Card-of-Raspberry-Pi/) to disabled the sound driver but it resulted in my menu bar disappearing so I reverted.

theochva, 2018-04-23:
I am thinking what else i had may have done. I know that at some point, i also ran `configure-driver.sh` (from the old commit) because i see the changes it performs in my `/boot/config.txt`

This is the "tail" part of my `/boot/config.txt` file:
```

hack-r, 2018-04-30:
I also cannot get the 04/13/2018 image to load on 3B+

hack-r-mypi, 2018-04-30:
I also have same problem as above. Latest image 04/13/2018 and a 3B+
Is there any step by step instructions on how to get this going. I tried following the instructions above theochva but I couldnt follow. I am certain many people who have bought this voice kit, and have bought a new PI3B+ would be as frustrated that the software isnt up to date with the latest pi PI3B+
Any help would be appreciated. Trying to get this working for a science project
Thankyou in advance

Ziocozzo, 2018-05-02:
I had the same problem. I was using a 4GB microSD card and all the procedure seems to be ok, but the bootstrap was blocking with the black screen and the cursor blinking in the left-top corner.
I have the PI3B+ with AIY Voice Kit 1.0 and I downloaded the last OS image:  https://dl.google.com/dl/aiyprojects/aiyprojects-latest.img.xz
The solution was very simple: I changed the 4GB  microSD card with 16GB one! (probably also a 8GB uSD card is enough)
Now everything works fine.

hack-r, 2018-05-02:
@Ziocozzo Nice! The problem for me / us must be a little different than yours though because I was on 8gb and 16gb

The weird thing is that I did get the 8gb flash card from the newest vision kit to work on the voice kit. But not the downloaded image.

hack-r-mypi, 2018-05-06:
Good afternoon, I had been told to reimage my SD card too see if the imaging was the problem. The aiyprojects 2018/04/13 image I reimaged with Etcher and it worked perfectly. So using rpi3b+ and V1 google voice kit and it finally worked.
Once I got to the raspbian desktop, went to terminal  and done an update 
Sudo apt-get update
This updated all systems packages and then my wifi worked. I couldn't get it to work before that.
And finally setup Google api and scripts and my son done the rest of the testing so he can now take in the Google voice for his assignment for school. Cheers

closed 2018-05-06 by dmitriykovalev

hsheikha1429, 2018-05-13:
Faced the same with Google voice kit V1 & Raspberry pi 3 B+.
Once I re-do the project with RaspBerry Pi 3 B instead of the B+ the issue solved and the kit booted normally.