google/aiyprojects-raspbian #311: Running Assistant On Boot-Up with voice-recognizer.service does not work

thecoconutdream, 2018-03-17:
Hi everyone.
So I followed every step of the HACKING.md but it simply does not work. I use this image: aiyproject-2018-02-21
The first problem I am dealing with is that: `source env/bin/activate` does not work-->No such file or directory
But I was thinking that as long as I am working in the desktop shell, it is activated anyway.
So as a next step I did:`python3 src/examples/voice/assistant_library_demo.py`-->That worked.
Next step:
`sudo systemct1 enable voice-recognizer.service`
And then:
`sudo service voice-recognizer status`:
[image]
This is my huge problem.....I tried to create my own service but I alway get this failed state.... Should I try to use an older image?

divx118, 2018-03-17:
@thecoconutdream 
See https://github.com/google/aiyprojects-raspbian/issues/295

drigz, 2018-03-26:
As @divx118 pointed out, this has been reported at #295. You can try the workaround in https://github.com/google/aiyprojects-raspbian/issues/295#issuecomment-375127938.

closed 2018-03-26 by drigz