google/aiyprojects-raspbian #313: Mic stopped working after software upgrade using Feb 21 2018 image

wesmininger, 2018-03-19:
Ive tried this 2x now to verify. With the stock image, using my already created cloud json files, it works. The audio test script works correctly as well. As soon as I do sudo apt-get update && sudo apt-get upgrade, the mic no longer works with the audio tests or using the example python scripts. Instead the Dev environment puts out an error

pi@voicebox:~/AIY-projects-python $ /home/pi/AIY-projects-python/src/examples/voice/assistant_library_demo.py Say "OK, Google" then speak, or press Ctrl+C to quit... [1075:1105:ERROR:mic_failure_detector.cc(140)] [assistant] Channel 1 has failed. Powers -inf and -inf

When I revert to the original image it works, so its not hardware. Something in the upgrade breaks the mic.

av4625, 2018-03-19:
I have the same issue on the new image.
I did the apt-get update and upgrade.
The 'my_assistant.py' file got deleted on the upgrade and my microphone wont work after.
I can run 'check_audio', she says front and center or (or whatever it is) and then i record me saying Testing 1, 2, 3 and nothing is played back.

All this worked before the update/upgrade

rhowlett, 2018-03-19:
I just ran into this as well.  Looked at my /var/log/user.log files and found this right after the upgrade:
This is the UPGRADE:
dpkg.log:2018-02-06 18:37:11 install libportaudiocpp0:armhf <none> 19.6.0-1
dpkg.log:2018-02-06 18:37:11 status half-installed libportaudiocpp0:armhf 19.6.0-1
dpkg.log:2018-02-06 18:37:11 status unpacked libportaudiocpp0:armhf 19.6.0-1
dpkg.log:2018-02-06 18:37:11 status unpacked libportaudiocpp0:armhf 19.6.0-1
dpkg.log:2018-02-06 18:37:14 install portaudio19-dev:armhf <none> 19.6.0-1
dpkg.log:2018-02-06 18:37:14 status half-installed portaudio19-dev:armhf 19.6.0-1
dpkg.log:2018-02-06 18:37:15 status unpacked portaudio19-dev:armhf 19.6.0-1
dpkg.log:2018-02-06 18:37:15 status unpacked portaudio19-dev:armhf 19.6.0-1
dpkg.log:2018-02-06 18:37:15 install pulseaudio-utils:armhf <none> 10.0-1+deb9u1
dpkg.log:2018-02-06 18:37:15 status half-installed pulseaudio-utils:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:37:15 status unpacked pulseaudio-utils:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:37:15 status unpacked pulseaudio-utils:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:37:15 install pulseaudio:armhf <none> 10.0-1+deb9u1
dpkg.log:2018-02-06 18:37:15 status half-installed pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:37:17 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:37:17 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:38:23 configure libportaudiocpp0:armhf 19.6.0-1 <none>
dpkg.log:2018-02-06 18:38:23 status unpacked libportaudiocpp0:armhf 19.6.0-1
dpkg.log:2018-02-06 18:38:23 status half-configured libportaudiocpp0:armhf 19.6.0-1
dpkg.log:2018-02-06 18:38:23 status installed libportaudiocpp0:armhf 19.6.0-1
dpkg.log:2018-02-06 18:38:24 configure pulseaudio-utils:armhf 10.0-1+deb9u1 <none>
dpkg.log:2018-02-06 18:38:24 status unpacked pulseaudio-utils:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:38:24 status half-configured pulseaudio-utils:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:38:24 status installed pulseaudio-utils:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:08 configure portaudio19-dev:armhf 19.6.0-1 <none>
dpkg.log:2018-02-06 18:40:08 status unpacked portaudio19-dev:armhf 19.6.0-1
dpkg.log:2018-02-06 18:40:08 status half-configured portaudio19-dev:armhf 19.6.0-1
dpkg.log:2018-02-06 18:40:08 status installed portaudio19-dev:armhf 19.6.0-1
dpkg.log:2018-02-06 18:40:11 configure pulseaudio:armhf 10.0-1+deb9u1 <none>
dpkg.log:2018-02-06 18:40:11 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:11 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:11 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:11 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:11 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:11 status unpacked pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:11 status half-configured pulseaudio:armhf 10.0-1+deb9u1
dpkg.log:2018-02-06 18:40:14 status installed pulseaudio:armhf 10.0-1+deb9u1

Then I get this after the upgrade (I changed the timezone and went back in time for the savvy observer):

user.log:Feb 22 00:48:59 raspberrypi pulseaudio[1046]: [pulseaudio] module-x11-publish.c: PulseAudio information vanished from X11!

Time to dig...

philfin42, 2018-03-21:
I also have this issue when upgrading. 

pi@raspberrypi:~ $ dmesg | grep -i 'voice'
[    8.121638] snd-googlevoicehat-soundcard soc:sound: ASoC: CPU DAI (null) not registered - will retry
[    8.121657] snd-googlevoicehat-soundcard soc:sound: snd_soc_register_card() failed: -517
[    8.123588] voicehat-codec voicehat-codec: property 'voicehat_sdmode_delay' not found default 5 mS
[    8.123906] snd-googlevoicehat-soundcard soc:sound: ASoC: CPU DAI (null) not registered - will retry
[    8.123917] snd-googlevoicehat-soundcard soc:sound: snd_soc_register_card() failed: -517
[    8.128454] snd-googlevoicehat-soundcard soc:sound: voicehat-hifi <-> 3f203000.i2s mapping ok
[   31.928912] voicehat-codec voicehat-codec: Enabling audio amp...
[   36.995480] voicehat-codec voicehat-codec: Disabling audio amp...
[  457.021527] voicehat-codec voicehat-codec: Enabling audio amp...
[  478.644135] voicehat-codec voicehat-codec: Disabling audio amp...
[ 2016.830154] voicehat-codec voicehat-codec: Enabling audio amp...

Playback is set to 100, and Capture is set to 100 in alsamixer

Emailed support, no response.

HansMolenveld, 2018-03-25:
I have also this issue after upgrading.
Mar 25 13:23:24 googlevoice systemd[1]: Started My awesome assistant app.
Mar 25 13:23:27 googlevoice pulseaudio[1612]: [pulseaudio] sink.c: Default and alternate sample rates are the same.
Mar 25 13:23:27 googlevoice pulseaudio[1612]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 25 13:23:27 googlevoice pulseaudio[1612]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 25 13:23:28 googlevoice bash[1600]: /opt/aiy/projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
Mar 25 13:23:28 googlevoice bash[1600]:   GPIO.setup(channel, GPIO.OUT)
Mar 25 13:24:29 googlevoice bash[1600]: [1600:1637:ERROR:mic_failure_detector.cc(140)] [assistant] Channel 1 has failed. Powers -inf and -inf

tmbrns, 2018-03-26:
I had the same problem.  It has something to do with the interaction between ALSA and PulseAudio.  I read somewhere that PulseAudio is no longer installed by default in Stretch because ALSA is supposed to handle Bluetooth audio itself.  So, maybe the interaction between them hasn't been well tested.  I found [this page](https://wiki.archlinux.org/index.php/PulseAudio/Troubleshooting#Low_volume_for_internal_microphone) with a lot of PulseAudio troubleshooting tips.  Under the heading "Microphone not detected by PulseAudio" it suggested adding the following line to /etc/pulse/default.pa:

load-module module-alsa-source device=hw:0,0"

I found a commented out version of that line so I removed the semi-colon and maybe edited the hw:X,Y part.  It turned out that I also needed to uncomment the line above it:

load-module module-alsa-sink

Then to restart PulseAudio:

pulseaudio -k; pulseaudio -D

Then I started the Assistant Library Demo and said "OK Google, can you hear me?" and she replied "Loud and Clear!".

Unfortunately the other demo where you have to push the button (Assistant GRPC Demo) no longer works.  The two demos access the microphone data differently.  According to the pulseaudio(1) manual page you can tell it to read an alternate configuration file using the -C option along with the -n option to disable reading of /etc/pulse/default.pa.  So, you could have a separate file for each demo if you want to run both.

HansMolenveld, 2018-03-26:
tmbrns thank you it worked.  There has still be work to be done because I get error messgages for the pulsaudio that should not happen. See the logging below:
Mar 26 15:28:24 googlevoice pulseaudio[888]: [pulseaudio] module.c: Failed to load module "module-alsa-source" (argument: "device=hw:1,0"): initialization failed.
Mar 26 15:28:24 googlevoice pulseaudio[888]: [pulseaudio] main.c: Module load failed.
Mar 26 15:28:24 googlevoice pulseaudio[888]: [pulseaudio] main.c: Failed to initialize daemon.
Mar 26 15:28:24 googlevoice pulseaudio[879]: [pulseaudio] main.c: Daemon startup failed.

drigz, 2018-03-26:
Thanks for the investigation! I've posted a link to @tmbrns's comment in #307 - please continue discussion there if you discover anything more.

closed 2018-03-26 by drigz

tmbrns, 2018-03-26:
HansMolenveld, on the load module line, try changing the device to hw:0,0 if you haven't already.

You may have noticed the comment in the file saying that those lines should not be needed because the udev module should take care of loading the sink and source modules.  So, maybe something changed in udev but I don't know much about that.

Also, I forgot to mention before that you can include set-source-volume and set-sink-volume lines in /etc/pulse/default.pa:

set-sink-volume 0 200000

That boosts the speaker volume to  200,000.  The default of 65536 was too low.  I didn't need to change the microphone volume, but I was sitting just a few feet from my kit.  I haven't tried from across the room yet.

av4625, 2018-03-28:
After doing the above mine still doesn't work.
When I start the assistant it listens to me then spams this log:
`[1344:1378:ERROR:audio_input_processor.cc(748)] Input error`

It did get what I say and prints it just before it spams the above.

When I run the check audio I get:
`Enabling audio driver for VoiceKit.
Playing a test sound...
Did you hear the test sound? (y/n) y
When you're ready, press enter and say 'Testing, 1 2 3'...
Recording...
Playing back recorded audio...
aplay: pcm_write:2011: write error: Input/output error
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 180, in <module>
    main()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 175, in main
    do_checks()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 157, in do_checks
    if not check_mic_works():
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 117, in check_mic_works
    aiy.audio.play_wave(temp_path)
  File "/opt/aiy/projects-python/src/aiy/audio.py", line 104, in play_wave
    player.play_wav(wave_file)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_player.py", line 71, in play_wav
    self.play_bytes(frames, wav.getframerate(), wav.getsampwidth())
  File "/opt/aiy/projects-python/src/aiy/_drivers/_player.py", line 52, in play_bytes
    aplay.stdin.write(audio_bytes)
BrokenPipeError: [Errno 32] Broken pipe`

av4625, 2018-03-28:
Sometimes I dont get that far when running check audio and I get this:
```Enabling audio driver for VoiceKit.
Playing a test sound...
aplay: pcm_write:2011: write error: Input/output error
aplay crashed - try checking your ALSA config.
Press Enter to close...```

hack-r, 2018-04-29:
After 

`pulseaudio -k; pulseaudio -D
`
I get 

```
overrun!!! (at least 12.958 ms long)
overrun!!! (at least 8.635 ms long)
overrun!!! (at least 9.721 ms long)
overrun!!! (at least 9.290 ms long)
overrun!!! (at least 10.018 ms long)
overrun!!! (at least 12.989 ms long)
overrun!!! (at least 9.451 ms long)
overrun!!! (at least 10.073 ms long)
```