google/aiyprojects-raspbian #314: compile my own pb model into binaryproto error

sainttelant, 2018-03-19:
i 've trained my model  via objection _detection of tensorflow, and it tested successfully in jupyter notebook,i intent to deploy it on vision kit, so i used compiler.par tool to generate binaryproto file, the test of compiler via using tutorial's model is okay, no error, however, when i tried to compile my own model, the error occurred as follows..

~/anaconda3/envs/tensorflow/models/object_detection/vehiclemodel$ ./bonnet_model_compiler.par \

--frozen_graph_path=./vehicledemo_graph.pb 
--output_graph_path=./vehicledemo_graph.binaryproto 
--input_tensor_name="input" 
--output_tensor_names="MobilenetV1/Predictions/Softmax" 
--input_tensor_size=160
Traceback (most recent call last):
File "/home/wx/anaconda3/lib/python3.5/runpy.py", line 184, in _run_module_as_main
"main", mod_spec)
File "/home/wx/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
exec(code, run_globals)
File "./bonnet_model_compiler.par/main.py", line 129, in 
File "./bonnet_model_compiler.par/main.py", line 75, in main
File "./bonnet_model_compiler.par/compiler/bonnet_model_compiler_utils.py", line 189, in user_friendly_error_msg
TypeError: a bytes-like object is required, not 'str'

what is wrong? anyone knows?

sainttelant, 2018-03-19:
by the way,  is that caused by the wrong configuration of pipeline.config? i checked the instruction, the supported common graphs of mobilenet, its input size is restrained in 160*160, and depth multiplier=0.5, is it a reason for error occurring? anyone knows? it wasn't ought to be a problem it looks! confused, my pipeline likes: it is modified by pets mobilenet

# SSD with Mobilenet v1, configured for Oxford-IIIT Pets Dataset.
# Users should configure the fine_tune_checkpoint field in the train config as
# well as the label_map_path and input_path fields in the train_input_reader and
# eval_input_reader. Search for "PATH_TO_BE_CONFIGURED" to find the fields that
# should be configured.

model {
  ssd {
    num_classes: 14
    box_coder {
      faster_rcnn_box_coder {
        y_scale: 10.0
        x_scale: 10.0
        height_scale: 5.0
        width_scale: 5.0
      }
    }
    matcher {
      argmax_matcher {
        matched_threshold: 0.5
        unmatched_threshold: 0.5
        ignore_thresholds: false
        negatives_lower_than_unmatched: true
        force_match_for_each_row: true
      }
    }
    similarity_calculator {
      iou_similarity {
      }
    }
    anchor_generator {
      ssd_anchor_generator {
        num_layers: 6
        min_scale: 0.2
        max_scale: 0.95
        aspect_ratios: 1.0
        aspect_ratios: 2.0
        aspect_ratios: 0.5
        aspect_ratios: 3.0
        aspect_ratios: 0.3333
      }
    }
    image_resizer {
      fixed_shape_resizer {
        height: 300
        width: 300
      }
    }
    box_predictor {
      convolutional_box_predictor {
        min_depth: 0
        max_depth: 0
        num_layers_before_predictor: 0
        use_dropout: false
        dropout_keep_probability: 0.8
        kernel_size: 1
        box_code_size: 4
        apply_sigmoid_to_scores: false
        conv_hyperparams {
          activation: RELU_6,
          regularizer {
            l2_regularizer {
              weight: 0.00004
            }
          }
          initializer {
            truncated_normal_initializer {
              stddev: 0.03
              mean: 0.0
            }
          }
          batch_norm {
            train: true,
            scale: true,
            center: true,
            decay: 0.9997,
            epsilon: 0.001,
          }
        }
      }
    }
    feature_extractor {
      type: 'ssd_mobilenet_v1'
      min_depth: 16
      depth_multiplier: 1.0
      conv_hyperparams {
        activation: RELU_6,
        regularizer {
          l2_regularizer {
            weight: 0.00004
          }
        }
        initializer {
          truncated_normal_initializer {
            stddev: 0.03
            mean: 0.0
          }
        }
        batch_norm {
          train: true,
          scale: true,
          center: true,
          decay: 0.9997,
          epsilon: 0.001,
        }
      }
    }
    loss {
      classification_loss {
        weighted_sigmoid {
          anchorwise_output: true
        }
      }
      localization_loss {
        weighted_smooth_l1 {
          anchorwise_output: true
        }
      }
      hard_example_miner {
        num_hard_examples: 3000
        iou_threshold: 0.99
        loss_type: CLASSIFICATION
        max_negatives_per_positive: 3
        min_negatives_per_image: 0
      }
      classification_weight: 1.0
      localization_weight: 1.0
    }
    normalize_loss_by_num_matches: true
    post_processing {
      batch_non_max_suppression {
        score_threshold: 1e-8
        iou_threshold: 0.6
        max_detections_per_class: 100
        max_total_detections: 100
      }
      score_converter: SIGMOID
    }
  }
}

train_config: {
  batch_size: 24
  optimizer {
    rms_prop_optimizer: {
      learning_rate: {
        exponential_decay_learning_rate {
          initial_learning_rate: 0.004
          decay_steps: 800720
          decay_factor: 0.95
        }
      }
      momentum_optimizer_value: 0.9
      decay: 0.9
      epsilon: 1.0
    }
  }
  fine_tune_checkpoint: "gs://autovehicle/dataset/model.ckpt"
  from_detection_checkpoint: true
  # Note: The below line limits the training process to 200K steps, which we
  # empirically found to be sufficient enough to train the pets dataset. This
  # effectively bypasses the learning rate schedule (the learning rate will
  # never decay). Remove the below line to train indefinitely.
  num_steps: 200000
  data_augmentation_options {
    random_horizontal_flip {
    }
  }
  data_augmentation_options {
    ssd_random_crop {
    }
  }
}

train_input_reader: {
  tf_record_input_reader {
    input_path: "gs://autovehicle/dataset/train.record"
  }
  label_map_path: "gs://autovehicle/dataset/vehicle_label_map.pbtxt"
}

eval_config: {
  num_examples: 2000
  # Note: The below line limits the evaluation process to 10 evaluations.
  # Remove the below line to evaluate indefinitely.
  max_evals: 10
}

eval_input_reader: {
  tf_record_input_reader {
    input_path: "gs://autovehicle/dataset/test.record"
  }
  label_map_path: "gs://autovehicle/dataset/vehicle_map.pbtxt"
  shuffle: false
  num_readers: 1
}

sainttelant, 2018-03-19:
looking forward to discussing with you guys

sainttelant, 2018-03-19:
i think it maybe a bug,plz check it, my pb file is definite a binary byte file, why can't it be compiled?weird.

weiranzhao, 2018-03-19:
@sainttelant 160x160 input size with 0.5 depthwise multiplier should be supported on-device. Do you mind share a version of your graph? It would be easier for us to debug it that way. You can set the weights of your model to some random number if you don't feel comfortable sharing the weights. Thanks.

sainttelant, 2018-03-20:
@weiranzhao  my config is almost the same with ssd_mobilenet_v1_pets of tensorflow's object_detection,  do you mean the weight's value 0.00004 which i used directly from pets_config? or classification_weight: 1.0
localization_weight: 1.0

weiranzhao, 2018-03-20:
@sainttelant The configs looks OK. Are you able to convert your TF pb file with our compiler?

sainttelant, 2018-03-22:
@weiranzhao ,yes, i am, i know how to configure input_tensor_name, and output_tensor_name,but have no idea how to specify the size of input

sainttelant, 2018-03-22:
@weiranzhao ,yes, i am, i will try again according you mentioned in 308 issue

sainttelant, 2018-03-22:
@weiranzhao ,unfortunately, it is still not successfully converted, i tried to compare my own graph with example mobilenetV1 graph on tensorboard, i found the main structure is almost same, the start node both are "input".however the output node of example is Mobilenet/Predictions/Softmax, i checked my graph, found above output as well, the difference is that Mobilenet/Predictions/Softmax is the last node for example, but the last output of mine is "num_detection".

i intent to use both of them as output_tensor_name, run compiler tool CMD, the errors are still there

weiranzhao, 2018-03-23:
@sainttelant
What are the command line argument you're using now? Could you paste it here?

sainttelant, 2018-03-26:
@weiranzhao  I printed all graph nodes as follows:

Const
image_tensor
ToFloat
Preprocessor/map/Shape
Preprocessor/map/strided_slice/stack
Preprocessor/map/strided_slice/stack_1
Preprocessor/map/strided_slice/stack_2
Preprocessor/map/strided_slice
Preprocessor/map/TensorArray
Preprocessor/map/TensorArrayUnstack/Shape
Preprocessor/map/TensorArrayUnstack/strided_slice/stack
Preprocessor/map/TensorArrayUnstack/strided_slice/stack_1
Preprocessor/map/TensorArrayUnstack/strided_slice/stack_2
Preprocessor/map/TensorArrayUnstack/strided_slice
Preprocessor/map/TensorArrayUnstack/range/start
Preprocessor/map/TensorArrayUnstack/range/delta
Preprocessor/map/TensorArrayUnstack/range
Preprocessor/map/TensorArrayUnstack/TensorArrayScatter/TensorArrayScatterV3
Preprocessor/map/Const
Preprocessor/map/TensorArray_1
Preprocessor/map/while/Enter
Preprocessor/map/while/Enter_1
Preprocessor/map/while/Merge
Preprocessor/map/while/Merge_1
Preprocessor/map/while/Less/Enter
Preprocessor/map/while/Less
Preprocessor/map/while/LoopCond
Preprocessor/map/while/Switch
Preprocessor/map/while/Switch_1
Preprocessor/map/while/Identity
Preprocessor/map/while/Identity_1
Preprocessor/map/while/TensorArrayReadV3/Enter
Preprocessor/map/while/TensorArrayReadV3/Enter_1
Preprocessor/map/while/TensorArrayReadV3
Preprocessor/map/while/ResizeImage/ExpandDims/dim
Preprocessor/map/while/ResizeImage/ExpandDims
Preprocessor/map/while/ResizeImage/size
Preprocessor/map/while/ResizeImage/ResizeBilinear
Preprocessor/map/while/ResizeImage/Squeeze
Preprocessor/map/while/TensorArrayWrite/TensorArrayWriteV3/Enter
Preprocessor/map/while/TensorArrayWrite/TensorArrayWriteV3
Preprocessor/map/while/add/y
Preprocessor/map/while/add
Preprocessor/map/while/NextIteration
Preprocessor/map/while/NextIteration_1
Preprocessor/map/while/Exit_1
Preprocessor/map/TensorArrayStack/TensorArraySizeV3
Preprocessor/map/TensorArrayStack/range/start
Preprocessor/map/TensorArrayStack/range/delta
Preprocessor/map/TensorArrayStack/range
Preprocessor/map/TensorArrayStack/TensorArrayGatherV3
Preprocessor/mul/x
Preprocessor/mul
Preprocessor/sub/y
Preprocessor/sub
FeatureExtractor/Shape
FeatureExtractor/strided_slice/stack
FeatureExtractor/strided_slice/stack_1
FeatureExtractor/strided_slice/stack_2
FeatureExtractor/strided_slice
FeatureExtractor/GreaterEqual/y
FeatureExtractor/GreaterEqual
FeatureExtractor/Shape_1
FeatureExtractor/strided_slice_1/stack
FeatureExtractor/strided_slice_1/stack_1
FeatureExtractor/strided_slice_1/stack_2
FeatureExtractor/strided_slice_1
FeatureExtractor/GreaterEqual_1/y
[... 2737 lines cut ...]
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/range/limit
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/range/delta
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/range
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/Gather/Switch
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/Gather
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/Rank
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/Shape/Switch
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/Shape
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/strided_slice/stack
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/strided_slice/stack_1
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/strided_slice/stack_2
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/strided_slice
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/sub/x
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/sub
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/ExpandDims/dim
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/ExpandDims
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/Greater/y
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/Greater
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/cond/Switch
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/cond/switch_t
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/cond/switch_f
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/cond/pred_id
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/cond/strided_slice/stack
Postprocessor/BatchMultiClassNonMaxSuppression/map/while/PadOrClipBoxList/cond/cond/strided_slice/stack_1
Postprocessor

sainttelant, 2018-03-26:
i tried 
./bonnet_model_compiler.par --frozen_graph_path=vehicledemo_graph.pb --output_graph_path=vehicledemo_graph.binaryproto --input_tensor_name="input" --output_tensor_names="num_detections" --input_tensor_size=160 
as well

sainttelant, 2018-03-26:
@weiranzhao 
could you plz help me to convert my graph into binaryproto file? i don't mind sending my graph to you via email as you wish, is that okay? my email address is  sainttelant@163.com, you can contact me as you want,thanks a lot

weiranzhao, 2018-03-26:
@sainttelant Could you try to compile your graph with 
--input_tensor_name="Preprocessor/sub" --output_tensor_names="concat,concat_1"

Some background info. Your inference graph consists of 3 parts. Usually preprocessing + inference + post-processing. The input and outputs name you want to use is the input and outputs for the inference part.
If you look at the TF graph, you will find layers with name prefix 'Preprocessor', 'FeatureExtractor', 'Postprocessor' that correspond to each of the phases. 

Let me know if this works or not. If not, we'll try something differently.

sainttelant, 2018-03-27:
@weiranzhao  here is my CMD argument 

 ./bonnet_model_compiler.par \
> --frozen_graph_path=./vehicledemo_graph.pb \
> --output_graph_path=./vehicledemo_graph.binaryproto \
> --input_tensor_name="Preprocessor/sub" \
> --output_tensor_names="concat,concat_1" \
> --input_tensor_size=160
Traceback (most recent call last):
  File "/home/wx/anaconda3/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/wx/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "./bonnet_model_compiler.par/__main__.py", line 129, in <module>
  File "./bonnet_model_compiler.par/__main__.py", line 75, in main
  File "./bonnet_model_compiler.par/compiler/bonnet_model_compiler_utils.py", line 189, in user_friendly_error_msg
TypeError: a bytes-like object is required, not 'str'
unfortunately , it still needs to be modified, i will try as well

weiranzhao, 2018-03-27:
@sainttelant I see you are using py3.5, could you try to install py2.7 and set it as default to run the compiler?

Regarding model size, you'll need to see the converted graph's size. As long as it is smaller than 12MB, it should be fine.

sainttelant, 2018-03-27:
@weiranzhao 
(py2) wx@wilson:~/anaconda3/envs/tensorflow/models/object_detection/vehiclemodel$ ./bonnet_model_compiler.par --frozen_graph_path=./mobilenet_v1_160res_0.5_imag
enet.pb --output_graph_path=./mobilenet_v1_160res_0.5_imagenet.binaryproto --input_tensor_name="input" --output_tensor_names="MobilenetV1/Predictions/Softmax" --input_tensor_size=160
VisionBonnet model binary ./mobilenet_v1_160res_0.5_imagenet.binaryproto generated.
(py2) wx@wilson:~/anaconda3/envs/tensorflow/models/object_detection/vehiclemodel$ 

i tried to compiler example graph to binaryproto file, it is okay i used py2 environmemt, however, when i tried to convert my own graph, it still doesn't work.

sainttelant, 2018-03-27:
@weiranzhao , i think it isn't ought to be the py version' s error, i will try to figure it out

weiranzhao, 2018-03-27:
@sainttelant, Now you get 'Unexpected error' instead of 'TypeError: a bytes-like object is required, not 'str''. Meaning the previous error was caused by python environment. 
But your graph is still not compiled. 

There are two things I noticed about your training config.
1) 
image_resizer {
fixed_shape_resizer {
height: 300
width: 300
}
}
This is still training a model that accepts 300x300 input.  I think you want a 160x160 model input.
2) 
depth_multiplier: 1.0
The mobilenet model is still using a depth multiplier of 1.0, you might want a 0.5 depth multiplier.

I'm planning to release a newer version of the compiler by EOW, which will add an option to print more debug info. I will keep you updated. But I think you'd need to fix above 2 issues as well, otherwise, the model is too big to run on VisionKit.

sainttelant, 2018-03-27:
@weiranzhao 
Thanks for your help, that means i will have to modify my training config as above mentioned, otherwise, it couldn't be converted successfully, i will try to retrain the model in the future, thanks, but i am still not sure whether there are more places needing to be modified in config file before putting it on training

weiranzhao, 2018-03-27:
@sainttelant 
Thanks for bring this to our attention! For your case, the pre-trained model you use is 300x300 with depth multiplier 1.0; it is not possible to fine tune such graph to have 160x160 with depth multiplier 0.5.

It is easier to train your graph from scratch. But it will take longer time to train. The tutorial in TF object_detection seems to support training from scratch.

sainttelant, 2018-03-28:
@weiranzhao .
thanks for your help, i suggest google could probably release some pre_trained models for our training models,it may save the training time,and avoid improper  configurations, thanks alot.

chadwallacehart, 2018-04-20:
I followed the recommended training config above using [MobileNet_v1_0.50_160](http://download.tensorflow.org/models/mobilenet_v1_2018_02_22/mobilenet_v1_0.5_160.tgz) as my `fine_tune_checkpoint` when training. I am able to train and export without issue, but when I try to convert the frozen graph with the bonnet compiler I get an `Unexpected error.`

```
./bonnet_model_compiler.par --frozen_graph_path=cat_detector_interference_graph_aiy/frozen_inference_graph.pb --output_graph_path=cat_detector.binaryproto --input_tensor_name="input" --output_tensor_names="MobilenetV1/Predictions/Reshape_1" --input_tensor_size=160
```

The `mobilenet_v1_0.5_160/mobilenet_v1_0.5_160_info.txt` file inside the [MobileNet_v1_0.50_160](http://download.tensorflow.org/models/mobilenet_v1_2018_02_22/mobilenet_v1_0.5_160.tgz) download indicates the `Output:` is `MobilenetV1/Predictions/Reshape_1` so I tried that in addition to `Softmax` as the `output_tensor_names`.

Just using the downloaded graph works without issue:
```
./bonnet_model_compiler.par --frozen_graph_path=mobilenet_v1_0.5_160_frozen.pb --output_graph_path=mobilenet_v1_0.5_160_frozen.binaryproto --input_tensor_name="input" --output_tensor_names="MobilenetV1/Predictions/Softmax" --input_tensor_size=160
```

👍 for @sainttelant's request for:

1. Vision-kit friendly pre-trained models that can be used for transfer learning
2. Some kind of reference guide on how to do this

@weiranzhao - please let us know when your new compiler is ready.

crobbiano, 2018-04-22:
I am also having the same issues as everyone else while trying to compile my own model.  

For those of you with this issue, you can unzip the bonnet_model_compilier.par to get a look at the scripts that are being run when you try and compile.  In particular, there are some debug options that you can turn on to get a more verbose output, as well as adding your own debugging.  I will see if I can get my model to compile today and report back if I make any progress.

crobbiano, 2018-04-23:
@minakhan01 Yup.  Using Ubuntu 16.04 I just simply ran `unzip bonnect_modile_compiler.par`.  You should get the following contents:

`~/Downloads - 08:52:48 - $ unzip bonnet_model_compiler.par 
Archive:  bonnet_model_compiler.par
  inflating: __main__.py             
  inflating: compiler/__init__.py    
  inflating: compiler/bonnet_model_compiler  
  inflating: compiler/bonnet_model_compiler.py  
  inflating: compiler/bonnet_model_compiler_utils.py  
  inflating: compiler/licenses.txt   
  inflating: compiler/tool_a         
  inflating: compiler/tool_b         
  inflating: subpar/__init__.py      
  inflating: subpar/runtime/__init__.py  
  inflating: subpar/runtime/support.py
`

minakhan01, 2018-04-25:
thank you, @crobbiano !

sainttelant, 2018-04-25:
@crobbiano ,Hi guy, have you fixed the error up right now?

crobbiano, 2018-04-25:
Unfortunately, no.  I dug into it a bit and was getting "Mognet" errors in compiled code that I couldn't access..  Comparing outputs to a successful compilation of the example nets also didn't get me that far.

chadwallacehart, 2018-04-26:
In the process of unzipping `bonnet_model_compiler.par` I discovered there are a bunch of log messages dumped to `/tmp/`. There is no need to unzip and run the compiler in python to see them.

All mine start with `/tmp/tool_a.bin.`

'cat /tmp/tool_a.bin.FATAL` showed me the compiler has issues with my output tensor that I am currently trying to figure out. There is also symbolic links for `/tmp/tool_a.bin.INFO`, '/tmp/tool_a.bin.WARNING`, and `/tmp/tool_a.bin.ERROR` that point to the corresponding log files for the last run.

sainttelant, 2018-04-26:
@chadwallacehart  that is right, that is why i strongly recommended vision kit team to release some pre-trained models, our users just need to train our own models through existing models, it not only saves amount of training time but also avoiding configuring parameters error.

chadwallacehart, 2018-04-27:
Hi @weiranzhao - sorry to pester, but do you have any recommendations on how we can retrain an existing model? Are there any details on how the models listed [here](https://aiyprojects.withgoogle.com/models/) were made? Perhaps it might be easier to start with one of the image classification models?

I am giving a talk about the AIY Vision Kit at a local Google Developers Group meetup here in Boston on Monday evening. I would like to reference how to do transfer learning with the kit - ideally with a demo. Is there anything beyond what is on the project page you can share or anyone else I should ask?

weiranzhao, 2018-04-27:
@sainttelant @chadwallacehart @crobbiano @minakhan01 
Folks, we released a new version of the model compiler. It should have fixed the compiling issue you guys are experiencing. Please give it a try and let me know if that works.

sainttelant, 2018-04-28:
@weiranzhao ,that is great, i will try it later, and let you guys know the result

sainttelant, 2018-04-28:
@weiranzhao  ， does that mean i should remove some unused nodes?

weiranzhao, 2018-04-28:
@sainttelant Is this a newly trained graph? I remember you once re-trained a graph using 224x224 pretrained check points. That is not going to work. You have to re-train your graph using 160x160 pretrained check points.

chadwallacehart, 2018-04-30:
@weiranzhao - thanks for the update. Here is the output log it gave:

```
2018-04-30 10:20:37.218799: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayV3
2018-04-30 10:20:37.218908: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayScatterV3
2018-04-30 10:20:37.218931: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayV3
2018-04-30 10:20:37.218946: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.218959: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.218978: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.218990: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: LoopCond
2018-04-30 10:20:37.219012: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayReadV3
2018-04-30 10:20:37.219026: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.219038: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.219064: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayWriteV3
2018-04-30 10:20:37.219141: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.219179: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Exit
2018-04-30 10:20:37.219194: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArraySizeV3
2018-04-30 10:20:37.219226: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayGatherV3
2018-04-30 10:20:37.229493: F external/org_tensorflow/tensorflow/contrib/lite/toco/tooling_util.cc:822] Check failed: d >= 1 (0 vs. 1)
```
Here is the bonnet compiler command I used (same as before):
`./bonnet_model_compiler.par --frozen_graph_path=cat_detector_interference_graph_aiy/frozen_inference_graph.pb --output_graph_path=cat_detector.binaryproto --input_tensor_name="input" --output_tensor_names="concat,concat_1" --input_tensor_size=160 --debug`

Here is a copy of my training folder contents with the `pipeline.config` I used for reference: https://www.dropbox.com/s/jwdzhtotwjvk5o3/aiy_train.zip?dl=0

minakhan01, 2018-04-30:
@chadwallacehart issue #335 may be relevant to you. it's not the complete solution, but a couple of steps in the right direction.

@weiranzhao thanks for the help! Please let us know if you have updates about the new error in #335

weiranzhao, 2018-05-01:
@chadwallacehart Did you train the graph from scratch, or did you fine tune your training based on a pre-trained model? If the second case, what is the config for your pre-trained model?

chadwallacehart, 2018-05-01:
@weiranzhao - I probably do not understand this correctly, but I wanted to fine tune based on a pre-trained model. 

I used the following checkpoint configuration:
```
  fine_tune_checkpoint: "/Users/chad/code/cv/cat_detector/models/mobilenet_v1_0.5_160/mobilenet_v1_0.5_160.ckpt"
  # from_detection_checkpoint: true
```

The full config is located in this gist: https://gist.github.com/chadwallacehart/7781bd59b425508c2f7b7cf31986969c

I can give the procedure @minakhan01 describes in [#335](https://github.com/google/aiyprojects-raspbian/issues/335) a try (thanks for sharing that!).

weiranzhao, 2018-05-01:
@chadwallacehart 
May I know where you get the pre-trained checkpoints? I know TF object detection team release some pretrained checkpoints. https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md

But their checkpoints is based on 300x300, depthwise 1.0 input. IIRC, it is not easy/possible to fine tune a 300x300, depthwise 1.0 pretrained model to take 160x160 depthwise 0.5 input.

chadwallacehart, 2018-05-01:
@weiranzhao - I grabbed the MobileNet_v1_0.50_160 
 model file with checkpoints from here: https://github.com/tensorflow/models/blob/master/research/slim/nets/mobilenet_v1.md

weiranzhao, 2018-05-01:
@chadwallacehart 
The checkpoints you got are for mobilenet_v1, usually for a classifier (without bounding box). The training script (you used) says it is SSD+mobilenet_v1, it is for detection, which outputs a bounding box+scores. I don't think they can work together.
If you want to do with a classifier, the Poet tutorial mentioned in #335 is a good starting point.
If you want a detector, I think you will have to train from scratch (instead of fine tuning) because there's no SSD+mobilenet_v1 with input 160x160 available yet.

chadwallacehart, 2018-05-02:
@weiranzhao
I retrained with no checkpoints using the object detection API training script as I did before. I am getting the same error output:

```
2018-04-30 10:20:37.218799: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayV3
2018-04-30 10:20:37.218908: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayScatterV3
2018-04-30 10:20:37.218931: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayV3
2018-04-30 10:20:37.218946: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.218959: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.218978: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.218990: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: LoopCond
2018-04-30 10:20:37.219012: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayReadV3
2018-04-30 10:20:37.219026: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.219038: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.219064: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayWriteV3
2018-04-30 10:20:37.219141: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Enter
2018-04-30 10:20:37.219179: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Exit
2018-04-30 10:20:37.219194: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArraySizeV3
2018-04-30 10:20:37.219226: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: TensorArrayGatherV3
2018-04-30 10:20:37.229493: F external/org_tensorflow/
```
I guess I'll need to wait until some details on how the cat/dog/human detector was made are shared.

weiranzhao, 2018-05-03:
@chadwallacehart 
Hmm, could you try the following command (for the ssd+mobilenet detector model you trained from scratch)?
./bonnet_model_compiler.par --frozen_graph_path=cat_detector_interference_graph_aiy/frozen_inference_graph.pb --output_graph_path=cat_detector.binaryproto --input_tensor_name="Preprocessor/sub" --output_tensor_names="concat,concat_1" --input_tensor_size=160 --debug

In previous comment, you were using 'input' for input_tensor_name.

If this does not work, do you mind to share a copy of the frozen graph? Thanks!

chadwallacehart, 2018-05-03:
@weiranzhao 
In compiled- thank you!!!

Next I went to try it out using the `object_detection.py` using the following structure and renaming to avoid conflicts:
`cat_detection.py` - from `/opt/aiy/projects-python/src/examples/vision/object_detection`
`models`
 - `cat_detector_v2.binaryproto` - what I just compiled
`cat_detector`
 - `aiy_cat_detection.py` - from `/opt/aiy/projects-python/src/aiy/vision/models/object_detection.py`
- `aiy_cat_detection_anchors.py` - from `/opt/aiy/projects-python/src/aiy/vision/models/object_detection_anchors.py`

I fixed all the source directories in these files so it ran, then I and gave it a try:
`./cat_detection.py --input cat2.jpeg`

But that gave me the following error:
```
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 128, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, self._tbuf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./cat_detection.py", line 50, in main
    result = inference.run(image_center)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 98, in run
    return self._engine.image_inference(self._key, image, params)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 317, in image_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 143, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./cat_detection.py", line 60, in <module>
    main()
  File "./cat_detection.py", line 56, in main
    image.save(args.output)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 108, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 101, in close
    self._engine.unload_model(self._key)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 241, in unload_model
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Model 'object_detection' is not loaded.
```
I verified this same test setup works fine when I point it to `mobilenet_ssd_256res_0.125_person_cat_dog.binaryproto`.

Note this graph has 0 steps of training - I just wanted to see if it would load. I am not sure if that is relevant.

The frozen graph I used prior to compiling is located here: https://drive.google.com/open?id=1sdGoN6tOVQ9lfdbv1Zwhi3d1aU_HrAaS

Any recommendations on how to debug the spicomm error?

weiranzhao, 2018-05-04:
@chadwallacehart Great progress!
I think if you flash your SD card with the latest image, 
https://dl.google.com/dl/aiyprojects/aiyprojects-latest.img.xz
It should work. The generated binaryproto (with latest compiler) needs the latest firmware to run.

chadwallacehart, 2018-05-04:
@weiranzhao 

ok, I installed the 4-13 image (previously I had one from February). The bonnet compiler I used is `bonnet_model_compiler_2018_04_26.tgz`. I still get the same error:

```
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./cat_detection.py", line 50, in main
    result = inference.run(image_center)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 98, in run
    return self._engine.image_inference(self._key, image, params)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 317, in image_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 152, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./cat_detection.py", line 60, in <module>
    main()
  File "./cat_detection.py", line 56, in main
    image.save(args.output)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 108, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 101, in close
    self._engine.unload_model(self._key)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 241, in unload_model
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Model 'object_detection' is not loaded.
```

weiranzhao, 2018-05-04:
@chadwallacehart 
There is (at least) one operator that uses a configuration that is not supported on VisionBonnet. In this case, a convolution operator that uses kernel height/width greater than the input tensor height/width. 
This should have been caught by our compiler, we will fix this with next compiler release.

There's one configuration you can try is to use input 256x256, depthwise multiplier 0.125 when you configure your training. This is the same configuration as the one we ship by default. It should (is more likely to) work this way.

chadwallacehart, 2018-05-05:
@weiranzhao 
I tried with 0.125 depthwise multiplier and 256x256 - same exact error output.
Here are some of the parts of the training config file for reference:
```
    image_resizer {
      fixed_shape_resizer {
        height: 256
        width: 256
      }
    }
```
```
 feature_extractor {
      type: 'ssd_mobilenet_v1'
      min_depth: 16
      depth_multiplier: 0.125
```

Was `mobilenet_ssd_256res_0.125_person_cat_dog.binaryproto` made with the standard [Object Detection API](https://github.com/tensorflow/models/tree/master/research/object_detection)? Perhaps it required some modifications?
Do you have access to the training config file for that for reference?

sainttelant, 2018-05-07:
@chadwallacehart ,yes, i also used tensorflow object detection api  to train the model, Maybe there exists some uncompatible  issues between vision kit teams supplied and tensorflow API.

weiranzhao, 2018-05-08:
@chadwallacehart @sainttelant 
Could you guys use this config when training your model 
https://github.com/tensorflow/models/blob/master/research/object_detection/samples/configs/embedded_ssd_mobilenet_v1_coco.config
This is the same config for our obj detector

chadwallacehart, 2018-05-09:
hi @weiranzhao - I just gave that a try, but I got the same error:
```
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/code/cat_detection.py", line 60, in <module>
    main()
  File "/home/pi/code/cat_detection.py", line 44, in main
    with ImageInference(aiy_cat_detection.model()) as inference:
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 95, in __init__
    self._key = self._engine.load_model(descriptor)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 225, in load_model
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 152, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError
```

For reference, the difference in the `embedded_ssd_mobilenet_v1_coco.config` and the last one I tried were the following:

* `feature_extractor { type:` `'embedded_ssd_mobilenet_v1'` vs.   `ssd_mobilenet_v1'`
* `anchor_generator {ssd_anchor_generator { num_layers: `: `5` vs `6`

chadwallacehart, 2018-05-09:
@weiranzhao - I am happy to debug this with you in real time if that is easier.  My gmail is chadwhart.

lleonid, 2018-05-09:
@chad
Does it work if you replace your cat_detector_v2.binaryproto with
/opt/aiy/models/mobilenet_ssd_256res_0.125_person_cat_dog.binaryproto ?
I wonder if the part where you "fixed all the source directories in these
files" introduced an error.

chadwallacehart, 2018-05-09:
@lleonid - I just tried that and it was fine:

I replaced: `#_COMPUTE_GRAPH_NAME = '/home/pi/code/models/aiy_cat_detector_embedded_ssd_v4.pb'`

with: `_COMPUTE_GRAPH_NAME = '/opt/aiy/models/mobilenet_ssd_256res_0.125_person_cat_dog.binaryproto'`

and it ran the same as the `object_detection.py` demo as it should:
```
pi@raspberrypi:~/code $ ./cat_detection.py --input cat2.jpeg 
Object #0: kind=PERSON(2), score=0.790692, bbox=(443, 220, 598, 549)
```

Its actually a cat, but I swapped the label order from the person_cat_dog model (mine only does cats & people).

weiranzhao, 2018-05-09:
@chadwallacehart Thanks for being patient with us! Let me use their training script to train a model myself. It could be their trained model contains some operation that is not supported by myriad firmware yet. If I get it work, I will document the steps and release it. Either way, I will get back to you on this thread by EOW.

chadwallacehart, 2018-05-09:
Thanks @weiranzhao - that documentation would be much appreciated and I'm here to try it out when it is ready.

weiranzhao, 2018-05-14:
@chadwallacehart 
I used [a] to retrain a detector model myself and deployed on Vision Kit without problem.

Here are a few things to check.
1) I retrained with pascal data set. Should change 'num_classes' to 20. And use pascal_train/eval data set.
2) I used method described here [b] to export the frozen graph
3) I used the compiler to compile the frozen graph.
    ./bonnet_model_compiler.par --frozen_graph_path=frozen_inference_graph.pb \
    --output_graph_path=try.binaryproto  \
    --input_tensor_name="Preprocessor/sub" \
    --output_tensor_names="concat,concat_1" \
     --input_tensor_size=256
4) You can hack 'examples/vision/mobilenet_based_classifier.py' to let it just load the model and run inference. (we will release a script just for loading and get inference result back soon) You will need to interpret the result yourself when using the model eventually, but for testing the model works, you can skip the interpreting step. It returns 2 tensors, 1 for bounding box encoding, the other has class scores for each bounding box. 
    Note, the returned tensors sizes are big for vision kit. Tensor a is 1278 bounding box encodings, which is 1278x4xsizeof(float). Tensor b has class scores (plus background) for each bounding box encoding, that is in total 1278x(20+1)xsizeof(float). That is in total ~125KB per inference.
If you are the mscoco data set, which has 90 different classes, the return tensors sizes is around ~475KB.

[a]: https://github.com/tensorflow/models/blob/master/research/object_detection/samples/configs/embedded_ssd_mobilenet_v1_coco.config
[b]: https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/exporting_models.md

I will publish a tutorial on this soon. Let me know if this is working for you or not.

sainttelant, 2018-05-14:
@weiranzhao  @chadwallacehart ,thanks you guys for excellent discussion,  i am busy with my current job right now, have no spare time to retrain and deploy the model,if you guys successfully fine tune your models, let us know and post the instruction we will appreciate so much, thanks you guys.

chadwallacehart, 2018-05-15:
Thanks @weiranzhao. I am also very busy with other projects this week so it may be a few days before I can try this.

My data only has 2 classes. I assume I don't need to use 20 when retraining like you did?

Also, are you saying we can reuse the aiy object detection code to interpret the bounding box encodings, you just haven't tried that yet?

[16 more replies]

closed 2018-05-30 by sainttelant