google/aiyprojects-raspbian #316: No play back recording on Google Voice Kit

melzoghbi, 2018-03-20:
Hi,

I have a new google voice kit and when i try to check the voice checkpoint (speaker and mic), i got the speaker working but i can't hear my recording. This is my second kit which has the same issue. The magpi does not contain any instructions on how to resolve this issue? 

I do not think it is a hardware problem since this is my second board. 

[image]

Any though on fixing this would be helpful since i need to use the mic on this board.

Thanks.

drigz, 2018-03-21:
We haven't looked in to this in detail, but from other reports it seems to be something to do with apt upgrades and pulseaudio. You may be able to find more info in other issues. Older images such as 2018-01-03 do not appear to have this issue.

melzoghbi, 2018-03-21:
@drigz any workaround or a fix for the current stretch image? I could not find any resolution on other threads unless you have something you can point me to it.

anthyava, 2018-03-23:
Hi, this problem is relative at last kernel update, because before update my raspberrry (sudo apt-get update && sudo apt-get upgrade) workerd properly. After updated the mic don't work

melzoghbi, 2018-03-23:
@anthyava yes we are stuck on this!

PeterMalkin, 2018-03-23:
We are working on a fix. But meanwhile, try to grab the original image and do not update? Sound should work

anthyava, 2018-03-23:
@PeterMalkin Yes Yes suond work properly

drigz, 2018-03-26:
Thanks for the report!

As @PeterMalkin said, you'll need to use the 2018-01-03 image or reflash 2018-02-21 and avoid running `apt-get upgrade`. Alternatively, you can try the steps in https://github.com/google/aiyprojects-raspbian/issues/313#issuecomment-376046952.

I'm closing this issue as a duplicate - please continue discussion on #307.

closed 2018-03-26 by drigz