google/aiyprojects-raspbian #317: MIC access for python container app in YIA Voice Kit

melzoghbi, 2018-03-23:
I am using Google Voice Kit on a Raspberry PI 3. I created a docker image of my python app. When i run my app inside a container, i get this error when trying to record from the mic:

**FileNotFoundError: [Errno 2] No such file or directory: 'arecord'**

Here is the actual line that throws the exception in recording.py:

 self._arecord = subprocess.Popen(self._cmd, stdout=subprocess.PIPE)

This line is called from recordSound method in recording.py file:

```
def recordSound(duration = 1, sr = AUDIO_SAMPLE_RATE_HZ):
    recorder = Recorder(sample_rate_hz = sr)
    dumper = assembleArray(duration)
    with recorder, dumper:
        recorder.add_processor(dumper)
        while not dumper.is_done():
            time.sleep(0.1)        # This is the line that throws the above exception

    byt_buffer_lst = dumper.byte_buffer    
    byte_buffer = b''.join(byt_buffer_lst)    
    dat = np.frombuffer(byte_buffer, np.int16)  
    ## return sample rate in tuple to mimic wavfile read    
    return((sr, dat))
```
Any idea how this driver issue could be fixed inside a docker container?

Is it related to configuring ALSA loopback? i am looking for direction to share my mic inside my container app since i think this is the main issue.

PeterMalkin, 2018-03-23:
I think you might have messed something up when making your docker container.
arecord is a default alsa utility that is present on any raspbian by default along with aplay and alsamixer. It is used (you guessed it) for recording audio. If you dont have it - it means it is missing from your container.

drigz, 2018-03-26:
We don't support running the tool inside docker, so we're going to close this issue.

If you do need to run inside docker, you'll need to install `alsa-utils` inside the container and allow the container to access the sound devices on the host. Some relevant info here: https://github.com/danielchalef/jasper-docker

closed 2018-03-26 by drigz