google/aiyprojects-raspbian #318: Voice Kit stops reproducing answer after a random interval with grpc example

ThePiLocator, 2018-03-27:
Audio is working fine when i play some song on youtube, but as soon as i use the example assistant_grpc_demo.py, if the answer of the assistant is longer than 5-10 seconds it stops reproducing 
audio without any error, then the conversation continue as if the sound was heared correctly.
I tried to reinstall the OS, nothing changed. 
i've a Raspberry pi 2 model B

drigz, 2018-03-28:
This looks like the same issue as #297. Could you try the following workaround?

This seems to be caused by the PulseAudio suspend-on-idle module.  Please try disabling it and restarting PulseAudio:

```
sudo sed -i -e "s/^load-module module-suspend-on-idle/#load-module module-suspend-on-idle/" /etc/pulse/default.pa
pulseaudio --kill
pulseaudio --start
```

ThePiLocator, 2018-03-28:
Works like a charm! Thank you!

drigz, 2018-03-28:
Nice! Let's close this issue, and we can close #297 once we've released an image that fixes the issue.

closed 2018-03-28 by drigz