google/aiyprojects-raspbian #319: Completely remove google audio drivers from RPI3

melzoghbi, 2018-03-27:
Hi,

I was using Voice Hat board for working with google assistant sdk on my RPI3. Now, i removed the voice hat hardware board and installed respeaker board. I am not able to reset my audio driver so i can install respeaker driver.  

Is there reference document that someone can point me to so i can install other audio drivers on RPI3?

Thanks.

drigz, 2018-03-28:
Hey @melzoghbi, thanks for the feedback. I'm not familiar with ReSpeaker, so I'll need some clarification: Could you point to the exact ReSpeaker hardware you're using, and explain exactly how you're trying to install the audio driver?

I expect you'll need to remove the AIY ALSA config, which is stored at /etc/asound.conf and ~/.asoundrc, and replace it with config suitable for the ReSpeaker.

How was the AIY driver installed? Are you using the AIY SD card image or did you follow HACKING.md? It may be easier to start from the vanilla RPi SD card image.

melzoghbi, 2018-03-28:
Hi @drigz 

Here is the details of respeaker board on PI: https://github.com/SeeedDocument/ReSpeaker-4-Mic-Array-for-Raspberry-Pi/blob/master/ReSpeaker-4-Mic-Array-for-Raspberry-Pi.md

I tried but failed to completely remove AIY drivers from the OS so i started from scratch with a new raspbian image.

drigz, 2018-03-29:
Thanks for info - glad you found another approach.

I'll leave the issue open in case anyone else encounters this.

dmitriykovalev, 2018-08-16:
We switched to PulseAudio, ALSA configs are not relevant anymore.

closed 2018-08-16 by dmitriykovalev