google/aiyprojects-raspbian #321: Forbidden for Url error

nickdex, 2018-03-29:
I followed steps in guide and when I ran `src/examples/voice/assistant_library_demo.py` it opened up browser for permission, it initialised credentials in cach directory and threw this exception.
```
Traceback (most recent call last):
  File "src/examples/voice/assistant_library_demo.py", line 73, in <module>
    main()
  File "src/examples/voice/assistant_library_demo.py", line 67, in main
    with Assistant(credentials) as assistant:
  File "/opt/aiy/projects-python/src/aiy/assistant/library.py", line 32, in __init__
    self._model_id = device_helpers.register_model_id(credentials)
  File "/opt/aiy/projects-python/src/aiy/assistant/device_helpers.py", line 93, in register_model_id
    r.raise_for_status()
  File "/home/pi/.local/lib/python3.5/site-packages/requests/models.py", line 928, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://embeddedassistant.googleapis.com/v1alpha2/projects/jarvis-133713/deviceModels/jarvis-133713-voice-kit
```

drigz, 2018-03-29:
Did you use the same Google account to create the cloud project and to log in from the Raspberry Pi?

If not, you'll need to use the same account, or to make the account running on the Pi a project editor, so that it can register itself. You can add the editor rights here: https://console.cloud.google.com/iam-admin/iam.

nickdex, 2018-03-30:
Yes, I used the same account. I removed .cache directory and tried again. I got same error (as above) after following log line.
```
[2018-03-30 15:05:23,065] INFO:root:OAuth credentials initialized: /home/pi/.cache/voice-recognizer/assistant_credentials.json
```

aJoohongKim, 2018-04-03:
I am getting the same error. In my case, I already set up voice recognition app in one google account and I want to change the account with another. I guess the problem is the consent step. It thinks it already have consent from previous account and send request but it has never consented.
My question is where is stores this consent. I guess it's my google account and I totally changed assistant.json.
Help me~~

drigz, 2018-04-06:
@proppy Any idea what could be causing this?

proppy, 2018-04-06:
can you confirm that account used to register the device is a viewer of the project where the device model was created?

drigz, 2018-04-06:
@proppy, I asked "Did you use the same Google account to create the cloud project and to log in from the Raspberry Pi?" and @nickdex responded that they did. Presumably that would guarantee that the account is a project owner, right?

proppy, 2018-04-06:
@drigz I would double check in the developer console for project `jarvis-133713` just to be sure

proppy, 2018-04-06:
Also it seems that the failure is on `register_model_id`, in order to be able to register a new model, you need to be a `Owner` or `Editor` of the project.

JamesNitsch, 2018-04-07:
Same issue here. Same account to create & log in. Additionally cleared the cache & re-downloaded the credentials to no avail.

mihilie, 2018-04-08:
I've run into the same issue - same symptoms - but the problem was that, somehow the APIs and services got disabled - I'd swear I didn't do it, but with all the going back and forth, who knows? After I went through “Enable APIS and Services” again, it started to work. The only tweak I would do on the client side is to delete the credentials .cache directory.

I hope this helps.

JamesNitsch, 2018-04-08:
@mihilie thank you for your input - seems I had the exact same issue. I'm not sure when or how the assistant API got disabled after I set it up, but it all works fine for me now. Didn't even need to clear the credentials. Hopefully this is the root of the problem for others in this thread.

yevx1, 2018-04-09:
Hi, have the same problem. switched accounts, checked if API is enabled and it is. but i don't seem to find how to delete the credentials cache .. i tried the command: # sync; echo 2 > /proc/sys/vm/drop_caches but doesn't seem to do anything. anyone any idea what to do ? i work as a freelancer on this project. the setup at my home was working, today i try to set everything up with the company credentials but won't work

dmitriykovalev, 2018-10-15:
Please make sure you can run this command using `sudo`. All credentials should be copied to the `/root` folder.

Nojahhh, 2018-10-19:
Hi Dmitri. Yes, my current command in rc.local looks like this:
`sudo -H -u pi python3 assistant_library_with_local_commands_demo2.py &`
I've tried to run it without -H and -user but get the same results.

What do you mean when you say "all credentials"? I've created a client ID through Google Cloud Platform and placed it in assistant.json at /root, is there another credential-file I'm missing?
I've even copied my working assistant.json from my user-folder to /root without success.

manoj7410, 2018-10-22:
@Nojahhh Did you try to run your application manually on terminal on rootlevel? Did it work ?

Can you share what happens when you perform below steps:

- `sudo su`
- `cp /home/pi/assistant.json /root`
- `cp /home/pi/AIY-projects-python/src/examples/voice/assistant_library_with_local_commands_demo.py /root`
- `cd /root`
- `./assistant_library_with_local_commands_demo.py`

I was able to run demo with above steps. Haven't tried it with rc.local yet.

closed 2019-11-14 by dmitriykovalev

ralgara, 2019-11-17:
@manoj7410 I tried your exact steps above to no avail. Any other suggestions? Thanks.