google/aiyprojects-raspbian #322: What are the differences between Asssistant.GRPC and Assistant.library?

ThePiLocator, 2018-03-29:
Maybe this is not the right place for this kind of question, but i need to clarify my ideas and this community is amazing. Hope someone can answer my 4 questions down here:

1) What are the key differences between aiy.assistant.grpc and aiy.assistant.library ?
2) Why does the first one works with italian language and why the second one doesn't? Am i doing something wrong or is this just not supported yet?
3) Should the "Okay Google" hotword work also with the grpc version?
4) Does exist a difference also in terms of time performances? 

Thank you.

drigz, 2018-04-12:
> Maybe this is not the right place for this kind of question

The best places for such questions would be the [AIY Forums](https://www.raspberrypi.org/forums/viewforum.php?f=114) or [StackOverflow](https://stackoverflow.com/questions/tagged/google-aiy). However, I'll try to answer these.  

> What are the key differences between aiy.assistant.grpc and aiy.assistant.library ?

aiy.assistant.grpc uses the Google Assistant Service, and aiy.assistant.library uses the Google Assistant Library. There's some more information here: https://developers.google.com/assistant/sdk/overview

> Why does the first one works with italian language and why the second one doesn't? Am i doing something wrong or is this just not supported yet?

For the Library, you need to change the language in the Google Assistant app. More information on [this StackOverflow answer](https://stackoverflow.com/questions/48017507/how-do-i-change-the-language-locale-in-the-google-assistant-sdk).

> Should the "Okay Google" hotword work also with the grpc version?

No, it doesn't.

> Does exist a difference also in terms of time performances?

Not that I'm aware of.

Hope that helps! Let me know if anything's still unclear - if not, let's close the issue.

ThePiLocator, 2018-04-15:
Everything is perfectly clear! Thank you 👍

closed 2018-04-15 by ThePiLocator