google/aiyprojects-raspbian #324: Why response is slow in cloudspeech demo?

varad1311, 2018-03-30:
Hi! I'm using cloudspeech for home automation ( to control various electrical appliances like lights,fan etc). When I give instruction like "turn on lights", it takes much time to execute that operation and make the corresponding GPIO pin HIGH. Why this happens? Is it due to slow internet connection? I'm using my mobiles hotspot to connect raspberry pi.

drigz, 2018-04-06:
I'd expect it would be limited by the network connection. It just uploads the audio Google's servers, and the action executes as soon as the transcription text is returned.

How slow is slow? eg how many seconds from command to response?

varad1311, 2018-04-06:
Hi! Sometimes it is very slow. It takes 10 to 15 seconds to respond.
Sometimes the program crashes.

dmitriykovalev, 2018-08-16:
@varad1311 Do you still have this issue even with the latest SD card image?

nunziofiore, 2018-09-17:
without any update to the code I have the same issue. Last week it was ok and I have a quick response, today I have 15/20seconds to wait. No problem on my wifi speed.

dmitriykovalev, 2018-09-17:
@nunziofiore Can you try check Cloud Speech web control panel and see if everything is fine there?

closed 2019-11-14 by dmitriykovalev