google/aiyprojects-raspbian #325: Volume level google assistant response

divx118, 2018-04-01:
Is there a place in the code where we can adjust the volume level of the assistant response, because it is very low.
I know I can up the volume level of alsa, but that would be way to loud for when I want to play music with for example mpd (music player daemon).
Or is this something I should request/ask at the google api repo.

Off topic: Can we still make pull requests here since it seems everything is first prepared on an internal repo? If so, then for now we don't exactly know on which issues there is work being done on the internal repo. Could that be made more transparent? So that we don't need to invest free time into doing something that is all ready being handled. Thanks for any clarification on this.

mpember, 2018-04-09:
I find that the volume of the voice response does sometimes become very low.  If I say a command like "turn up the volume" or "set the volume to maximum", it appears to resolve it.

divx118, 2018-04-22:
@mpember Thanks, yeah that solved it, at least the volume is higher now. Still it would be nice to have some adjustment possible, so I could turn the range a bit up. Now at max it is still lower then my music. I know, I can turn on volume control in my mpd server, but that means more cpu power gets used.

NorioFujii, 2018-05-10:
"Unify volume between assistant_{grpc,library}_demo" for src/aiy/audio.py 's update is invalid.
line113 db_range = -60.0 - (-60.0 * (volume / 100.0))   (Fault)
             db_range = -6.6  - (-6.6   * (volume / 100.0))    (Right)