google/aiyprojects-raspbian #327: Keyboard and mouse are not working with Vision image

kodukulav, 2018-04-05:
Hi @dmitriykovalev 

I assembled the vision kit as per the directions given in the website. After that, I flashed the provided image onto a SD card and hooked the kit to a monitor. I also, inserted my keyboard and mouse dongle to the microUSB port on the Pi through a microUSB to female USB OTG cable. With all of this, I'm not able to use my keyboard and mouse after the system boots up.

I looked up online and added modules_load = g_hid in the cmdline.txt under the boot folder. But this didn't fix the problem.

Is there any extra setting that we need to add to make the keyboard/mouse work?

Thanks,
Venky

dmitriykovalev, 2018-04-17:
@kodukulav 

SD card image should work fine with USB keyboard and mouse, you don't need to modify cmdline.txt for that. Try to download the latest image (there was an update recently) and make sure you use good power supply (e.g. wall power adapter) and good USB hub (try to plug in keyboard and mouse separately without a hub to troubleshoot).

dmitriykovalev, 2018-04-26:
Please reopen if you still have this problem.

closed 2018-04-26 by dmitriykovalev