google/aiyprojects-raspbian #328: Vision_Bonnet_compiler Link Broken?

leecarraher, 2018-04-06:
The vision_bonnet_compiler download link
https://dl.google.com/dl/aiyprojects/vision/bonnet_model_compiler_2017_12_13.tgz
appears to be broken. Does anyone have an alternate link?

kodukulav, 2018-04-10:
Yes, even I'm not able to download at the link given in the website.

sainttelant, 2018-04-13:
as i know, Google teams are trying to update the compiler tool to solve some compiler issues,plz refer to the issue #314

chadwallacehart, 2018-04-20:
Link is broken for me too. I found a copy I downloaded a while ago: [](https://www.dropbox.com/s/sedgmjaf9ng2mnd/bonnet_model_compiler_2017_12_13.tgz?dl=0)

kodukulav, 2018-04-20:
@chadwallacehart :Thanks for sharing!

weiranzhao, 2018-04-27:
@leecarraher @kodukulav @chadwallacehart @sainttelant @minakhan01 
Folks, we just released an updated version of the compiler. It fixed some issues people are experiencing when compiler customized TF model. It also added an option to print more debug information when the tool fails.
https://dl.google.com/dl/aiyprojects/vision/bonnet_model_compiler_latest.tgz

leecarraher, 2018-04-28:
@chadwallacehart  @weiranzhao Thank you for the links.

dmitriykovalev, 2018-05-02:
Closing, compiler download link is working now.

closed 2018-05-02 by dmitriykovalev