google/aiyprojects-raspbian #331: Documentation for Voice Bonnet pinouts

htruong, 2018-04-18:
Hi,

I have the Voice AIY kit v2 on hand, but I have found no documentation for the pinouts for the Voice Bonnet anywhere. I have found the v1 hardware image, though, but that gives no hints for where the v2 docs should be.

Thanks.

tei-chino, 2018-04-24:
Hi, 
On https://aiyprojects.withgoogle.com/voice/, The HARDWARE EXTENSIONS is still showing the v.1 voice board. Where can I find the the new Voice Bonnet hardware information. For example, the GPIO pins and its drive current limit.
Thanks.

dawzrd, 2018-05-06:
yeah I am curious about this too...
extend the `hackability'

sfjadi2010, 2018-05-14:
Hi, I'm curious to know is anyone come across latest voice bonnet documentation. Any help will be appreciated.
Thanks

dmitriykovalev, 2018-05-14:
We are working on a website fix. VoiceBonnet has exactly the same GPIOs as VisionBonnet, you can take a look here: https://aiyprojects.withgoogle.com/vision#makers-guide--gpio-diagrams 

Python code examples can be found here (they work for both VisionBonnet and VoiceBonnet): https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/src/examples/vision/gpiozero

tei-chino, 2018-05-14:
Thanks you  for updating the GPIO information for VoiceBonnet.

tei-chino, 2018-05-15:
Hi Dear Sir,
Would you like tell me how about the drive limits for these GPIOs(A, B, C, D).
Are they same with HAT board (25mA)?
Thanks.

SK-Github, 2018-07-13:
Thank you @dmitriykovalev for updating the header pins info (A, B, C, D) here for the AIY version 2 kit: https://aiyprojects.withgoogle.com/vision#makers-guide--gpio-diagrams 
Much appreciated! Is is possible to get information of the corresponding pin numbers on the Pi Zero WH for the voice HAT used in AIY Voice kit version 2, similar to the one available for version 1 here: https://pinout.xyz/pinout/voice_hat
Thank you!

dmitriykovalev, 2018-07-14:
@SK-Github We'll try to put Voice V2 info to pinout.xyz (it's pretty similar to V1). For now please use the following:

```
       3.3V --> 1    2 <-- 5V
    I2C_SDA --> 3    4 <-- 5V
    I2C_SCL --> 5    6 <-- GND
                7    8
        GND --> 9   10
                11  12 <-- I2S_BCLK
                13  14 <-- GND
                15  16 <-- BUTTON_GPIO (GPIO_23)
       3.3V --> 17  18
                19  20 <-- GND
                21  22
                23  24
        GND --> 25  26
     ID_SDA --> 27  28 <-- ID_SCL
                29  30 <-- GND
                31  32
  I2S_LRCLK --> 33  34 <-- GND
                35  36
                37  38 <-- I2S_DIN
        GND --> 39  40 <-- I2S_DOUT
```

SK-Github, 2018-07-14:
@dmitriykovalev Thank you so much for the quick response on the button pin # information! I really appreciate it! May I assume the following for the header pins and onboard LEDs on the voice HAT (based on code here: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/pins.py ):

AIY voice kit version 2:  <————>  BCM GPIO pin # on Pi Zero WH
(Voice HAT)                                         

		PIN_A         <————>    BCM 2
		PIN_B         <————>    BCM 3
		PIN_C         <————>    BCM 8
		PIN_D         <————>    BCM 9

		LED_1         <————>    BCM 13
		LED_2         <————>    BCM 14

Thank you so much!

dmitriykovalev, 2018-07-14:
@SK-Github No, your mapping is incorrect. Pins A, B, C, D, and both LEDs are connected to the separate microcontroller on the bonnet board. There is a separate Linux driver for it and `pins.py` communicates with the driver via sysfs.

SK-Github, 2018-07-14:
@dmitriykovalev Thank you so much for the quick reply!

dmitriykovalev, 2018-07-15:
@SK-Github No problem, all this information will be added to the doc files in the next release.

dmitriykovalev, 2018-07-18:
@SK-Github Thank you for very detailed description! You are correct, pull up/down resistor is required in this case. I guess we need to specify this somewhere in the docs, so I'm putting "enhancement" label to this issue. And let me also double check gpiozero API contract again.

SK-Github, 2018-07-18:
@dmitriykovalev With much gratitude for your prompt response and advice. Thank you!

SK-Github, 2018-07-19:
@dmitriykovalev 

The behaviors of the APIs led.on( ) and led.off( ) for the two LEDs on the voice bonnet:
- Green LED connected to aiy.pins.LED_1
-   Red LED connected to aiy.pins.LED_2

are exactly as I would expect from the gpiozero documentation (section 13.1 here: https://gpiozero.readthedocs.io/en/stable/api_output.html)

However, if I hook up an external LED to PIN_A of the GPIO header, led.on( ) and led.off( ) behave exactly in the opposite manner compared to the behavior described for their gpiozero API. I wired the external LED to PIN_A as is specified in the gpiozero documentation: “Connect the cathode (short leg, flat side) of the LED to a ground pin; connect the anode (longer leg) to a limiting resistor; connect the other side of the limiting resistor to a GPIO pin (the limiting resistor can be placed either side of the LED).” I found the same behavior across 3 different pins that I tested (PIN_A, PIN_B, PIN_C). I did not test on PIN_D as I had a button hooked up to that pin to control the LED :-)

Please let me know if there is an expected difference in behavior between led.on( ) and led.off( ) for the onboard LEDs on the voice bonnet vs external LEDs that can be hooked up to the GPIO header pins on the kit. Thank you very much for your help!

SK-Github, 2018-07-19:
@dmitriykovalev 
I am hoping to control a micro servo using the GPIO header pins on AIY voice kit V2 and the code here (https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/gpiozero/servo_example.py) based on your comment on this thread on May 14. I get the following errors when I run it in Python3 from dev terminal on the Pi desktop. Can you please let me know what needs to be fixed to interface with a servo? Thank you for your help!

pi@raspberrypi:~/AIY-voice-kit-python/src/examples/voice/gpiozero_sk $ python3 AIY15_ctrl_1_servo.py
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/pins.py", line 120, in export
    with open(self.root_path('export'), 'w') as export:
PermissionError: [Errno 13] Permission denied: '/sys/class/pwm/pwmchip0/export'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "AIY15_ctrl_1_servo.py", line 13, in <module>
    simple_servo = Servo(PIN_A)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1012, in __init__
    pin, frequency=int(1 / frame_width), pin_factory=pin_factory
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 331, in __init__
    self.pin.frequency = frequency
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 300, in <lambda>
    lambda self, value: self._set_frequency(value),
  File "/opt/aiy/projects-python/src/aiy/pins.py", line 573, in _set_frequency
    self._enable_pwm()
  File "/opt/aiy/projects-python/src/aiy/pins.py", line 502, in _enable_pwm
    self.pwm_pin.open()
  File "/opt/aiy/projects-python/src/aiy/pins.py", line 343, in open
    super(SysFsPwmPin, self).open()
  File "/opt/aiy/projects-python/src/aiy/pins.py", line 130, in open
    self.export()
  File "/opt/aiy/projects-python/src/aiy/pins.py", line 123, in export
    raise GPIOPinInUse('Pin already in use')
gpiozero.exc.GPIOPinInUse: Pin already in use

dmitriykovalev, 2018-07-20:
@SK-Github Regarding `on()` and `off()` for external LEDs — this behavior is a bug. It is already fixed internally and updated code will be released with the new SD card image.

SK-Github, 2018-07-24:
@dmitriykovalev  Thank you so much for the info on the LEDs on the voice bonnet! Any update on the code that interfaces with the servos (description above): (https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/gpiozero/servo_example.py)? Thank you!

manoj7410, 2018-07-25:
@SK-Github Servo APIs can be seen in detail here : https://gpiozero.readthedocs.io/en/stable/api_output.html#servo

SK-Github, 2018-07-25:
@manoj7410 Thank you for the link! I believe the code here (https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/gpiozero/servo_example.py) follows the gpiozero servo documentation that you linked above. My question is on why the example code in **servo_example.py** does not run with Python3. I am copying the errors below. for your reference. Can you please let me know if you are able to run it successfully on AIY Voice kit V2? I made sure to connect the 3 wires of the servo motor correctly (5 Volt, Ground and GPIO pin). Thank you for your advice!

Traceback (most recent call last):
File "**servo_example.py**", line 14, in 
simple_servo = Servo(PIN_A)
File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in call
self = super(GPIOMeta, cls).call(*args, **kwargs)
File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1012, in init
pin, frequency=int(1 / frame_width), pin_factory=pin_factory
File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in call
self = super(GPIOMeta, cls).call(*args, **kwargs)
File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 331, in init
self.pin.frequency = frequency
File "/usr/lib/python3/dist-packages/gpiozero/pins/init.py", line 300, in 
lambda self, value: self._set_frequency(value),
File "/opt/aiy/projects-python/src/aiy/pins.py", line 573, in _set_frequency
self._enable_pwm()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 502, in _enable_pwm
self.pwm_pin.open()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 343, in open
super(SysFsPwmPin, self).open()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 130, in open
self.export()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 123, in export
raise GPIOPinInUse('Pin already in use')
gpiozero.exc.GPIOPinInUse: Pin already in use

manoj7410, 2018-07-25:
@SK-Github I just tried to run Servos on Voice V2 and Vision V1.1 using sdcard image 04-13 and it is working fine. Are you trying with the latest image ? Or did you make any changes in servo_example.py ?
Your error is saying "Pin already in use". Not sure but did you configure this pin for some other purpose earlier ?

SK-Github, 2018-07-25:
@manoj7410 Thank you so much for verifying this on your end and for the feedback! The issue was that I was not running with sudo. When I run with sudo, I do not run into the error with pwmchip0 highlighted below.  However, there are some runs when things seem to work just fine without having to use sudo, but it is a hit or a miss. But, with sudo, things work consistently. I thought to try sudo after reading this: https://www.udoo.org/forum/threads/pwm-export-permission-denied.3507/

Thank you again!

Traceback (most recent call last):
File "/opt/aiy/projects-python/src/aiy/pins.py", line 120, in export
with open(self.root_path('export'), 'w') as export:
**PermissionError: [Errno 13] Permission denied: '/sys/class/pwm/pwmchip0/export'**

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "AIY15_ctrl_1_servo.py", line 13, in 
simple_servo = Servo(PIN_A)
File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in call
self = super(GPIOMeta, cls).call(*args, **kwargs)
File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1012, in init
pin, frequency=int(1 / frame_width), pin_factory=pin_factory
File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in call
self = super(GPIOMeta, cls).call(*args, **kwargs)
File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 331, in init
self.pin.frequency = frequency
File "/usr/lib/python3/dist-packages/gpiozero/pins/init.py", line 300, in 
lambda self, value: self._set_frequency(value),
File "/opt/aiy/projects-python/src/aiy/pins.py", line 573, in _set_frequency
self._enable_pwm()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 502, in _enable_pwm
self.pwm_pin.open()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 343, in open
super(SysFsPwmPin, self).open()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 130, in open
self.export()
File "/opt/aiy/projects-python/src/aiy/pins.py", line 123, in export
raise GPIOPinInUse('Pin already in use')
gpiozero.exc.GPIOPinInUse: Pin already in use

SK-Github, 2018-07-26:
@manoj7410 I am facing a new issue with AIY Voice kit V2. I am using the 04/13/2018 version of the SD card image now and using ssh (just like your set up). I am trying to control an LED (on PIN_A) and a servo (on PIN_B) using voice commands uttered after the press of a button (connected to PIN_D). If I do not use sudo, I get the same issue as before (PWM permissions). So I tried to run it with sudo. This time I get a different error: **File /root/cloud_speech.json was not found.**

I made sure that the assistant.json and the cloud_speech.json files are in the /home/pi directory as usual. I am puzzled as to why I get the error of **File /root/cloud_speech.json was not found.** Please let me know if you have any suggestions. Thank you!

I am attaching the following for your reference:
~ My code
~ Error log without sudo (PWM permission error)
~ Error log with sudo (cloud_speech.json error)
~ Software version info of the image on my SD card

[Control_Servo.zip](https://github.com/google/aiyprojects-raspbian/files/2229964/Control_Servo.zip)

manoj7410, 2018-07-26:
@SK-Github Below is my understanding and please correct me if it is incorrect:
- You burned your sdcard with a fresh image (version 04-13)
- You did not make any changes into the environment of the AIY OS after booting it.
- You ssh into the kit.
- You executed cloud_speech or servo_example without sudo and ran into issue.
- Then you tried with sudo and it gave a different error (Which is because application is looking for cloud_speech in /root instead of /home/pi).

My kit's observation:
- No changes in the environment.
- I am able to run servo_example with and without sudo.
- To run assistant_grpc using sudo we need to put assistant.json into /root because $HOME for root (sudo) is /root and not /home/pi (You can check this using "sudo printenv" command. Working with and without sudo)

I haven't tried cloud_speech since it requires billing and I haven't configured that into my account yet. But you can try to copy the cloud_speech.json into /root to make it work (Although not recommended to use sudo).

I think there is some environment change done by you and because of that you are running into user access related issues.

I saw some people facing PWM permission error but that was long back and due to udev related changes into the kernel. I am trying to reproduce this here but couldn't see that yet.

SK-Github, 2018-07-26:
@manoj7410 Thank you so much for the detailed and useful response! I really appreciate all of your advice and suggestions!

First, my response to the confirmation of my setup:
- You burned your sdcard with a fresh image (version 04-13): **Yes**
- You did not make any changes into the environment of the AIY OS after booting it:  **Yes, except** this...I copied leds.py and pins.py from here (https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/src/aiy)  into the aiy directory, otherwise the code fails compilation (does not recognize PIN_A etc)
- You ssh into the kit: **Yes**
- You executed cloud_speech or servo_example without sudo and ran into issue: I tried **servo_example**
Then you tried with sudo and it gave a different error (Which is because application is looking for cloud_speech in /root instead of /home/pi): **You are right!**

Now....update since last night...

~ Today for the first time, I could directly ssh into the Pi and got my code (in my previous post) to run **without sudo**. I was able to do this 10 times consecutively and things worked great and the LEDs and servo behaved exactly as I would expect in response to the appropriate voice commands in my code. I scratched my head because I did not know what changed on my end to make the PWM permission error problem go away ! To try to recreate the problem that I posted most recently above, I opened a dev terminal on the Pi (by attaching a monitor) and ran the same code and boom...the PWM permission error showed up (not just on the dev terminal now, but even when I run it via ssh subsequently)

~ That made me think....ok! perhaps some environment variables are being set to weird stuff when I open a dev terminal on the Pi. I can work around that.....I can always just ssh into the Pi. Alas!...it does not always work. After I rebooted the Pi and directly sshed into the Pi (after those 10 successful runs that I mention above), I still ran into the same PWM permission error. 

~ I tried starting from scratch again (re-imaged the SD card with the 04/13 version, copied the leds.py, pins.py files into the aiy directory, and directly sshed into the Pi, **without** opening a regular or dev terminal directly on the Pi) and the same PWM permission error showed up again. **Is there any way that you could send me your relevant environment variables so that I can double check with mine, if that is indeed the source of all this mystery?**

~ On another note, it was not at all straight forward to put the cloud_speech.json file into the /root directory without running into permission issues. Given that, and your recommendation that I should be able to run the servo code without sudo, I abandoned the task of trying to run it with sudo. I'd rather wait, so that we figure what in my environment variables are different from yours, than mess around with sudo permissions on the /root directory.

Thank you so much for your advice and I look forward to your suggestions!

closed 2018-08-16 by dmitriykovalev

yooniscali, 2020-04-27:
Hi there, i am doing robotic arm voice control, i am using google aiy voice kit v2, i have 6 servo motors to control the robotic arm via the voice kit. I want to use servo motor driver however i can't find the pins for the driver on the voice bonnet, for example the A,B, C, D pins which one is sda or scl because i want to connect straight into the voice bonnet gpio pins, any advise is appreciated