google/aiyprojects-raspbian #334: Response stream does not actually use deadline

jaymwong, 2018-04-20:
See : https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/_apis/_speech.py#L327:L329

As a result, this defaults to I'm guessing 185 seconds (the GRPC default), which is probably undesirable...

jaymwong, 2018-04-20:
The reason I point this out is that this fix should allow users to exit quickly on failures. Sometimes my response stream hangs and it takes several minutes to exit out with one of the following errors, 
```grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with (StatusCode.DEADLINE_EXCEEDED, Deadline Exceeded)>

```

```error {
  code: 3
  message: "Client GRPC deadline too short. Should be at least: 3 * audio-duration + 5 seconds. Current deadline is: 132.99840800778475 second(s). Required at least: 139 second(s)."
}```

drigz, 2018-04-24:
This is an intentional choice, to allow users to use the full 60 seconds allowed for a query. However, it's possible that most queries are much shorter so we don't need such a large deadline.

@proppy, do you have any data on the durations of queries used on the Assistant Service?