google/aiyprojects-raspbian #335: Error creating a binaryproto for retrained/rounded graph

minakhan01, 2018-04-22:
Hi!

I retrained mobilenet_0.50_160 using instructions from to Peter Waden's blog and code labs: https://codelabs.developers.google.com/codelabs/tensorflow-for-poets/ and https://codelabs.developers.google.com/codelabs/tensorflow-for-poets-2/

The entire repo is here: https://github.com/minakhan01/LabMembersFaceRecognition and the .pb files are as follows: 
Retrained graph: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/tf_files/retrained_graph.pb
Optimized graph: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/tf_files/optimized_graph.pb
Rounded graph: 
https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/tf_files/rounded_graph.pb

I'm trying to convert rounded_graph.bp to rounded_graph.binary proto using the following command:
./bonnet_model_compiler.par \
    --frozen_graph_path=./rounded_graph.pb \
    --output_graph_path=./rounded_graph.binaryproto \
    --input_tensor_name="input" \
    --output_tensor_names="final_result" \
    --input_tensor_size=160

But I keep getting this error: Input model uses unsupported dimension in at least ONE tf.matmul. USED unsupported_K_value = 1001

I would really appreciate your help!

Thanks :)

minakhan01, 2018-04-23:
Fixed it by padding the 1001 dimension tensor. still getting the unexpected error, but at least this one is fixed:

[image]

minakhan01, 2018-04-23:
I'm trying to figure out why I'm getting the 'unexpected error' when I compile my .pb graph using the bonnet compiler. My suspicion is that it has something to do with the placeholder_with_default we are using for the BottleneckInputPlaceholder. Does anyone know how I can avoid using placeholder_with_default in my transfer learning code or remove the placeholder node later on using optimize_for_inference?

Thanks!

[image]

minakhan01, 2018-04-23:
update: I was able to remove the placeholder node

here the original protobuf file
a. original from training (optimized for inference and quantized): https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/tf_files/rounded_graph.pb
b. pb a minus the placeholder node: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/tf_files/rounded_graph_without_dropout.pb

[image]

sainttelant, 2018-04-25:
@minakhan01 Hi, i am still confused about how to remove some specific nodes where you want in a trained model, how did you do that?

minakhan01, 2018-04-27:
@sainttelant you've to modify line # 16 and 1 for your respective nodes, but here's the file. take a looks: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/tf_files/remove.py

weiranzhao, 2018-04-27:
@minakhan01 There's a new version of model compiler, could you give that a try?

minakhan01, 2018-04-30:
@weiranzhao the compiler worked and I was able to create a binary proto file. However, the binaryprotos create the following error (same one when I compile my own model or the mobilenet_v1_160res_0.5_imagenet pb file). I have checked my code with the binaryproto files that you already have on the pi so I think the code is fine. I think the problem is with the binaryproto files from the compiler because even the compiled version of unmodified imagenet does not work. 

**Here are my compiled protobuf files in case you'd like to try it out:**
1. custom model: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/NEW_COPY_TO_PI/rounded_graph.binaryproto
2. imagenet: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/NEW_COPY_TO_PI/mobilenet_v1_160res_0.5_imagenet_own.binaryproto

**Here's the error:**
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 128, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, self._tbuf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./person_recognition_camera.py", line 77, in main
    for i, result in enumerate(inference.run()):
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 76, in run
    yield self._engine.camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 257, in camera_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 143, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./person_recognition_camera.py", line 81, in main
    print_classes(classes, args.num_objects)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 87, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 79, in close
    self._engine.stop_camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 263, in stop_camera_inference
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: StopCameraInference: Camera inference is not running.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./person_recognition_camera.py", line 87, in <module>
    main()
  File "./person_recognition_camera.py", line 83, in main
    camera.stop_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "

**Here are my files:**
1. person_recognition.py which I put in /home/pi/AIY-projects-python/src/aiy/vision/models/: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/NEW_COPY_TO_PI/person_recognition.py
2. person_recognition_camera.py which I put in pi@192.168.86.170:/home/pi/AIY-projects-python/src/examples/vision/: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/NEW_COPY_TO_PI/person_classification_camera.py

weiranzhao, 2018-04-30:
Is your SD card upgraded to the latest, i.e. Apr 2018 release?

minakhan01, 2018-04-30:
@weiranzhao yes, it is. I recently bought the device (Vision Kit 1.1) from your new April 2018 batch and haven't changed the SD card contents. Is there any other updates that I don't know about?

crobbiano, 2018-04-30:
@minakhan01 Thanks for some of the pioneering work.  I was able to retrain the provided mobilenet imagenet model, similar to how you have (pruning the bottleneck node from the graph), and get it to compile with the new compiler.  After loading the binary onto the bonnet and running with modified version of your scripts, I also get the same errors.

minakhan01, 2018-04-30:
@crobbiano team work! :)

Google is the one doing the pioneering work. thank you, @weiranzhao, for helping!

weiranzhao, 2018-04-30:
@minakhan01 @crobbiano 
The firmware that comes with Vision Kit 1.1 does not support Gemm operator with common dimension that is not a multiple of 8. And that's why when you used the old compiler, it is complaining 
"Input model uses unsupported dimension in at least ONE tf.matmul. USED unsupported_K_value = 1001"

But newer firmware supports it, and that's why the model compiles with latest compiler.

Could you update your SD card using the instructions here?
https://aiyprojects.withgoogle.com/help/#updating-your-sd-card

You can run 
dpkg -l aiy*
before and after you update your SD card to confirm SD card gets updated.

crobbiano, 2018-04-30:
@weiranzhao Will update my SD image tonight and report back.  Thanks!

crobbiano, 2018-05-01:
@minakhan01 @weiranzhao After updating the SD card it would appear that I can at least run with my compiled model.  I had to comment out the assertion checking the shape of the tensor == (1, 1, 1, 1001) to get some output as my graph is not conformal at this point.  

It correctly classifies blackground. hah :/ more to come as I dig in later this week.

weiranzhao, 2018-05-01:
@crobbiano Great job, thanks for the update!
You can also check 
src/examples/vision/mobilenet_based_classifier.py
It is written to run mobilenet based classifier, it takes model + class labels, so you don't have to hard code a CLASSES file.

crobbiano, 2018-05-01:
@weiranzhao can you recommend a method for automating the retaining and removal of the bottleneck node in preparation for the bonnet compiler? Possibly share a retraining script?

 I'm using a hacked retrain.py from the tensorflow python examples that loads a local frozen mobilenet instead of downloading it from tfhub. In doing the retaining, it places the bottleneck node between the mobilenet block and the final layer which ultimately needs to be manually removed for compatibility.

weiranzhao, 2018-05-01:
@crobbiano Not sure if you followed Pete's Poet tutorial or not. You can use the retraining script there to get the retrained frozen graph. Then feed the graph to bonnet compiler, it should work. I tried this before, it worked for me.

I'm not completely sure about the removal you are talking about. But if you want only part of the graph to run inference, you can do this with --input_tensor and --output_tensors flags, just need to specify the correct node names.

weiranzhao, 2018-05-01:
@crobbiano I just followed the TF poet tutorial again, I have no trouble converting the retrained_graph.pb with bonnet compiler.

minakhan01, 2018-05-04:
@weiranzhao , sorry for the late reply -- I'm finishing up my thesis. I updated my pi as per your instructions. this is the output I get:
[image]

But I'm still getting the same error as last time. Seems that it is not able to read the binaryproto. Would it be possible for you to try to run this one: https://github.com/minakhan01/LabMembersFaceRecognition/blob/master/NEW_COPY_TO_PI/rounded_graph.binaryproto

[image]

minakhan01, 2018-05-05:
Got it to work! A little caught up right now, but I'll post more details on how so that everyone else can use it too.

thank you so much, @weiranzhao ! you were incredibly helpful, and also your file src/examples/vision/mobilenet_based_classifier.py Couldn't have done it without you.

jkim0120, 2018-06-10:
Hey @minakhan01 -- I've been running into the same issues. I was wondering if you can share how you got it to work in the end?

dvillevald, 2018-06-24:
Hi @minakhan01 -- I am having same issues. Can you share how you resolve them?

weiranzhao, 2018-06-25:
@dvillevald 
We put up instructions and short tutorial on our website on deploying customized network. Could you check that?
https://aiyprojects.withgoogle.com/vision/#makers-guide--tensorflow-model-compiler

BigVulcanDeal, 2018-06-27:
I found this thread to be really helpful .. thanks.  

For me, the key turned out to be ensuring that the SD card used the April image.  Running "update" and "upgrade" was not helpful .. in fact, it seemed to overwrite some stuff that was needed for successful operation. I had to download the new SD image.

The updated instructions might benefit from a mention of the importance of having the latest image, and some tests to ensure that everything is "in order".  For example, the screen-shot that Mina Kahn posted above was my clue that I needed to burn a new SD (copy of said screenshot included below).

Thanks again!

[image]

ritwikbiswas, 2018-07-07:
I had a pretty basic question. I've used transfer learning on Mobilenet for a simple male and female face classification model. I'm attempting to compile the model to run on the vision kit (using the bonnet model compiler). How do I determine what to use for the `--output_tensor_names` command line argument? My full call to the compiler looks something like this:

```
 ./bonnet_model_compiler.par \
  --frozen_graph_path=<graph path> \
  --output_graph_path=retrained_graph.binaryproto \
  --input_tensor_name="input" \
  --output_tensor_names=<???> \
  --input_tensor_size=160
```
Any direction or help would be appreciated. Thanks!

dvillevald, 2018-07-07:
It is just the name of the new last layer of your retrained model (with two neurons to classify female/male faces). If I understand it correctly you can give this layer any name (e.g. "final_result" as on AIY Vision page) - just make sure you use the same name when you test/run your model on the vision kit.

ritwikbiswas, 2018-07-07:
Thanks for the tip. So my error changes from `Output node named "foo" does not appear in the graph.` to `Unexpected error, use --debug flag to see crash log.`

This error indicates a more fundamental issue with either my training phase or arguments for the compiler. However running with the the debug flag doesn't seem to give me any useful crash information (unless I'm missing something). I've pasted the complete crash log below.

```
2018-07-07 05:06:47.942362: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 713 operators, 1031 arrays (0 quantized)
2018-07-07 05:06:47.972094: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 713 operators, 1031 arrays (0 quantized)
2018-07-07 05:06:47.988996: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 103 operators, 213 arrays (0 quantized)
2018-07-07 05:06:47.991367: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before dequantization graph transformations: 103 operators, 213 arrays (0 quantized)
```

Aileenlingyu, 2018-07-11:
Hi @minakhan01 @jkim0120 @dvillevald, I am having the same issue here, test my own compiled binary model and still have spicomm error and camera inference error, I upgrade to the latest Pi image but the error is still there, have you found out the solution?

algila, 2018-07-21:
TF 1.8, Ubuntu 18.04, bonnet_model_compiler 2018_04_26
I create my own model of SDD MobileNet 256x256 alpha 0,125 in keras then converted in TF. When I try to compile the file .pb the following error occur.

> ./bonnet_model_compiler.par \
>     --frozen_graph_path=./model_OLI.pb \
>     --output_graph_path=./model_OLI.binaryproto \
>     --input_tensor_name="input_1" \
>     --output_tensor_names="output_node0" \
>     --input_tensor_size=256 \
>     --debug
2018-07-21 22:51:06.232817: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.232946: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233054: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233245: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233336: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233444: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233528: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233608: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233686: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233741: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233793: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.233839: I external/org_tensorflow/tensorflow/contrib/lite/toco/import_tensorflow.cc:1268] Converting unsupported operation: Prod
2018-07-21 22:51:06.240375: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 765 operators, 1167 arrays (0 quantized)
2018-07-21 22:51:06.261099: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 765 operators, 1167 arrays (0 quantized)
2018-07-21 22:51:06.289453: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 177 operators, 370 arrays (0 quantized)
2018-07-21 22:51:06.293105: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before dequantization graph transformations: 177 operators, 370 arrays (0 quantized)
2018-07-21 22:51:06.295892: F external/org_tensorflow/tensorflow/contrib/lite/toco/toco_tooling.cc:46] Check failed: unsupported_ops.empty() These unsupported ops were not removed by graph transformations: Prod

See below the graph of my model .pb

[image]

closed 2020-08-19 by minakhan01